espressif/esp-idf #5615: Download firmwareGitHub release asset

kaizoku-oh, 2020-07-19:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [ESP32-Wrover-Kit|ESP32-DevKitC|ESP32-PICO-Kit|ESP32-LyraT|ESP32-LyraTD-MSC|none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROOM-32|ESP32-WROOM-32D|ESP32-WROOM-32U|ESP32-WROVER|ESP32-WROVER-I|ESP32-WROVER-B|ESP32-WROVER-IB|ESP32-SOLO-1|ESP32-PICO-D4|ESP32]
- IDF version (run ``git describe --tags`` to find it):
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [Make|CMake|idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Windows|Linux|macOS]
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: [No|Yes (please give details)]
- Power Supply:         [USB|external 5V|external 3.3V|Battery]

## Problem Description

//Detailed problem description goes here.

### Expected Behavior

### Actual Behavior

### Steps to reproduce

1. step1
2. ...

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{

}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

mahavirj, 2020-07-20:
@kaizoku-oh  

Can you please try with:
 - Certificate from https://github.com/espressif/esp-jumpstart/blob/master/6_ota/main/cloud_cfg/github_server.cert
 - URL: `https://raw.githubusercontent.com/wiki/espressif/esp-jumpstart/images/hello-world.bin`

(CC: @shubhamkulkarni97)

kaizoku-oh, 2020-07-20:
@mahavirj thanks for answering, the above certificate is already the same one that I'm using, and I just tried your URL and it worked, but still my problem is that I want to download the binary from github release page like the following: **https://github.com/kaizoku-oh/pio-esp32-ota-test/releases/download/v1.1.0/firmware.bin** and it dosen't work with github certificate, mbetls fails to verify the certificate.

shubhamkulkarni97, 2020-07-20:
@kaizoku-oh Please use the same URL you have shared above (i.e. https://github-production-release-asset-2e65be.s3.amazonaws.com/280750808/6f943180-c949-11ea-89d0-baff5f5c8431?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20200720%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20200720T055718Z&X-Amz-Expires=300&X-Amz-Signature=bd4701edebd72e2e5b87e9c6349a0a73a3b8d5c6d721d5e555eea16f56272907&X-Amz-SignedHeaders=host&actor_id=0&repo_id=280750808&response-content-disposition=attachment%3B%20filename%3Dfirmware.bin&response-content-type=application%2Foctet-stream) and certificate pasted below.

```
-----BEGIN CERTIFICATE-----
MIIEYzCCA0ugAwIBAgIQAYL4CY6i5ia5GjsnhB+5rzANBgkqhkiG9w0BAQsFADBa
MQswCQYDVQQGEwJJRTESMBAGA1UEChMJQmFsdGltb3JlMRMwEQYDVQQLEwpDeWJl
clRydXN0MSIwIAYDVQQDExlCYWx0aW1vcmUgQ3liZXJUcnVzdCBSb290MB4XDTE1
MTIwODEyMDUwN1oXDTI1MDUxMDEyMDAwMFowZDELMAkGA1UEBhMCVVMxFTATBgNV
BAoTDERpZ2lDZXJ0IEluYzEZMBcGA1UECxMQd3d3LmRpZ2ljZXJ0LmNvbTEjMCEG
A1UEAxMaRGlnaUNlcnQgQmFsdGltb3JlIENBLTIgRzIwggEiMA0GCSqGSIb3DQEB
AQUAA4IBDwAwggEKAoIBAQC75wD+AAFz75uI8FwIdfBccHMf/7V6H40II/3HwRM/
sSEGvU3M2y24hxkx3tprDcFd0lHVsF5y1PBm1ITykRhBtQkmsgOWBGmVU/oHTz6+
hjpDK7JZtavRuvRZQHJaZ7bN5lX8CSukmLK/zKkf1L+Hj4Il/UWAqeydjPl0kM8c
+GVQr834RavIL42ONh3e6onNslLZ5QnNNnEr2sbQm8b2pFtbObYfAB8ZpPvTvgzm
+4/dDoDmpOdaxMAvcu6R84Nnyc3KzkqwIIH95HKvCRjnT0LsTSdCTQeg3dUNdfc2
YMwmVJihiDfwg/etKVkgz7sl4dWe5vOuwQHrtQaJ4gqPAgMBAAGjggEZMIIBFTAd
BgNVHQ4EFgQUwBKyKHRoRmfpcCV0GgBFWwZ9XEQwHwYDVR0jBBgwFoAU5Z1ZMIJH
WMys+ghUNoZ7OrUETfAwEgYDVR0TAQH/BAgwBgEB/wIBADAOBgNVHQ8BAf8EBAMC
AYYwNAYIKwYBBQUHAQEEKDAmMCQGCCsGAQUFBzABhhhodHRwOi8vb2NzcC5kaWdp
Y2VydC5jb20wOgYDVR0fBDMwMTAvoC2gK4YpaHR0cDovL2NybDMuZGlnaWNlcnQu
Y29tL09tbmlyb290MjAyNS5jcmwwPQYDVR0gBDYwNDAyBgRVHSAAMCowKAYIKwYB
BQUHAgEWHGh0dHBzOi8vd3d3LmRpZ2ljZXJ0LmNvbS9DUFMwDQYJKoZIhvcNAQEL
BQADggEBAC/iN2bDGs+RVe4pFPpQEL6ZjeIo8XQWB2k7RDA99blJ9Wg2/rcwjang
B0lCY0ZStWnGm0nyGg9Xxva3vqt1jQ2iqzPkYoVDVKtjlAyjU6DqHeSmpqyVDmV4
7DOMvpQ+2HCr6sfheM4zlbv7LFjgikCmbUHY2Nmz+S8CxRtwa+I6hXsdGLDRS5rB
bxcQKegOw+FUllSlkZUIII1pLJ4vP1C0LuVXH6+kc9KhJLsNkP5FEx2noSnYZgvD
0WyzT7QrhExHkOyL4kGJE7YHRndC/bseF/r/JUuOUFfrjsxOFT+xJd1BDKCcYm1v
upcHi9nzBhDFKdT3uhaQqNBU4UtJx5g=
-----END CERTIFICATE-----
```

If you observe errors like `HTTP_HEADER: Buffer length is small to fit all the headers`, please increase value of `buffer_size_tx` in `esp_http_client_config_t` to 1024 (by default set to 512).

Please also note that the URL returns HTTP 403 Forbidden error even on host machine.

kaizoku-oh, 2020-07-20:
@shubhamkulkarni97 thanks for answering,
- in fact the url above returns 403 because it's available only for 300 secondes after you query https://github.com/kaizoku-oh/pio-esp32-ota-test/releases/download/v1.1.0/firmware.bin and next time it'll change, for this I have created a little webservice that can extract that final destination url from the github binary url and return it to esp32 as json, here is an example: https://github-ota-api.herokuapp.com/firmware/latest?github_username=kaizoku-oh&github_repository=pio-esp32-ota-test&device_current_fw_version=0.0.0 
- As for the above certificate, I tried and got the following error: **E (17112) esp_https_ota: Failed to establish HTTP connection**

shubhamkulkarni97, 2020-07-21:
> ```
>       esp_http_client_config_t ota_client_config =
>       {
>         .url = download_url,
>         .cert_pem = root_ca_cert_pem_start,
>         .buffer_size = 1024,
>       };
> ```

In shared configuration, you have set `buffer_size` to 1024. `buffer_size_tx` should be set to 1024 and not `buffer_size`.

Also, instead of using custom code, can you use `simple_ota_example` by entering OTA URL in menuconfig and using certificate shared above?

This will help eliminate issues with custom code.

kaizoku-oh, 2020-07-21:
@shubhamkulkarni97 @mahavirj  Thank you very much for your support, indeed increasing the `buffer_size_tx` did solve the problem.

mahavirj, 2020-07-22:
@kaizoku-oh Thanks for confirmation, will close the issue.

closed 2020-07-22 by mahavirj