espressif/esp-idf #5617: "Stack smashing" during cpu startup.

ataweg, 2020-07-20:
I did an update to the todays head revision of esp-idf and my project crash during the startup phase. The last commit which works fine was cf155161c40251d8f778a572f25d2821e70718d4  from 19.06.2020. Commit 4bc53d44e3de267651d4f499782065f41d09fcdd   also from 19.06.2020 has an compile error and commit 8bfccba598dc4f175cde202b4204ef73f9bf2eaf from 20.06.2020 already ends in the crash.
SInce 19.06.2020 there were a lot of changes in the startup procedure.
I think this is an esp-idf problem in the startup code. The crash happend while  "Intialize random stack guard".

The debug output after reset is released:
```

Hard resetting via RTS pin...
MONITOR
?[0;33m--- WARNING: GDB cannot open serial ports accessed as COMx?[0m
?[0;33m--- Using \\.\COM7 instead...?[0m
--- idf_monitor on \\.\COM7 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:9804
load:0x40078000,len:17408
load:0x40080400,len:4932
entry 0x40080720
I (28) boot: ESP-IDF v4.3-dev-472-gcf056a7d0-dirty 2nd stage bootloader
I (29) boot: compile time 22:52:24
D (30) bootloader_flash: mmu set block paddr=0x00000000 (was 0xffffffff)
I (37) boot: chip revision: 1
I (40) boot_comm: chip revision: 1, min. bootloader chip revision: 0
D (47) boot.esp32: magic e9
D (50) boot.esp32: segments 04
D (53) boot.esp32: spi_mode 02
D (56) boot.esp32: spi_speed 00
D (60) boot.esp32: spi_size 02
I (63) boot.esp32: SPI Speed      : 40MHz
I (67) boot.esp32: SPI Mode       : DIO
I (72) boot.esp32: SPI Flash Size : 4MB
D (76) boot: Enabling RTCWDT(9000 ms)
I (80) boot: Enabling RNG early entropy source...
D (85) bootloader_flash: mmu set paddr=00010000 count=1 size=c00 src_addr=10000 src_addr_aligned=10000
D (95) boot: mapped partition table 0x10000 at 0x3f400000
D (100) flash_parts: partition table verified, 4 entries
I (106) boot: Partition Table:
I (109) boot: ## Label            Usage          Type ST Offset   Length
D (117) boot: load partition table entry 0x3f400000
D (122) boot: type=1 subtype=2
I (125) boot:  0 nvs              WiFi data        01 02 00011000 00006000
D (132) boot: load partition table entry 0x3f400020
D (137) boot: type=1 subtype=1
I (141) boot:  1 phy_init         RF data          01 01 00017000 00001000
D (148) boot: load partition table entry 0x3f400040
D (153) boot: type=0 subtype=0
I (156) boot:  2 factory          factory app      00 00 00020000 00200000
I (164) boot: End of partition table
D (168) boot: Trying partition index -1 offs 0x20000 size 0x200000
D (174) esp_image: reading image header @ 0x20000
D (179) bootloader_flash: mmu set block paddr=0x00020000 (was 0xffffffff)
D (186) esp_image: image header: 0xe9 0x06 0x02 0x02 400812d0
I (192) boot_comm: chip revision: 1, min. application chip revision: 0
V (199) esp_image: loading segment header 0 at offset 0x20018
V (205) esp_image: segment data length 0x3df3c data starts 0x20020
V (211) esp_image: segment 0 map_segment 1 segment_data_offs 0x20020 load_addr 0x3f400020
I (219) esp_image: segment 0: paddr=0x00020020 vaddr=0x3f400020 size=0x3df3c (253756) map
D (228) esp_image: free data page_count 0x00000032
D (233) bootloader_flash: mmu set paddr=00020000 count=4 size=3df3c src_addr=20020 src_addr_aligned=20000
V (333) esp_image: loading segment header 1 at offset 0x5df5c
D (334) bootloader_flash: mmu set block paddr=0x00050000 (was 0xffffffff)
V (335) esp_image: segment data length 0x20b4 data starts 0x5df64
V (341) esp_image: segment 1 map_segment 0 segment_data_offs 0x5df64 load_addr 0x3ffb0000
I (349) esp_image: segment 1: paddr=0x0005df64 vaddr=0x3ffb0000 size=0x020b4 (  8372) load
[... 350 lines cut ...]
I (1871) spi_flash(154): flash io: dio
D (1876) stack_chk(29): Intialize random stack guard

Stack smashing protect failure!

abort() was called at PC 0x400d22e8 on core 0
0x400d22e8: __stack_chk_fail at C:/Espressif/esp-idf/components/esp_common/src/stack_check.c:36

Backtrace:0x40089284:0x3ffe3a70 0x40089c8d:0x3ffe3aa0 0x40092631:0x3ffe3ad0 0x400d22e8:0x3ffe3b50 0x40081553:0x3ffe3b70 0x400817ac:0x3ffe3ba0 0x40081419:0x3ffe3be0 0x40079ca9:0x3ffe3c20 |<-CORRU
PTED
0x40089284: panic_abort at C:/Espressif/esp-idf/components/esp_system/panic.c:349

0x40089c8d: esp_system_abort at C:/Espressif/esp-idf/components/esp_system/system_api.c:105

0x40092631: abort at C:/Espressif/esp-idf/components/newlib/abort.c:46

0x400d22e8: __stack_chk_fail at C:/Espressif/esp-idf/components/esp_common/src/stack_check.c:36

0x40081553: do_global_ctors at C:/Espressif/esp-idf/components/esp_system/startup.c:135

0x400817ac: start_cpu0_default at C:/Espressif/esp-idf/components/esp_system/startup.c:333

0x40081419: call_start_cpu0 at C:/Espressif/esp-idf/components/esp_system/port/cpu_start.c:418

```

ataweg, 2020-07-21:
My ESP32-Camera project based on https:\\github.com\espressif\esp32-camera.git and  https:\\github.com\espressif\esp-who\tree\master\examples\single_chip\camera_web_server doesn't have this problem.  So I think now it can be project specific issue. An interesting problem, how far is the execution of the startup code project-specific, apart from the memory initialization of the data segment and memory allocation?

igrr, 2020-07-21:
Hi @ataweg, this is indeed an issue introduced in the startup code. As a workaround, you may temporarily disable stack smashing protection using CONFIG_COMPILER_STACK_CHECK_MODE Kconfig option: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-compiler-stack-check-mode.

ataweg, 2020-07-21:
it works fine with CONFIG_COMPILER_STACK_CHECK_MODE_NORM=y or CONFIG_COMPILER_STACK_CHECK_MODE_STRONG=y
it fails with CONFIG_COMPILER_STACK_CHECK_MODE_ALL=y

gecko242, 2020-07-21:
@igrr I too am seeing this issue. Disabling stack smashing protection works as a workaround.

renzbagaporo, 2020-07-22:
An internal fix has been submitted and is under review.

closed 2020-08-03 by espressif-bot