espressif/esp-idf #5618: Add a function to read bytes form uart with a time limit from last byte

romocitto88, 2020-07-21:
**UART bytes collection**

I can read a maximum of only 128 bytes per times when "int uart_read_bytes(uart_port_t uart_num, uint8_t *buf, uint32_t length, TickType_t ticks_to_wait)"  is executed because Uart Fifo is 128 , the maximum support on hardware is 128 bytes.

I think that in future should be usefull implemented a function to read the bytes form uart with a time limit from last byte and not a maximum limit time of waiting.

wilhelmw201, 2020-07-21:
hi,
have you tried using the RX buffer by setting the argument "rx_buffer_size" in uart_driver_install to a large number (e.g. 1024)? You can then call uart_read_bytes to recieve more data.

wilhelmw201, 2020-07-21:
hi,
this should not be a problem in the newest version of the IDF. Please take a look the example uart_async_rxtxtasks, it is able to recieve more than 128 bytes. (Maybe you need to change the sent message in tx_task to a longer one.)

wilhelmw201, 2020-07-21:
hi,
can you be more specific in what you are trying to do? What do you mean by "from last byte"?

romocitto88, 2020-07-21:
Ciao,
la funzione "int uart_read_bytes (uart_port_t uart_num, uint8_t * buf, uint32_t length, TickType_t ticks_to_wait)" scrive al massimo 128 byte in "* buf" per volta (c'è qualche argomento a riguardo). Vorrei che la funzione restituisse tutti i byte in * buf e se possibile quando, dopo aver ricevuto l'ultimo byte, passa un tempo definito dall'utente, anzichè impostare il tempo massimo di ascolto.

romocitto88, 2020-07-21:
For the second question, I try to explain with an example:
I'm getting a frame of xxx bytes and I don't know how many are coming.
For each byte received, the function should start a timer defined by a parameter passed to the function before declaring the end of the frame, so as to have the whole complete frame at one time

wilhelmw201, 2020-07-22:
> The function "int uart_read_bytes (uart_port_t uart_num, uint8_t * buf, uint32_t length, TickType_t ticks_to_wait)" "writes a maximum of 128 bytes in" * buf "at a time (there is some argument about it). I would like the function to return all the bytes in * buf and if possible when, after receiving the last byte, a user-defined time passes, instead of setting the maximum listening time.

This is already the behavior in the current IDF. uart_driver_install lets you specify the software buffer length. You are reading from the software buffer (and not from the hardware FIFO), when you call uart_read_bytes. As a result, you can read more than 128 bytes. 

The wait behavior of the current uart_read_bytes is, if it does not read anything from the software buffer for ticks_to_wait, then it returns. Any successful read resets this timer back to ticks_to_wait.

> For the second question, I try to explain with an example:
> I'm getting a frame of xxx bytes and I don't know how many are coming.
> For each byte received, the function should start a timer defined by [...]

Just to check that I have understood you correctly, do you need a way of knowing that the driver recieved nothing for a predefined amount of time?

wilhelmw201, 2020-07-22:
By the way, you can use a break signal to signal the end of a frame :)

romocitto88, 2020-07-22:
> This is already the behavior in the current IDF. uart_driver_install lets you specify the software buffer length. You are reading from the software buffer (and not from the hardware FIFO), when you call uart_read_bytes. As a result, you can read more than 128 bytes. If you really can't, it may be because of some strange problems, please post some code excerpts so that I can have a look at it.

If you search in google, you can find a lot of result with this problem. The function returns always 120 bytes per times. I next day I try to prepare some codes to reproduce this behavior.

>Just to check that I have understood you correctly, do you need a way of knowing that the driver recieved nothing for a predefined amount of time?

Yes

wilhelmw201, 2020-07-27:
Hi,
for your first problem, try this lightly modified version of the example uart_async_rxtxtasks:
`#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"
#include "esp_log.h"
#include "driver/uart.h"
#include "string.h"
#include "driver/gpio.h"

static const int RX_BUF_SIZE = 1024;

#define TXD_PIN (GPIO_NUM_4)
#define RXD_PIN (GPIO_NUM_5)

void init(void) {
    const uart_config_t uart_config = {
        .baud_rate = 115200,
        .data_bits = UART_DATA_8_BITS,
        .parity = UART_PARITY_DISABLE,
        .stop_bits = UART_STOP_BITS_1,
        .flow_ctrl = UART_HW_FLOWCTRL_DISABLE,
        .source_clk = UART_SCLK_APB,
    };
    // We won't use a buffer for sending data.
    uart_driver_install(UART_NUM_1, RX_BUF_SIZE * 2, 0, 0, NULL, 0);
    uart_param_config(UART_NUM_1, &uart_config);
    uart_set_pin(UART_NUM_1, TXD_PIN, RXD_PIN, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE);
}

static void tx_task(void *arg)
{
    static const char *TX_TASK_TAG = "TX_TASK";
    static const char some_data[2000] = {};
    esp_log_level_set(TX_TASK_TAG, ESP_LOG_INFO);
    while (1) {
        uart_write_bytes(UART_NUM_1, some_data, 2000);
        vTaskDelay(2000 / portTICK_PERIOD_MS);
    }
}

static void rx_task(void *arg)
{
    static const char *RX_TASK_TAG = "RX_TASK";
    esp_log_level_set(RX_TASK_TAG, ESP_LOG_INFO);
    uint8_t* data = (uint8_t*) malloc(RX_BUF_SIZE+1);
    while (1) {
        const int rxBytes = uart_read_bytes(UART_NUM_1, data, RX_BUF_SIZE, 1000 / portTICK_RATE_MS);
        if (rxBytes > 0) {
            data[rxBytes] = 0;
            ESP_LOGI(RX_TASK_TAG, "Read %d bytes: '%s'", rxBytes, data);
        }
    }
    free(data);
}

void app_main(void)
{
    init();
    xTaskCreate(rx_task, "uart_rx_task", 1024*2, NULL, configMAX_PRIORITIES, NULL);
    xTaskCreate(tx_task, "uart_tx_task", 1024*2, NULL, configMAX_PRIORITIES-1, NULL);
}
`
You should get outputs like this:
I (89124) RX_TASK: Read 1024 bytes: ''
I (90204) RX_TASK: Read 976 bytes: ''
I (91284) RX_TASK: Read 1024 bytes: ''

wilhelmw201, 2020-07-27:
For the second problem, again a lightly modified version of the example

```
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"
#include "esp_log.h"
#include "driver/uart.h"
#include "string.h"
#include "driver/gpio.h"

static const int RX_BUF_SIZE = 256;

#define TXD_PIN (GPIO_NUM_4)
#define RXD_PIN (GPIO_NUM_5)

void init(void) {
    const uart_config_t uart_config = {
        .baud_rate = 115200,
        .data_bits = UART_DATA_8_BITS,
        .parity = UART_PARITY_DISABLE,
        .stop_bits = UART_STOP_BITS_1,
        .flow_ctrl = UART_HW_FLOWCTRL_DISABLE,
        .source_clk = UART_SCLK_APB,
    };
    // We won't use a buffer for sending data.
    uart_driver_install(UART_NUM_1, RX_BUF_SIZE * 2, 0, 0, NULL, 0);
    uart_param_config(UART_NUM_1, &uart_config);
    uart_set_pin(UART_NUM_1, TXD_PIN, RXD_PIN, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE);
}

static void tx_task(void *arg)
{
    static const char *TX_TASK_TAG = "TX_TASK";
    static const char hi[] = "hi! ";
    esp_log_level_set(TX_TASK_TAG, ESP_LOG_INFO);
    while (1) {
        uart_write_bytes(UART_NUM_1, hi, 4);
        vTaskDelay(90 / portTICK_PERIOD_MS);
    }
}

static void rx_task(void *arg)
{
    static const char *RX_TASK_TAG = "RX_TASK";
    esp_log_level_set(RX_TASK_TAG, ESP_LOG_INFO);
    uint8_t* data = (uint8_t*) malloc(RX_BUF_SIZE+1);
    while (1) {
        const int rxBytes = uart_read_bytes(UART_NUM_1, data, RX_BUF_SIZE, 100 / portTICK_RATE_MS);
        if (rxBytes > 0) {
            data[rxBytes] = 0;
            ESP_LOGI(RX_TASK_TAG, "Read %d bytes: '%s'", rxBytes, data);
        } else
        {
            ESP_LOGI(RX_TASK_TAG, "Timeout.");
        }
        
    }
    free(data);
}

void app_main(void)
{
    init();
    xTaskCreate(rx_task, "uart_rx_task", 1024*2, NULL, configMAX_PRIORITIES, NULL);
    xTaskCreate(tx_task, "uart_tx_task", 1024*2, NULL, configMAX_PRIORITIES-1, NULL);
}

```

You should recieve a lot of hi-s at the same time. This is because every 90ms data is received, which sets back the 100-ms timer of the recieve function.

Alvin1Zhang, 2020-08-13:
@romocitto88  Thanks for reporting, would you please help share if any updates? Thanks.

o-marshmallow, 2021-11-18:
Closing this issue, feel free to create a new issue if the problem persists.

closed 2021-11-18 by o-marshmallow