espressif/esp-idf #5619: esp-idf 4.2   ppp 拨号

coffice12, 2020-07-21:
esp3.3能够拨号成功

然后4.2版本：ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:6964
load:0x40078000,len:14332
load:0x40080400,len:3980
entry 0x40080680
I (29) boot: ESP-IDF v4.2-dev-1856-g00148cd0c 2nd stage bootloader
I (29) boot: compile time 15:16:27
I (29) boot: chip revision: 1
I (33) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (40) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 4MB
I (54) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot: End of partition table
I (97) boot_comm: chip revision: 1, min. application chip revision: 0
I (104) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x191cc (102860) map
I (152) esp_image: segment 1: paddr=0x000291f4 vaddr=0x3ffbdb60 size=0x0235c (  9052) load
I (156) esp_image: segment 2: paddr=0x0002b558 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /Users/tangjunhua/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730

I (159) esp_image: segment 3: paddr=0x0002b960 vaddr=0x40080400 size=0x046b8 ( 18104) load
I (176) esp_image: segment 4: paddr=0x00030020 vaddr=0x400d0020 size=0x612c8 (398024) map
0x400d0020: _stext at ??:?

I (327) esp_image: segment 5: paddr=0x000912f0 vaddr=0x40084ab8 size=0x09f50 ( 40784) load
0x40084ab8: prvCheckItemFitsDefault at /Users/tangjunhua/esp-idf/components/esp_ringbuf/ringbuf.c:277 (discriminator 3)

I (353) boot: Loaded app from partition at offset 0x10000
I (353) boot: Disabling RNG early entropy source...
I (353) cpu_start: Pro cpu up.
I (357) cpu_start: Application information:
I (362) cpu_start: Project name:     pppos_client
I (367) cpu_start: App version:      v4.2-dev-1856-g00148cd0c-dirty
I (374) cpu_start: Compile time:     Jul 21 2020 15:19:29
I (380) cpu_start: ELF file SHA256:  4e1deab603af1fbd...
I (386) cpu_start: ESP-IDF:          v4.2-dev-1856-g00148cd0c-dirty
I (393) cpu_start: Starting app cpu, entry point is 0x400815dc
0x400815dc: call_start_cpu1 at /Users/tangjunhua/esp-idf/components/esp32/cpu_start.c:282

I (385) cpu_start: App cpu up.
I (404) heap_init: Initializing. RAM available for dynamic allocation:
I (410) heap_init: At 3FFAE6E0 len 0000F480 (61 KiB): DRAM
I (417) heap_init: At 3FFC1858 len 0001E7A8 (121 KiB): DRAM
I (423) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (429) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (436) heap_init: At 4008EA08 len 000115F8 (69 KiB): IRAM
I (442) cpu_start: Pro cpu start user code
I (460) spi_flash: detected chip: generic
I (461) spi_flash: flash io: dio
I (461) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (469) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (479) gpio: GPIO[5]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (489) gpio: GPIO[23]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (499) uart: queue free spaces: 30
E (659) esp-modem: esp_dte_handle_line(91): handle line failed
W (659) pppos_example: Unknow line received: +CPIN: READY

E (1459) esp-modem: esp_dte_handle_line(91): handle line failed
W (1459) pppos_example: Unknow line received: +COPS: 0

I (1759) pppos_example: Module: SIMCOM_SIM800L
I (1759) pppos_example: Operator: 
I (1759) pppos_example: IMEI: 863576043219489
I (1759) pppos_example: IMSI: 460012592624115
I (1959) pppos_example: rssi: 0, ber: 0
I (2159) pppos_example: Battery voltage: 4049 mV
I (2359) pppos_example: Modem PPP Started
I (8559) esp-netif_lwip-ppp: Connection lost
I (8559) pppos_example: Modem Disconnect from PPP Server

是什么原因呢 ？硬件确定没有问题。

Alvin1Zhang, 2020-07-21:
Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

gecko242, 2020-07-21:
I experienced something similar to this, and it seemed to be due to the lack of an option to set the APN. 

Delving into the modem component:

`
esp_err_t esp_modem_start_ppp(modem_dte_t *dte)
{
    modem_dce_t *dce = dte->dce;
    MODEM_CHECK(dce, "DTE has not yet bind with DCE", err);
    esp_modem_dte_t *esp_dte = __containerof(dte, esp_modem_dte_t, parent);
    /* Set PDP Context */
    MODEM_CHECK(dce->define_pdp_context(dce, 1, "IP", CONFIG_EXAMPLE_COMPONENT_MODEM_APN) == ESP_OK, "set MODEM APN failed", err);
    /* Enter PPP mode */
    MODEM_CHECK(dte->change_mode(dte, MODEM_PPP_MODE) == ESP_OK, "enter ppp mode failed", err);

    /* post PPP mode started event */
    esp_event_post_to(esp_dte->event_loop_hdl, ESP_MODEM_EVENT, ESP_MODEM_EVENT_PPP_START, NULL, 0, 0);
    return ESP_OK;
err:
    return ESP_FAIL;
}
`
`CONFIG_EXAMPLE_COMPONENT_MODEM_APN` does not appear to be an option in menuconfig. 

replacing manually with the correct APN fixed the example for me.

will-emmerson, 2020-08-12:
CONFIG_EXAMPLE_COMPONENT_MODEM_APN is in menuconfig under component config > esp-modem

david-cermak, 2020-08-20:
@coffice12 Any update for your side? Have you tried to set the correct `APN` in menuconfig?

@will-emmerson @gecko242 Thanks for your help with this issue!

will-emmerson, 2020-08-20:
Off topic but I feel like it would make more sense if the modem component inside the pppos tutorial was just a normal idf component, then it might be more obvious where `CONFIG_EXAMPLE_COMPONENT_MODEM_APN` is.

gecko242, 2020-08-21:
@will-emmerson I think this is due to the fact that the modem component is not finished yet, (its still pretty buggy), and normally components do not get included until they are fully tested.

coffice12, 2020-09-08:
> @ coffice12您身边有任何更新吗？您是否尝试`APN`在menuconfig中设置正确的？
> 
> @ will-emmerson @ gecko242感谢您在此问题上的帮助！

apn config UNICOM    is ok

closed 2020-09-08 by coffice12