espressif/esp-idf #5620: Nearly same function name

ankit-thealchemist, 2020-07-21:
https://github.com/espressif/esp-idf/blob/cf056a7d0b90261923b8207f21dc270313b67456/examples/bluetooth/esp_ble_mesh/ble_mesh_sensor_model/sensor_server/main/main.c#L339

Hello,
In the above example, the two function name "example_ble_mesh_send_sensor_setting_status" and "example_ble_mesh_send_sensor_settings_status" having the almost same name, only difference is the 's' in the settings. Please avoid these similar names of the function/variable as they create the confusion in the readers mind. 
Thanks!

Alvin1Zhang, 2020-07-21:
Thanks for reporting and letting us know, we will look into.

Campou, 2020-07-22:
Hi @ankit-thealchemist 

Sorry for the confusion caused.

`Sensor Settings Status` & `Sensor Setting Status` are just two SIG-defined model status messages, so in the example I just named the functions based on the names of these two messages.

And if some better ideas come on your side, please feel free to submit a PR for the changes.

Thanks.

ankit-thealchemist, 2020-07-22:
@Campou thanks for the reply!
I can understand it now! As I am new to the ble mesh, I don't know it. Do we have any resource that have all the models related knowledge such as model status messages, structure, working etc.

closed 2020-07-22 by ankit-thealchemist

Campou, 2020-07-30:
Hi @ankit-thealchemist 

You can go to the bluetooth official website and download the Mesh Model specification, which includes all the model related definitions (i.e. messages, behaviors, etc.)

Thanks.