espressif/esp-idf #5621: BLE Gap Scan stops and locks esp32

fbgencer, 2020-07-21:
## Environment

- Development Kit:      [none]
- Module or chip used:  [ESP32-WROOM-32U]
- IDF version:
    // v4.2-dev-1206-g741960d5c
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 8.2.0
- Operating System:     [Linux]

## Problem Description
Normally, I am using BLE scan near my house and encountered no problem and saw at most 10 devices in the scan results.  Now I changed my location and there are at least 200 BLE devices around and during the scan results (gap callback function invokes ESP_GAP_BLE_SCAN_RESULT_EVT) stops at some point and esp does not respond anything until the next reset.

//Detailed problem description goes here.
I thought my code could not be identify the error I am getting, I tried to regenerate the same behaviour with the provided esp32 examples. Here I used gattc_multi_connect example but there is no connection established, I only used scan part of the code. 

### Expected Behavior
In the menuconfig section there are two parts "BLE adv report flow control number" and "BLE adv lost event threshold value", I assumed this is related with the scan results and increased them both 1000, but still problem continues.

### Code to reproduce this issue
My code is here.
https://gist.github.com/fbgencer/01b5da0f0d35c464363f3e86475c9b9d

Additionally I added a flag when scan stops task stars another scan immediately. This may seem redundant but my actual device used this kind of approach for ble scan.

## Debug Logs
For the debug logs, in the menuconfig, Component config → Bluetooth → Bluedroid Options → BT DEBUG LOG LEVEL list are changed for BTM and GAP Layer (now it is VERBOSE).
```
I (2741) BT_BTM: BTM_CheckAdvData type=0x01
D (2751) BT_BTC: btc_transfer_context msg 0 5 2 0x3ffce0b8

I (2751) BT_BTM: BTM_CheckAdvData type=0x19
D (2761) BT_BTC: btc_transfer_context msg 0 5 2 0x3ffce0b8

I (2761) BT_BTM: BTM_CheckAdvData type=0x03
D (2771) BT_BTC: btc_transfer_context msg 0 5 2 0x3ffce0b8

D (2771) BT_BTM: BR/EDR NOT SUPPORT bit set, LE only device
D (2781) BT_BTC: btc_transfer_context msg 0 5 2 0x3ffce0b8

D (2791) BT_BTC: btc_transfer_context msg 0 5 2 0x3ffce0b8

I (2791) BT_BTM: BTM_CheckAdvData type=0x01
D (2801) BT_BTC: btc_transfer_context msg 0 5 2 0x3ffce0b8

D (2801) BT_APPL: bta_dm_observe_results_cb
D (2811) BT_BTC: btc_transfer_context msg 0 5 2 0x3ffce0b8

I (2811) BT_BTM: BTM_InqDbRead: bd addr [788d5b7230d7]

D (2821) BT_BTC: btc_transfer_context msg 0 5 2 0x3ffce0b8

D (2821) BT_BTC: btc_transfer_context msg 1 5 3 0x3ffcc110

D (2831) BT_BTC: btc_transfer_context msg 0 5 2 0x3ffce0b8

D (2841) BT_BTC: btc_transfer_context msg 0 5 2 0x3ffce0b8

I (2841) BT_BTM: BTM_CheckAdvData type=0x01
I (2841) BT_BTM: BTM_CheckAdvData type=0x19
D (2851) BT_BTC: btc_transfer_context msg 0 5 2 0x3ffce0b8

I (2861) BT_BTM: BTM_CheckAdvData type=0x03
D (2861) BT_BTM: Random address, treating device as LE only
I (2871) BT_BTM: BTM_CheckAdvData type=0x01
D (2871) BT_APPL: bta_dm_observe_results_cb
I (2881) BT_BTM: BTM_InqDbRead: bd addr [223457cf7ac0]

D (2881) BT_BTC: btc_transfer_context msg 1 5 3 0x3ffcc110

D (2891) BT_BTC: Unhandled deep free 2

D (2891) BT_BTC: btc_thread_handler msg 0 5 2 0x3ffba244

D (2901) BT_BTC: btc_transfer_context msg 0 5 2 0x3ffce0b8

D (2901) BT_BTC: btc_gap_ble_call_handler act 2

I (2911) BT_APPL: BTA_DmBleScan:start = 1 
D (3941) BT_BTC: btc_transfer_context msg 0 10 128 0x3ffb7628
```

## sdkconfig and elf file

[gattc_multi_connect_to_github.zip](https://github.com/espressif/esp-idf/files/4953657/gattc_multi_connect_to_github.zip)

Alvin1Zhang, 2020-07-22:
Thanks for reporting and letting us know, we will look into.

WCCWCC, 2020-07-22:
Hi, @fbgencer 

**reason**: 
If there are too many devices in the environment, the event ESP_GAP_BLE_SCAN_RESULT_EVT will be triggered frequently.

You cannot do time-consuming operations in ESP_GAP_BLE_SCAN_RESULT_EVT, such as printing. It takes about 10ms to print once

**Solution**：
You can create a new task to receive ESP_GAP_BLE_SCAN_RESULT_EVT information to the queue, and then print the scan results.

You also need to change the baud rate for printing to 921600.

fbgencer, 2020-07-22:
Thank you for your responses. 

Actually I also realized printing the results causes the issue, so it is about timing. In the original code I am using `esp_ble_resolve_adv_data` to get the name of the device (in my case their MAC address is not important ) and check the name. If it satisfies then their properties are saved in an array. I do not need to print them, it is just for debugging. 

But I also tried after scan start, I put a semaphore in order to wait the end of the scan results and when `ESP_GAP_SEARCH_INQ_CMPL_EVT` is triggered I am giving the semaphore to make sure I don't do anything during `ESP_GAP_BLE_SCAN_RESULT_EVT`. There was no printing just checking the name and saving into an array. Is this also problematic since in RTOS level probably there are some timing managements for semaphores. 

I will try your suggestions and notify you about the result.

goro9, 2020-11-27:
さっきのメモ
起動後3時間くらいでscanが死んでるっぽい

```
2020-11-21 00:55:29,773 : [0;32mI (554) cpu_start: App version:      v1.2.6[0m
```

```
2020-11-21 03:48:44,952 : [0;32mI (10395451) ble_qrio: investigation: qrio_exec_log, 860[0m
2020-11-21 03:48:44,991 : [0;32mI (10395451) ble_qrio: investigation: qrio_exec_log, 872[0m
2020-11-21 03:48:44,992 : [0;32mI (10395461) ble_qrio: investigation: qrio_main, 912[0m
2020-11-21 03:48:44,993 : [0;32mI (10395461) ble_linking: --- print scanned devices ---[0m
2020-11-21 03:48:44,993 : [0;32mI (10395471) ble_linking: device=0[0m
2020-11-21 03:48:44,994 : [0;32mI (10395471) ble_linking: device=1[0m
2020-11-21 03:48:44,995 : [0;32mI (10395481) ble_linking: addr=[0m
2020-11-21 03:48:44,995 : [0;32mI (10395481) ble_linking: 0x3ffbb4c1   7b 08 d4 05 26 70                                 |{...&p|[0m
2020-11-21 03:48:44,996 : [0;32mI (10395491) ble_linking: last_detected_at=10335625483, adv_data=[0m
2020-11-21 03:48:45,034 : [0;32mI (10395501) ble_linking: 0x3ffbb4e4   02 01 05 02 0a 00 13 ff  35 02 cf 1b 36 b5 99 1a  |........5...6...|[0m
2020-11-21 03:48:45,035 : [0;32mI (10395511) ble_linking: 0x3ffbb4f4   30 e7 d8 4a 10 00 06 04  00 00                    |0..J......|[0m
2020-11-21 03:48:45,036 : [0;32mI (10395521) ble_linking: addr=[0m
2020-11-21 03:48:45,037 : [0;32mI (10395521) ble_linking: 0x3ffe8b65   37 8c d4 05 26 70                                 |7...&p|[0m
2020-11-21 03:48:45,037 : [0;32mI (10395531) ble_linking: last_detected_at=10335567062, adv_data=[0m
2020-11-21 03:48:45,061 : [0;32mI (10395541) ble_linking: 0x3ffedb1c   02 01 05 02 0a 00 13 ff  35 02 5d 2e e3 7b 2c 47  |........5.]..{,G|[0m
2020-11-21 03:48:45,062 : [0;32mI (10395551) ble_linking: 0x3ffedb2c   0e 66 94 ce 10 00 44 01  00 00                    |.f....D...|[0m
2020-11-21 03:48:46,081 : [0;32mI (10396561) ble_qrio: investigation: qrio_main, 909[0m
2020-11-21 03:48:46,082 : [0;32mI (10396561) ble_qrio: investigation: qrio_exec_log, 850[0m
2020-11-21 03:48:46,083 : [0;32mI (10396561) ble_qrio: investigation: qrio_advertisings_new, 790[0m
2020-11-21 03:48:46,083 : [0;32mI (10396561) ble_qrio: investigation: qrio_advertisings_new, 792[0m
2020-11-21 03:48:46,084 : [0;32mI (10396571) ble_qrio: investigation: qrio_advertisings_new, 821[0m
2020-11-21 03:48:46,084 : [0;32mI (10396581) ble_qrio: investigation: qrio_exec_log, 860[0m
2020-11-21 03:48:46,085 : [0;32mI (10396581) ble_qrio: investigation: qrio_exec_log, 872[0m
2020-11-21 03:48:46,097 : [0;32mI (10396591) ble_qrio: investigation: qrio_main, 912[0m
2020-11-21 03:48:47,108 : [0;32mI (10397591) ble_qrio: investigation: qrio_main, 909[0m
2020-11-21 03:48:47,109 : [0;32mI (10397591) ble_qrio: investigation: qrio_exec_log, 850[0m
2020-11-21 03:48:47,110 : [0;32mI (10397591) ble_qrio: investigation: qrio_advertisings_new, 790[0m
2020-11-21 03:48:47,111 : [0;32mI (10397591) ble_qrio: investigation: qrio_advertisings_new, 792[0m
2020-11-21 03:48:47,111 : [0;32mI (10397601) ble_qrio: investigation: qrio_advertisings_new, 821[0m
2020-11-21 03:48:47,112 : [0;32mI (10397611) ble_qrio: investigation: qrio_exec_log, 860[0m
2020-11-21 03:48:47,128 : [0;32mI (10397611) ble_qrio: investigation: qrio_exec_log, 872[0m
2020-11-21 03:48:47,129 : [0;32mI (10397621) ble_qrio: investigation: qrio_main, 912[0m
2020-11-21 03:48:48,138 : [0;32mI (10398621) ble_qrio: investigation: qrio_main, 909[0m
2020-11-21 03:48:48,140 : [0;32mI (10398621) ble_qrio: investigation: qrio_exec_log, 850[0m
2020-11-21 03:48:48,140 : [0;32mI (10398621) ble_qrio: investigation: qrio_advertisings_new, 790[0m
2020-11-21 03:48:48,141 : [0;32mI (10398621) ble_qrio: investigation: qrio_advertisings_new, 792[0m
2020-11-21 03:48:48,142 : [0;32mI (10398631) ble_qrio: investigation: qrio_advertisings_new, 821[0m
2020-11-21 03:48:48,143 : [0;32mI (10398641) ble_qrio: investigation: qrio_exec_log, 860[0m
2020-11-21 03:48:48,160 : [0;32mI (10398641) ble_qrio: investigation: qrio_exec_log, 872[0m
2020-11-21 03:48:48,161 : [0;32mI (10398651) ble_qrio: investigation: qrio_main, 912[0m
2020-11-21 03:48:49,168 : [0;32mI (10399651) ble_qrio: investigation: qrio_main, 909[0m
2020-11-21 03:48:49,169 : [0;32mI (10399651) ble_qrio: investigation: qrio_exec_log, 850[0m
2020-11-21 03:48:49,170 : [0;32mI (10399651) ble_qrio: investigation: qrio_advertisings_new, 790[0m
2020-11-21 03:48:49,171 : [0;32mI (10399651) ble_qrio: investigation: qrio_advertisings_new, 792[0m
2020-11-21 03:48:49,171 : [0;32mI (10399661) ble_qrio: investigation: qrio_advertisings_new, 821[0m
2020-11-21 03:48:49,172 : [0;32mI (10399671) ble_qrio: investigation: qrio_exec_log, 860[0m
2020-11-21 03:48:49,188 : [0;32mI (10399671) ble_qrio: investigation: qrio_exec_log, 872[0m
2020-11-21 03:48:49,189 : [0;32mI (10399681) ble_qrio: investigation: qrio_main, 912[0m
2020-11-21 03:48:50,198 : [0;32mI (10400681) ble_qrio: investigation: qrio_main, 909[0m
2020-11-21 03:48:50,199 : [0;32mI (10400681) ble_qrio: investigation: qrio_exec_log, 850[0m
2020-11-21 03:48:50,200 : [0;32mI (10400681) ble_qrio: investigation: qrio_advertisings_new, 790[0m
2020-11-21 03:48:50,201 : [0;32mI (10400681) ble_qrio: investigation: qrio_advertisings_new, 792[0m
2020-11-21 03:48:50,201 : [0;32mI (10400691) ble_qrio: investigation: qrio_advertisings_new, 821[0m
2020-11-21 03:48:50,202 : [0;32mI (10400701) ble_qrio: investigation: qrio_exec_log, 860[0m
[... cut ...]
2020-11-21 03:48:50,233 : [0;32mI (10400711) ble_linking: --- print scanned devices ---[0m
2020-11-21 03:48:50,233 : [0;32mI (10400721) ble_linking: device=0[0m
2020-11-21 03:48:50,234 : [0;32mI (10400721) ble_linking: device=1[0m
2020-11-21 03:48:51,235 : [0;32mI (10401731) ble_qrio: investigation: qrio_main, 909[0m
2020-11-21 03:48:51,236 : [0;32mI (10401731) ble_qrio: investigation: qrio_exec_log, 850[0m
2020-11-21 03:48:51,237 : [0;32mI (10401731) ble_qrio: investigation: qrio_main, 912[0m
```

WCCWCC, 2020-11-27:
Hi ,
Your question is scan stops abnormally? 
In the scenario of coexistence with WiFi, scan may stop abnormally, and then the data packet cannot be scanned
This commit can solve your problem:https://github.com/espressif/esp-idf/commit/e5af56db411daf9bdabbf4d8dc020e2fc30f84d7

If it is not the problem, please describe it in detail.

Alvin1Zhang, 2021-02-26:
@fbgencer Thanks for reporting, would you please help share if any further updates for this issue? Thanks.

Alvin1Zhang, 2021-08-18:
Thanks for reporting, will close due to short of feedback, feel free to reopen with more details.

closed 2021-08-18 by Alvin1Zhang