espressif/esp-idf #5622: tcp_server.c SO_REUSEADDR not working as expected

ZweiEuro, 2020-07-21:
## Environment

- Development Kit:     ESP32 Devkit V1 -Doit (I think, it's soldered i can't look at it anymore) (here)[https://docs.espressif.com/projects/esp-idf/en/latest/esp32/hw-reference/modules-and-boards.html#esp32-wroom-32]
- Kit version -
- Module or chip used:  ESP32-WROOM-32

- IDF version (run ``git describe --tags`` to find it): framework-espidf 3.40001.200521 (4.0.1)  (PIO)
- Build System:         platformio
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0

- Operating System:   Linux

- Using an IDE?:  Vscode platformio
- Power Supply:         USB

## Problem Description
(The configs:
 CONFIG_LWIP_SO_REUSE=y
CONFIG_LWIP_SO_REUSE_RXTOALL=y 
are set)
the port i am using is : 60583
i am also using the ipv4 example
adding:
```
{
          int i = 1;
          int ret = setsockopt(listen_sock, SOL_SOCKET, SO_REUSEADDR, &i, sizeof(i));
          ESP_LOGE(TAG, "setsockopt %d  %d %s ", ret, errno, strerror(errno));
        }
```

into:
https://github.com/espressif/esp-idf/blob/48ea44f3d1e4b5763a2d0b65bc5c56a84690e910/examples/protocols/sockets/tcp_server/main/tcp_server.c#L65

Does not make the address reusable.

### Expected Behavior

the socket should be reusing the address freely.

### Actual Behavior

Without socketopt:

```

I (2166) wifi:AP's beacon interval = 102400 us, DTIM period = 2
␛[0;32mI (3076) tcpip_adapter: sta ip: 192.168.43.23, mask: 255.255.255.0, gw: 192.168.43.1␛[0m
␛[0;32mI (3076) example_connect: Connected to Meiner␛[0m
␛[0;32mI (3076) example_connect: IPv4 address: 192.168.43.23␛[0m
␛[0;32mI (3086) example: Socket created␛[0m
␛[0;32mI (3086) example: Socket bound, port 60583␛[0m
␛[0;32mI (3096) example: Socket listening␛[0m
␛[0;32mI (5976) example: Socket accepted␛[0m
␛[0;32mI (5986) example: Received 86 bytes from 192.168.43.62:␛[0m
␛[0;32mI (5986) example: {"event":"mod_control","id":1336146744,"h_id":"25289830","is_old":false,"state":false}␛[0m
␛[0;32mI (8216) example: Connection closed␛[0m
␛[0;31mE (8216) example: Shutting down socket and restarting...␛[0m
␛[0;32mI (8216) example: Socket created␛[0m
␛[0;31mE (8216) example: Socket unable to bind: errno 112␛[0m
```

112 -> Addr in use ? I thought the socket was shut down when it was reacreated.
This is where I found the flag reuse and thought it may fix my problem.

with socketopt:
```
␛[0;32mI (3079) example_connect: Connected to Meiner␛[0m
␛[0;32mI (3079) example_connect: IPv4 address: 192.168.43.23␛[0m
␛[0;31mE (3089) example: setsockopt 0  0 Success ␛[0m
␛[0;32mI (3089) example: Socket created␛[0m
␛[0;32mI (3099) example: Socket bound, port 60583␛[0m
␛[0;32mI (3099) example: Socket listening␛[0m
␛[0;32mI (11099) example: Socket accepted␛[0m
␛[0;32mI (11129) example: Received 86 bytes from 192.168.43.62:␛[0m
␛[0;32mI (11129) example: {"event":"mod_control","id":1336146744,"h_id":"25289830","is_old":false,"state":false}␛[0m
␛[0;32mI (13349) example: Connection closed␛[0m
␛[0;31mE (13349) example: Shutting down socket and restarting...␛[0m
␛[0;31mE (13349) example: setsockopt 0  128 Socket is not connected ␛[0m
␛[0;32mI (13349) example: Socket created␛[0m
␛[0;32mI (13359) example: Socket bound, port 60583␛[0m
␛[0;31mE (13359) example: Error occurred during listen: errno 112␛[0m
```
first socket connection let me set setsockopt.
received the data like i want it to

then after restarting i get 128 -> Enotconn ... how is the socket not connected anymore? 
As i understood it the example just shutdowns the previous socket and then starts a new one.

The error from listen is not listet in
https://pubs.opengroup.org/onlinepubs/007908799/xns/listen.html
unless i missread

### Steps to reproduce

1. step1
2. ...

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

Add the {...} into the example and that's my code.
Additionally i have connect.c in, so the example connect works with my WiFi.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [x] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/4954248/sdkconfig.txt)

ZweiEuro, 2020-07-21:
For reproduction I tried https://play.google.com/store/apps/details?id=com.simplesockettester&hl=en_US
and it produces the exact same results, connecting to the socket twice  makes it die. ( As written above with the setsockopt)

Alvin1Zhang, 2020-07-22:
Thanks for the detailed report, we will look into.

david-cermak, 2020-08-24:
Hi @ZweiEuro 

The `SO_REUSEADDR` works as expected. 
The issue with the tcp-server example is that it does close the socket [here](https://github.com/espressif/esp-idf/blob/48ea44f3d1e4b5763a2d0b65bc5c56a84690e910/examples/protocols/sockets/tcp_server/main/tcp_server.c#L125), but this is the  *accepted* socket, the original *listener* socket is still open and bound to the original address/port (So in fact we could bind to it -- `SO_REUSEADDR=1` allows us to bind to already used endpoint -- but the actual listen would fail as the endpoint is actively used by an open socket) 

Adding 
```
            shutdown(listen_sock, 0);
            close(listen_sock);
```
after the
``` c
            ESP_LOGE(TAG, "Shutting down socket and restarting...");
```
Should solve the issue.

In general, please check the latest version of the example here
https://github.com/espressif/esp-idf/blob/master/examples/protocols/sockets/tcp_server/main/tcp_server.c
where the listener socket is kept open, which is I believe a preferred option for common a tcp server.

david-cermak, 2020-09-09:
@ZweiEuro Any further observation from your side or can we close the issue?

Alvin1Zhang, 2020-09-17:
Thanks for reporting, will close due to short of feedback, feel free to reopen if the issue still happens. Thanks.

closed 2020-09-17 by Alvin1Zhang