espressif/esp-idf #5623: Sources of secure_cert_mfg.bin used in esp-cryptoauth utility

mvacha, 2020-07-21:
Are the sources of secure_cert_mfg.bin and secure_cert_mfg_esp32.bin available somewhere? I'm having trouble communicating with ATECC608 from ESP32, but I had no problem running the esp-cryptoauth utility made by @AdityaHPatwardhan, so I wanted to see where is the difference, but I cannot find the source code anywhere.

AdityaHPatwardhan, 2020-07-22:
@mvacha right, the source code has been kept in an internal repo, If I may ask what problems are you facing with connecting ATECC608 to ESP32,  maybe you can share some code and we will help fix it.

mvacha, 2020-07-22:
I wanted to run the atecc608_ecdsa example from this repo on HW I have (WROOM32 devkit) with 
ATECC608 on a ATCRYPTOAUTH-XPRO-B devboard connected over I2C. I couldn't found a devkit with ESP32-WROOM32-SE, so I cannot use the exactly same HW as the demo recommends. I had no trouble running the cryptoauth utility you made or this demo based on Platformio and ESP-IDF 3.3 (https://github.com/kmwebnet/ECC608-Configure), so I assumed that it should work.

In the demo, I have changed the atcai2 bus to 1 (same as I have used with the platformio demo):
```
cfg_ateccx08a_i2c_default.atcai2c.bus = 1;
ret = atcab_init(&cfg_ateccx08a_i2c_default);
```
But the return value from `atcab_init` is always `ATCA_COMM_FAIL (0xF0)` or `ATCA_GEN_FAIL (0xE1)`.
I have also tried do explicitly configure the I2C port (again same values that worked for me in with the platformio demo) as I wasn't sure if the HAL does it implictly or not:
```
i2c_config_t i2c_config = {
    .mode = I2C_MODE_MASTER,
    .sda_io_num = 21,
    .scl_io_num = 22,
    .sda_pullup_en = GPIO_PULLUP_DISABLE,
    .scl_pullup_en = GPIO_PULLUP_DISABLE,
    .master.clk_speed = 100000
    };
        
i2c_param_config(I2C_NUM_0 , &i2c_config);
i2c_driver_install(I2C_NUM_0 , I2C_MODE_MASTER, 0, 0, 0);
```
But calling `atcab_init` keeps failing. I have also tried other demos [[1]](https://github.com/anozaki/esp32-atecc608a-iot/blob/master/main/main.c),[[2]](https://github.com/PBearson/eccx08-with-esp32) doing the same (ESP32 + ATECC608A over I2C), but I wasn't able to find another one that would work for me.

My "hunch" is that the issue is related to incompatible library/esp-idf versions, but I wasn't able to figure it.

I'm using the latest (master) version of ESP-IDF with the latest tooling. 

Any help or pointing in the right direction would be greatly appriciated, thanks.

mvacha, 2020-07-22:
I have just succesfully run https://github.com/PBearson/eccx08-with-esp32 with `SDA = 4` and `SCL = 15` (defaults in the project). But still no luck with the example from this repo.

AdityaHPatwardhan, 2020-07-23:
At what pins of ESP32 have you connected the ATECC608 chip ?, ( generally GPIO 21 (SDA) and GPIO22 (SCL) are used for that purpose) 
If you were able to run the `secure_cert_mfg.py` script on you chip,as It is internally set as `SDA = 21`, `SCL = 22`, I wonder how you were able to again connect to `atecc608` with providing `SDA=4` and `SCL=15`.

Meanwhile, you may change [this_line](https://gitlab.espressif.cn:6688/espressif/esp-cryptoauthlib/-/blob/master/cryptoauthlib/lib/hal/hal_esp32_i2c.c#L23) with your value for SDA pin, respectively for SCL, I think with that you might be able to connect to ATECC608 from ESP32,

I am assuming in the XPRO-B board, `WROOM32 devkit` has been directly connected with `atecc608` over I2C.

mvacha, 2020-07-23:
I didn't have to change anything in the linked demo (https://github.com/PBearson/eccx08-with-esp32) it just worked. 

Also changing the pins to `SDA = 21`, `SCL = 22` in `hal_esp32_i2c.c` fixed the demo for me, thanks :)

AdityaHPatwardhan, 2020-07-24:
@mvacha if your issue is now resolved, can you please close the issue. Thank you.

closed 2020-07-24 by mvacha

debug-ito, 2021-10-22:
@AdityaHPatwardhan Sorry for commenting on a close issue, but is there any plan to publish the source code of secure_cert_mfg.bin ? I'm curious of the specification of its serial-line interface. It looks so useful for manipulating ATECC608 from a host PC.