espressif/esp-idf #5624: GPIO wakeup from light sleep does not work for GPIO16/17 when timer and GPIO wakeup are enabled

jimparis, 2020-07-21:
Hi,

I couldn't get wakeup from light sleep to work reliably for GPIO16, when both GPIO and timer wakeups are enabled (as is the case for automatic light sleep).  Eventually I discovered that this is because `esp_light_sleep_start` disables `VDD_SDIO` if the sleep duration is long enough, and GPIO16/GPIO17 are in the `VDD_SDIO` power domain.

I think that `esp_light_sleep_start` should leave `VDD_SDIO` enabled if any of the GPIOs in that power domain are configured as wakeup sources.  It would also be nice to document this more clearly, perhaps [here](https://docs.espressif.com/projects/esp-idf/en/v4.0.1/api-reference/system/sleep_modes.html#gpio-wakeup-light-sleep-only), as I would have avoided using those pins in my design if I realized this earlier.

lennyJL, 2022-04-01:
Hi Jim. If you use GPIOs powered by VDD_SDIO during light sleep(such as `hold gpio` or `use gpio as wakeup source`), you need to keep VDD_SDIO ON by calling `esp_sleep_pd_config(ESP_PD_DOMAIN_VDDSDIO, ESP_PD_OPTION_ON);`.

You can find related document: `docs/en/api-reference/system/sleep_modes.rst`(Lines 137 to 143).

Alvin1Zhang, 2022-04-01:
Thanks for reporting, feel free to reopen.

closed 2022-04-01 by Alvin1Zhang

jimparis, 2022-04-01:
Well, yeah, that's what I said, and it was not in the documentation at the time -- thanks for adding it.