espressif/esp-idf #5625: expose FF_USE_FASTSEEK in menuconfig

yetifrisstlama, 2020-07-21:
Without `FF_USE_FASTSEEK`, seeking backwards or forwards on a FAT32 FS takes __hundreds of ms__.

This shows up, for example, when streaming animations from a SD card and frames need to be shown in a non-sequential way.

Enabling `FF_USE_FASTSEEK` costs ~64 words of memory per open file and reduces seek time by a factor of ~100.

I tested this by overriding the `ff.c` file with my patched version:
https://github.com/yetifrisstlama/espirgbani_pio/commit/2c957e3f77fba3410e91f84a3a8304582586eda8

See also:
http://www.elm-chan.org/fsw/ff/doc/lseek.html

I've opened the same issue for arduino-esp32:
https://github.com/espressif/arduino-esp32/issues/4097

Here is a benchmark project:
https://github.com/yetifrisstlama/fast_seek_test

Needs platform.io and an ESP32 with SD card connected.

```
------------------------
 without FAST_SEEK
------------------------
[I][main.cpp:34] setup(): Creating 200 MB file: /sd/test.dat ...
[I][main.cpp:58] setup(): fseek( 104857750) dt: 128002 us, read 4096
[I][main.cpp:58] setup(): fseek( 104857751) dt: 128022 us, read 4096
[I][main.cpp:58] setup(): fseek( 104857752) dt: 128044 us, read 4096
[I][main.cpp:58] setup(): fseek( 104857753) dt: 128061 us, read 4096
[I][main.cpp:58] setup(): fseek( 104857754) dt: 128012 us, read 4096
[I][main.cpp:58] setup(): fseek( 104857755) dt: 127987 us, read 4096
[I][main.cpp:58] setup(): fseek( 104857756) dt: 128001 us, read 4096

------------------------
 with FAST_SEEK enabled
------------------------
(comment last line in platform.ini)
[I][main.cpp:34] setup(): Creating 200 MB file: /sd/test.dat ...
[V][ff.c:3736] f_open(): FAST_SEEK enabled
[I][main.cpp:58] setup(): fseek( 104857750) dt:   1143 us, read 4096
[I][main.cpp:58] setup(): fseek( 104857751) dt:    692 us, read 4096
[I][main.cpp:58] setup(): fseek( 104857752) dt:    684 us, read 4096
[I][main.cpp:58] setup(): fseek( 104857753) dt:    681 us, read 4096
[I][main.cpp:58] setup(): fseek( 104857754) dt:    679 us, read 4096
[I][main.cpp:58] setup(): fseek( 104857755) dt:    682 us, read 4096
[I][main.cpp:58] setup(): fseek( 104857756) dt:    681 us, read 4096
```

uLipe, 2020-08-19:
Hey @yetifrisstlama , thanks for your contribution. We're working internally to enable this option. Our plan is to make the fast seek capabilities to be enabled using the ESP-VFS API. 

The idea is to allow the application to decide when to use fast-seek or not given the limitations of this mechanism as stated in ELM-Chan documentation.

This will take some more time, but we will update the status of this feature once it is getting ready.

yetifrisstlama, 2020-08-19:
That's fantastic news! Looking forward to it :+1:

Alvin1Zhang, 2021-08-24:
Thanks for reporting, fix is available at https://github.com/espressif/esp-idf/commit/6ce34a227e10108b5c457db2195dcd8346d3da38, feel free to reopen.

closed 2021-08-24 by Alvin1Zhang