espressif/esp-idf #5627: Guru Meditation Error: Core  1 panic'ed (LoadProhibited) when test wpa2_enterprise example code

AxelLin, 2020-07-22:
## Environment

- Module or chip used: ESP32-WROOM-32
- IDF version: v4.3-dev-472-gcf056a7d0b90
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0
- Operating System: Linux
- Using an IDE?: NO
- Power Supply: USB

## Problem Description

Guru Meditation Error: Core  1 panic'ed (LoadProhibited) when test wpa2_enterprise example code

Environment
Module or chip used: ESP32-WROOM-32
IDF version: v4.3-dev-472-gcf056a7d0b90
Build System: idf.py
Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r1) 8.2.0
Operating System: Linux
Using an IDE?: NO
Power Supply: USB

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7476
load:0x40078000,len:13988
ho 0 tail 12 room 4
load:0x40080400,len:4000
0x40080400: _init at ??:?

entry 0x40080684
I (30) boot: ESP-IDF v4.3-dev-472-gcf056a7d0b90 2nd stage bootloader
I (30) boot: compile time 14:32:25
I (30) boot: chip revision: 1
I (34) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot.esp32: SPI Speed      : 40MHz
I (46) boot.esp32: SPI Mode       : DIO
I (51) boot.esp32: SPI Flash Size : 2MB
I (55) boot: Enabling RNG early entropy source...
I (61) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (79) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 00100000
I (94) boot: End of partition table
I (98) boot_comm: chip revision: 1, min. application chip revision: 0
I (105) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x1e86c (125036) map
I (163) esp_image: segment 1: paddr=0x0002e894 vaddr=0x3ffb0000 size=0x01784 (  6020) load
I (166) esp_image: segment 2: paddr=0x00030020 vaddr=0x400d0020 size=0x8a214 (565780) map
0x400d0020: _stext at ??:?

I (392) esp_image: segment 3: paddr=0x000ba23c vaddr=0x3ffb1784 size=0x023a4 (  9124) load
I (396) esp_image: segment 4: paddr=0x000bc5e8 vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at /home/axel/esp/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730

I (399) esp_image: segment 5: paddr=0x000bc9f4 vaddr=0x40080404 size=0x15ce4 ( 89316) load
I (460) boot: Loaded app from partition at offset 0x10000
I (460) boot: Checking flash encryption...
I (460) flash_encrypt: flash encryption is enabled (0 plaintext flashes left)
I (467) boot: Disabling RNG early entropy source...
I (484) cpu_start: Pro cpu up.
I (484) cpu_start: Starting app cpu, entry point is 0x40081430
0x40081430: call_start_cpu1 at /home/axel/esp/esp-idf/components/esp_system/port/cpu_start.c:109

I (0) cpu_start: App cpu up.
I (500) cpu_start: Pro cpu start user code
I (500) cpu_start: Application information:
I (500) cpu_start: Project name:     wpa2-enterprise
I (506) cpu_start: App version:      v4.3-dev-472-gcf056a7d0b90
I (513) cpu_start: Compile time:     Jul 22 2020 14:32:09
I (519) cpu_start: ELF file SHA256:  8c3bdf8a5337e530...
I (525) cpu_start: ESP-IDF:          v4.3-dev-472-gcf056a7d0b90
I (532) heap_init: Initializing. RAM available for dynamic allocation:
I (539) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (545) heap_init: At 3FFB9658 len 000269A8 (154 KiB): DRAM
I (551) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (557) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (564) heap_init: At 400960E8 len 00009F18 (39 KiB): IRAM
W (571) flash_encrypt: Flash encryption mode is DEVELOPMENT (not secure)
I (578) spi_flash: detected chip: generic
I (582) spi_flash: flash io: dio
W (586) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (600) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (665) wifi:wifi driver task: 3ffc1ce0, prio:23, stack:6656, core=0
I (665) system_api: Base MAC address is not set
I (665) system_api: read default base MAC address from EFUSE
I (715) wifi:wifi firmware version: bbe4b67
I (715) wifi:wifi certification version: v7.0
I (715) wifi:config NVS flash: enabled
I (715) wifi:config nano formating: disabled
I (725) wifi:Init data frame dynamic rx buffer num: 32
I (725) wifi:Init management frame dynamic rx buffer num: 32
I (735) wifi:Init management short buffer num: 32
I (735) wifi:Init dynamic tx buffer num: 32
I (735) wifi:Init static rx buffer size: 1600
I (745) wifi:Init static rx buffer num: 10
I (745) wifi:Init dynamic rx buffer num: 32
I (755) example: Setting WiFi configuration SSID LAB-WPA2-ENT...
I (755) wpa: WPA2 ENTERPRISE VERSION: [v2.0] enable

W (765) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration
I (905) phy: phy_version: 4182, f1ba940, Jun  4 2020, 19:40:07, 0, 2
I (915) wifi:mode : sta (f0:08:d1:79:8c:28)
I (1895) wifi:new:<8,2>, old:<1,0>, ap:<255,255>, sta:<8,2>, prof:1
I (2415) wifi:state: init -> auth (b0)
I (2425) wifi:state: auth -> assoc (0)
I (2435) wifi:state: assoc -> run (10)
I (2435) wpa: wpa2_task prio:2, stack:6656

Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4015877d  PS      : 0x00060c30  A0      : 0x800e02e4  A1      : 0x3ffc9780  
0x4015877d: mbedtls_pk_get_bitlen at /home/axel/esp/esp-idf/components/mbedtls/mbedtls/library/pk.c:502 (discriminator 1)

A2      : 0x000000bc  A3      : 0x3ffb1404  A4      : 0x3ffb57f4  A5      : 0x3ffbb064  
A6      : 0x3ffba700  A7      : 0xffff0000  A8      : 0x00000000  A9      : 0x00000001  
A10     : 0x3f41294c  A11     : 0x00000000  A12     : 0x00000028  A13     : 0x3ffba6a2  
A14     : 0x00002019  A15     : 0x00000000  SAR     : 0x0000001f  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x000000bc  LBEG    : 0x4000c428  LEND    : 0x4000c42d  LCOUNT  : 0x00000000  

Backtrace:0x4015877a:0x3ffc9780 0x400e02e1:0x3ffc97a0 0x400e03be:0x3ffc97c0 0x400e041d:0x3ffc97e0 0x400e0809:0x3ffc9800 0x400e4e1c:0x3ffc9820 0x400e4ee8:0x3ffc9840 0x400e3c94:0x3ffc98c0 0x400dd465:0x3ffc98e0 0x400dd5f7:0x3ffc9910 0x400dd743:0x3ffc9930 0x40087655:0x3ffc9970
0x4015877a: mbedtls_pk_get_bitlen at /home/axel/esp/esp-idf/components/mbedtls/mbedtls/library/pk.c:499

0x400e02e1: tls_set_ciphersuite at /home/axel/esp/esp-idf/components/wpa_supplicant/src/crypto/tls_mbedtls.c:387 (discriminator 1)

0x400e03be: set_client_config at /home/axel/esp/esp-idf/components/wpa_supplicant/src/crypto/tls_mbedtls.c:431

0x400e041d: tls_create_mbedtls_handle at /home/axel/esp/esp-idf/components/wpa_supplicant/src/crypto/tls_mbedtls.c:449

0x400e0809: tls_connection_set_params at /home/axel/esp/esp-idf/components/wpa_supplicant/src/crypto/tls_mbedtls.c:764

0x400e4e1c: eap_tls_init_connection at /home/axel/esp/esp-idf/components/wpa_supplicant/src/eap_peer/eap_tls_common.c:144

0x400e4ee8: eap_peer_tls_ssl_init at /home/axel/esp/esp-idf/components/wpa_supplicant/src/eap_peer/eap_tls_common.c:202

0x400e3c94: eap_peap_init at /home/axel/esp/esp-idf/components/wpa_supplicant/src/eap_peer/eap_peap.c:165

0x400dd465: eap_sm_process_request at /home/axel/esp/esp-idf/components/wpa_supplicant/src/esp_supplicant/esp_wpa2.c:457

0x400dd5f7: eap_sm_rx_eapol_internal at /home/axel/esp/esp-idf/components/wpa_supplicant/src/esp_supplicant/esp_wpa2.c:645

0x400dd743: wpa2_task at /home/axel/esp/esp-idf/components/wpa_supplicant/src/esp_supplicant/esp_wpa2.c:217

0x40087655: vPortTaskWrapper at /home/axel/esp/esp-idf/components/freertos/xtensa/port.c:169

ELF file SHA256: 8c3bdf8a5337e530

Rebooting...
ets Jun  8 2016 00:22:57

### Expected Behavior
It should work.
### Actual Behavior
Hit panic and keep rebooting.

### Steps to reproduce

Just run the wpa2_enterprise example code.

Attached my config.
(I have flash encryption on my testing device, you can disable flash encryption if necessary. (It's not related to this problem))

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/4958277/sdkconfig.txt)

AxelLin, 2020-07-22:
The panic only happen when CONFIG_WPA_MBEDTLS_CRYPTO=y (which is the default setting now).
But the example code still fails to join AP with enterprise security when CONFIG_WPA_MBEDTLS_CRYPTO is not set.

Alvin1Zhang, 2020-07-22:
Thanks for reporting, we already had a fix internally under reviewing, will update when the fix is available on GitHub. Thanks.

sagb2015, 2020-07-22:
@AxelLin Thanks for reporting the issue. 
This seems to be a regression from the recent TLS porting using mbedlts for the case - CONFIG_WPA_MBEDTLS_CRYPTO=y and CONFIG_EXAMPLE_VALIDATE_SERVER_CERT=n.  We already have fixed this during out testing and fix should be available for this soon. 
However, for the case CONFIG_WPA_MBEDTLS_CRYPTO=n. There should not be any issue. Can you share the logs for the same please?

AxelLin, 2020-07-22:
@sagb2015 

For the case CONFIG_WPA_MBEDTLS_CRYPTO=n, the log is the same as I reported on https://github.com/espressif/esp-idf/issues/5273

(9) eap: Peer sent packet with method EAP PEAP (25)
(9) eap: Calling submodule eap_peap to process data
(9) eap_peap: Continuing EAP-TLS
(9) eap_peap: Peer indicated complete TLS record size will be 60 bytes
(9) eap_peap: Got complete TLS record (60 bytes)
(9) eap_peap: [eaptls verify] = length included
(9) eap_peap: (other): before SSL initialization
(9) eap_peap: TLS_accept: before SSL initialization
(9) eap_peap: TLS_accept: before SSL initialization
(9) eap_peap: <<< recv TLS 1.3  [length 0037]
(9) eap_peap: >>> send TLS 1.2  [length 0002]
(9) eap_peap: ERROR: TLS Alert write:fatal:internal error
tls: TLS_accept: Error in error
(9) eap_peap: ERROR: Failed in __FUNCTION__ (SSL_read): error:14201044:SSL routines:tls_choose_sigalg:internal error
(9) eap_peap: ERROR: System call (I/O) error (-1)
(9) eap_peap: ERROR: TLS receive handshake failed during operation
(9) eap_peap: ERROR: [eaptls process] = fail
(9) eap: ERROR: Failed continuing EAP PEAP (25) session.  EAP sub-module failed
(9) eap: Sending EAP Failure (code 4) ID 108 length 4

I tried all your suggestions, but it still does not work.
So I'm wondering if it works for you with FreeRADIUS v3.0.21 now? (I it works what settings did you change?)
(Since you mentioned that you can reproduce the same issue:
we tested with FreeRadius 3.0.12 and 3.0.21 versions and results were different (3.0.12 => no issues and 3.0.21=> same issue as you reported).

AxelLin, 2020-08-03:
It's fixed in v4.3-dev-771-gc77c4ccf6c43, thanks.

closed 2020-08-03 by AxelLin