espressif/esp-idf #5629: SPIFFS conflicts with WIFI when using SPIRAM

asheetov, 2020-07-22:
Environment
---
Module or chip used: ESP32-WROOVER-32
IDF version: v3.3.2
Build System: makefile
Compiler version: xtensa-esp32-elf-gcc (crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
Operating System: Linux
Power Supply: USB
Problem Description
Guru Meditation Error: Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed) when running example code

Core 0 register dump:
PC      : 0x4008a404  PS      : 0x00060c34  A0      : 0x80084826  A1      : 0x3ffb7450  
0x4008a404: esp_rom_spiflash_read_data at /home/san/termt/eftlite/esp-idf/components/spi_flash/spi_flash_rom_patch.c:234
 (inlined by) esp_rom_spiflash_read at /home/san/termt/eftlite/esp-idf/components/spi_flash/spi_flash_rom_patch.c:587

A2      : 0x00040000  A3      : 0x3ffb7470  A4      : 0x00000004  A5      : 0x0000003f  
A6      : 0x00040000  A7      : 0x1f50fc00  A8      : 0x3ff42000  A9      : 0x3ffb7420  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x3ff42004  A13     : 0x000001ff  
A14     : 0x00040000  A15     : 0x3ff42000  SAR     : 0x00000010  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x40093724  LEND    : 0x40093752  LCOUNT  : 0xffffffff  
0x40093724: memcpy at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/memcpy.S:168

0x40093752: memcpy at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/memcpy.S:201

ELF file SHA256: ad0bd0baad0bd0baad0bd0baad0bd0baad0bd0baad0bd0baad0bd0baad0bd0ba

Backtrace: 0x4008a404:0x3ffb7450 0x40084823:0x3ffb7470 0x4012cc6d:0x3ffb74d0 0x400e3b71:0x3ffb7500 0x400e3d1d:0x3ffb7530 0x400e5201:0x3ffb7560 0x400e31a0:0x3ffb75a0 0x400e2743:0x3ffb75d0 0x400e2d71:0x3ffb7620 0x400d2938:0x3ffb76e0 0x400d298a:0x3ffb7710 0x400d0c54:0x3ffb7730
0x4008a404: esp_rom_spiflash_read_data at /home/san/termt/eftlite/esp-idf/components/spi_flash/spi_flash_rom_patch.c:234
 (inlined by) esp_rom_spiflash_read at /home/san/termt/eftlite/esp-idf/components/spi_flash/spi_flash_rom_patch.c:587

0x40084823: spi_flash_read at /home/san/termt/eftlite/esp-idf/components/spi_flash/flash_ops.c:520

0x4012cc6d: esp_partition_read at /home/san/termt/eftlite/esp-idf/components/spi_flash/partition.c:242

0x400e3b71: spiffs_api_read at /home/san/termt/eftlite/esp-idf/components/spiffs/spiffs_api.c:36

0x400e3d1d: spiffs_phys_rd at /home/san/termt/eftlite/esp-idf/components/spiffs/spiffs/src/spiffs_cache.c:146

0x400e5201: spiffs_obj_lu_scan at /home/san/termt/eftlite/esp-idf/components/spiffs/spiffs/src/spiffs_nucleus.c:1164

0x400e31a0: SPIFFS_mount at /home/san/termt/eftlite/esp-idf/components/spiffs/spiffs/src/spiffs_hydrogen.c:997

0x400e2743: esp_spiffs_init at /home/san/termt/eftlite/esp-idf/components/spiffs/esp_spiffs.c:327

0x400e2d71: esp_vfs_spiffs_register at /home/san/termt/eftlite/esp-idf/components/spiffs/esp_spiffs.c:359

0x400d2938: spiffs_test() at /home/san/termt/eftlite/tests/main/appmain.cpp:65

0x400d298a: app_main at /home/san/termt/eftlite/tests/main/appmain.cpp:81

0x400d0c54: main_task at /home/san/termt/eftlite/esp-idf/components/esp32/cpu_start.c:551

Rebooting...

Expected Behavior
---
It should work.

Actual Behavior
---
Hit panic and keep rebooting.

Steps to reproduce
---
1. Run the example code to see the panic. 
2. Comment out wifi_start() call to make the example work.

Attached my config, partition table and example code
I have flash encryption and secure boot on my testing device, you can disable it if necessary. (It's not related to this problem).

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/4959644/sdkconfig.txt)
[appmain.cpp.txt](https://github.com/espressif/esp-idf/files/4959648/appmain.cpp.txt)
[partitions.csv.txt](https://github.com/espressif/esp-idf/files/4959646/partitions.csv.txt)

chegewara, 2020-07-22:
https://github.com/espressif/esp-idf/issues/5107

igrr, 2020-09-09:
@asheetov Sorry that we didn't reply to this issue earlier. As a workaround, you may apply the following patch:
```diff
diff --git a/components/esp32/esp_adapter.c b/components/esp32/esp_adapter.c
index 915666b3db..28c7862dea 100644
--- a/components/esp32/esp_adapter.c
+++ b/components/esp32/esp_adapter.c
@@ -177,7 +177,7 @@ static void set_isr_wrapper(int32_t n, void *f, void *arg)
 static void * spin_lock_create_wrapper(void)
 {
     portMUX_TYPE tmp = portMUX_INITIALIZER_UNLOCKED;
-    void *mux = malloc(sizeof(portMUX_TYPE));
+    void *mux = heap_caps_malloc(sizeof(portMUX_TYPE), MALLOC_CAP_8BIT|MALLOC_CAP_INTERNAL);
 
     if (mux) {
         memcpy(mux,&tmp,sizeof(portMUX_TYPE));
```

This issue will be updated once the fix is merged.

liuzfesp, 2020-09-14:
Hi @asheetov, can @igrr 's suggestion solve your issue?

closed 2020-10-14 by espressif-bot