espressif/esp-idf #5630: Reading Bluetooth Classic RSSI value unreliable

chedda61, 2020-07-22:
## Environment

- Development Kit:         DOIT ESP32 Dev Kit
- Kit version:                   v1
- Module or chip used:  ESP32-WROOM-32
- IDF version:                  v4.0.1
- Build System:               CMake
- Compiler version:       8.2.0
- Operating System:     macOS
- Using an IDE?:            PlatformIO
- Power Supply:             USB

## Problem Description

Trying to read RSSI value of a non-paired, non-discoverable device sometimes returns 0, although it is too far away.
Depending on the delay between sending the hci commands, the ratio of reasonable values and 0 can change.
With vTaskDelay(1000 / portTICK_PERIOD_MS) there is a ~50% change to get value !=0.
Waiting longer will disconnect from the device, waiting shorter will decrease the chance of getting a value !=0.
Using linux hcitool, the rssi value can be obtained about once per second with the same remote device.
With the esp32 and and vTaskDelay(1000 / portTICK_PERIOD_MS) it takes about 6-7 seconds for one reading.

### Expected Behavior

True RSSI value each time, or a status code != 0 in case of a problem.
Ideally the wait time between commands should be much smaller.

### Actual Behavior

Returned value is sometimes 0.

### Steps to reproduce

1. Setup bt (esp_bt_controller_init, esp_bt_controller_enable)
2. loop hci commands in this order (and wait for ok status in callback):
 hci reset
 hci create connection
 hci read remote features
 hci read remote extended features
 hci read remote name
 hci read rssi
 hci disconnect
  

### Code to reproduce this issue

https://gist.github.com/chedda61/188607f4f061bb00052097d4915679e2

## Other items if possible

- [x] sdkconfig file (https://github.com/espressif/esp-idf/files/4959869/sdkconfig.zip)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

Alvin1Zhang, 2020-07-23:
Thanks for reporting, we will look into.

munho-espressif, 2020-09-07:
Hi @chedda61 ,

You can look at the Core specification 4.2, a rssi value of zero is within the normal range.
[image]

chedda61, 2020-09-07:
Hi,

sorry I didn't describe the problem clear enough. Let me explain it with two scenarios:
Scenario 1:
RSSI print of the esp32 scanning a phone that is 10cm away:
-3
-5
0
-1
-4

This is reasonable, I have no problem with that. Now let's look at 
Scenario 2:
RSSI print of the esp32 scanning a phone that is **5m** away:
-23
-22
0
-19
0
-22
0
0
-20

In this scenario "0" can't be true, which is why I think there is a bug.
I tried the same scenario and procedure on a Raspberry Pi 3B+ and Raspberry Pi Zero W. There i don't get 0s in between lower values. It will be constantly -25 to -18.

I would rather have it print an error than 0, as 0 is indicating very close proximity, as described in the specification above.

munho-espressif, 2020-12-01:
Hi @chedda61 ,

I didn't detect the problem with the latest branch Release4.0. Maybe you should try the latest branch Release4.0.

Thanks.

closed 2021-01-28 by Alvin1Zhang