espressif/esp-idf #5631: decode topic MQTT for multi-sensor

SERIDJ, 2020-07-22:
hi  ,

iam using application with multi-sensor so i need to subscribe to may topic ,
usign ** mqtt_event_handler(esp_mqtt_event_handle_t event) ** https://github.com/espressif/espidf/blo ... app_main.c

the examples works fine, but to be honest I have not understood the concept how to use it right.

for subscribe iam using case MQTT_EVENT_DATA i dont know if there another method ?

so i want to decode topic but i have problem : 

[image]

david-cermak, 2020-07-23:
Hi @SERIDJ 

Yes, the only way to receive published data is to register to `MQTT_DATA_EVENT`. Please check the documentation on this event https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/mqtt.html#events, the general idea is to check the actual and total data length. you can check an internal test code in idf how the longer data are reconstructed: https://github.com/espressif/esp-idf/blob/master/tools/test_apps/protocols/mqtt/publish_connect_test/main/publish_test.c#L74. Please note that the topic and message id is valid only for the first data event.

laukik-hase, 2020-07-27:
You can use a FreeRTOS queue for handling MQTT data.
```c

// Set the queue size and stack size according to your data-length
# define MQTT_QUEUE_SIZE 8
# define MQTT_DATA_TASK_SIZE 2048

static const char * TAG_MQTT_UTILS = "MQTT_UTILS"

// MQTT Data struct
typedef struct
{
    char * topic
    char * payload
} mqtt_event_data_t;

// MQTT Data queue handle
static QueueHandle_t mqtt_data_queue;

esp_err_t mqttHandleDataReceived(const char * topic, const char * data)
{
    // Process your data....
}

void mqttEventDataTask(void * arg)
{
    mqtt_event_data_t recv_data = {0}

    while (true)
    {
        if (xQueueReceive(mqtt_data_queue, & recv_data, (portTickType)portMAX_DELAY))
        {
            mqttHandleDataReceived(recv_data.topic, recv_data.payload);
        }
    }
}

esp_err_t mqttCallback(esp_mqtt_event_handle_t event)
{
    ... 

    mqtt_event_data_t send_data = {0};

    switch(event -> event_id)
    {
        ...

        case MQTT_EVENT_DATA:
            ESP_LOGI(TAG_MQTT_UTILS,  "MQTT_EVENT_DATA");
            ESP_LOGI(TAG_MQTT_UTILS, "Topic: %.*s, Data length: %d", event -> topic_len, event -> topic, event -> data_len);

            // Loading Queue
            send_data.topic = event -> topic;
            send_data.payload = event -> data;

            ESP_LOGI(TAG_MQTT_UTILS,  "MQTT Data sending through queue");
            xQueueSend(mqtt_data_queue, & send_data, (portTickType)portMAX_DELAY);

            break;

        ...
    }
    ...
}

void initMQTT(void)
{
    // ESP-IDF MQTT Initialisation functions...

    mqtt_data_queue = xQueueCreate(MQTT_QUEUE_SIZE, sizeof(mqtt_event_data_t));
    xTaskCreate(mqttEventDataTask, "mqtt_event_data_task", MQTT_DATA_TASK_SIZE, NULL, 6, NULL);
}
```

SERIDJ, 2020-07-29:
@laukik-hase  yes that work well thanks , please do u know what is min time between 2 publish with mqtt  (esp_mqtt_client_publish) 
bcs i have until 32 value and if i use vTaskDelay 1s so that take 32s to send all value , and i dont know if mqtt library is Thread safety too ?

laukik-hase, 2020-07-30:
@SERIDJ About the thread safety, functions like the esp_mqtt_subscribe(...) and esp_mqtt_publish(...) are thread-safe as given in [ESP-IDF MQTT](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/mqtt.html) docs.
 
Could be more clear about the first question?

SERIDJ, 2020-07-30:
@laukik-hase yes i mean what is minimum time between 2 publish data to broker mqtt like  

'''    		
while (1) {

esp_mqtt_client_publish(client,topic,data1,strlen(data1), 0, 0);
vTaskDelay(X/ portTICK_PERIOD_MS);

esp_mqtt_client_publish(client,topic,data2,strlen(data2), 0, 0);
vTaskDelay(X/ portTICK_PERIOD_MS);

.
.
.
'

esp_mqtt_client_publish(client,topic,data32,strlen(data32), 0, 0);
vTaskDelay(X/ portTICK_PERIOD_MS);
''

}

laukik-hase, 2020-07-30:
@SERIDJ  I don't think there is any minimum interval between two esp_mqtt_publish(...) calls. You can test this in a ```while(true)``` loop if you want. Just take care that you are not subscribed to the topic you are publishing to.

SERIDJ, 2020-07-30:
@laukik-hase  yes, and that not saturated mqtt protocol if i send data without delay like : 

while(1) {

esp_mqtt_client_publish(client,topic,data1,strlen(data1), 0, 0);
esp_mqtt_client_publish(client,topic,data1,strlen(data2), 0, 0);
esp_mqtt_client_publish(client,topic,data1,strlen(data3), 0, 0);
.
.
.
esp_mqtt_client_publish(client,topic,dataXstrlen(dataX), 0, 0);

}

there is some execution time for fonction but for me  I **thought** that creates some issue with time .

laukik-hase, 2020-07-30:
Good. You can close the issue if you have no further queries.

closed 2020-07-31 by SERIDJ