espressif/esp-idf #5632: esp_wifi_sta_get_ap_info doesn't update RSSI value

dyarkovoy, 2020-07-22:
## Environment
- Development Kit:      [none]
- Module or chip used:  [ESP32-WROOM-32D]
- IDF version (run ``git describe --tags`` to find it): 3.40001.200521 (4.0.1)
- Build System:         [CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 2.80200.200226 (8.2.0)
- Operating System:     [Windows]
- (Windows only) environment type: PlatformIO
- Using an IDE?: [Yes] PlatformIO
- Power Supply:         [USB]

## Problem Description

`wifi_ap_record_t.rssi` value obtained by calling `esp_wifi_sta_get_ap_info()` is valid only for the moment ESP32 associated with the AP, and doesn't update when signal strength changes (ESP32 moves away/closer to the AP). If this is intended behavior, docs don't seem to be mentioning it. ESP8266 behaves differently and returns up-to-date value of RSSI.

liuzfesp, 2020-07-23:
Hi @dyarkovoy, is the AP hidden AP?

dyarkovoy, 2020-07-23:
Hi, @liuzfesp , you are right, the AP is hidden and once I un-hide it, I get real-time RSSI readings. 
So is it expected behavior for hidden APs? 
Is there any other way to get real-time RSSI for hidden AP?

esp-HarveyRong, 2020-08-11:
Hi @dyarkovoy, This is  bug, we will fix it soon, thanks for your feedback.

Alvin1Zhang, 2020-08-13:
@dyarkovoy Thanks for reporting, we had a fix under internal reviewing. This issue will be closed automatically once the fix is available on GitHub. Thanks.

closed 2020-08-20 by projectgus