espressif/esp-idf #5633: [PPPoS] Reboot after 3 attempts to connect

Kunislaw, 2020-07-22:
Hello, I test the connection of the SIM800L modem to the network, everything works fine until the 3 connection attempts, then unfortunately it gets a reboot. Here are my logs

```
Executing "/home/karol/.espressif/python_env/idf4.3_py3.8_env/bin/python /home/karol/esp/esp-idf/tools/idf_monitor.py -p /dev/ttyUSB0 -b 115200 --toolchain-prefix xtensa-esp32-elf- /home/karol/esp/projekty/ESP32_GPS_SIM/build/pppos_client.elf -m '/home/karol/.espressif/python_env/idf4.3_py3.8_env/bin/python' '/home/karol/esp/esp-idf/tools/idf.py' '-p' '/dev/ttyUSB0'"...
--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7096
load:0x40078000,len:13732
load:0x40080400,len:4000
0x40080400: _init at ??:?

entry 0x40080684
I (29) boot: ESP-IDF v4.3-dev-472-gcf056a7d0-dirty 2nd stage bootloader
I (29) boot: compile time 18:36:38
I (30) boot: chip revision: 1
I (34) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 4MB
I (54) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot: End of partition table
I (97) boot_comm: chip revision: 1, min. application chip revision: 0
I (104) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x12078 ( 73848) map
I (141) esp_image: segment 1: paddr=0x000220a0 vaddr=0x3ffb0000 size=0x0230c (  8972) load
I (145) esp_image: segment 2: paddr=0x000243b4 vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at /home/karol/esp/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730

I (149) esp_image: segment 3: paddr=0x000247c0 vaddr=0x40080404 size=0x0b624 ( 46628) load
I (177) esp_image: segment 4: paddr=0x0002fdec vaddr=0x00000000 size=0x0022c (   556) 
I (178) esp_image: segment 5: paddr=0x00030020 vaddr=0x400d0020 size=0x3eb28 (256808) map
0x400d0020: _stext at ??:?

I (288) boot: Loaded app from partition at offset 0x10000
I (288) boot: Disabling RNG early entropy source...
I (299) cpu_start: Pro cpu up.
I (299) cpu_start: Starting app cpu, entry point is 0x4008145c
0x4008145c: call_start_cpu1 at /home/karol/esp/esp-idf/components/esp_system/port/cpu_start.c:109

I (0) cpu_start: App cpu up.
I (314) cpu_start: Pro cpu start user code
I (314) cpu_start: Application information:
I (314) cpu_start: Project name:     pppos_client
I (319) cpu_start: App version:      bb4602a-dirty
I (325) cpu_start: Compile time:     Jul 22 2020 20:09:58
I (331) cpu_start: ELF file SHA256:  104a530401a55fde...
I (337) cpu_start: ESP-IDF:          v4.3-dev-472-gcf056a7d0-dirty
I (344) heap_init: Initializing. RAM available for dynamic allocation:
I (351) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (357) heap_init: At 3FFB3B70 len 0002C490 (177 KiB): DRAM
I (363) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (370) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (376) heap_init: At 4008BA28 len 000145D8 (81 KiB): IRAM
I (383) spi_flash: detected chip: generic
I (387) spi_flash: flash io: dio
I (392) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (403) pppos_example: RESET
I (403) gpio: GPIO[27]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 1| Intr:0 
[... 421 lines cut ...]
    ESP_ERROR_CHECK(dce->get_battery_status(dce, &bcs, &bcl, &voltage));
    ESP_LOGI(TAG, "Battery voltage: %d mV", voltage);
    
    esp_netif_ppp_set_auth(esp_netif, auth_type, CONFIG_EXAMPLE_MODEM_PPP_AUTH_USERNAME, CONFIG_EXAMPLE_MODEM_PPP_AUTH_PASSWORD);
    esp_netif_attach(esp_netif, modem_netif_adapter);
}
```

Deinitiating function
```
void gsm_deinit(){
    esp_modem_stop_ppp(dte);
    esp_modem_netif_clear_default_handlers(modem_netif_adapter);
    esp_modem_netif_teardown(modem_netif_adapter);
    esp_modem_remove_event_handler(dte, modem_event_handler);
    esp_event_handler_unregister(IP_EVENT, ESP_EVENT_ANY_ID, &on_ip_event);
    esp_event_handler_unregister(NETIF_PPP_STATUS, ESP_EVENT_ANY_ID, &on_ppp_changed);
    dce->deinit(dce);
    dte->deinit(dte);
    dce = NULL;
    dte = NULL;
}
```

Any ideas what is wrong?

david-cermak, 2020-07-31:
Hi @Kunislaw 

Could you please share your implementation of event handlers as well? Especially what you do to reconnect on disconnecting, here:
```
I (169543) pppos_example: Modem Disconnect from PPP Server, Attempting to connect again
```
There seems to be a race condition somewhere, as according to the backtrace it looks like the esp-netif is trying to set the link down for an interface which already has been destroyed.

Kunislaw, 2020-07-31:
```
#include "string.h"
#include "freertos/FreeRTOS.h"
#include "freertos/event_groups.h"
#include "esp_netif.h"
#include "esp_netif_ppp.h"
#include "mqtt_client.h"
#include "esp_modem.h"
#include "esp_modem_netif.h"
#include "esp_log.h"
#include "sim800.h"
#include "nmea_parser.h"
#include "driver/gpio.h"

#define TIME_ZONE (+2)
#define YEAR_BASE (2000)

bool makeReconnect = false;

uint8_t mqtt_connection_status;
esp_mqtt_client_handle_t mqtt_client = NULL;
int64_t last_send_mqtt = 0;
modem_dte_t *dte;
modem_dce_t *dce;
void *modem_netif_adapter;
esp_netif_t *esp_netif;

static const char *TAG = "pppos_example";
static EventGroupHandle_t event_group = NULL;
static const int CONNECT_BIT = BIT0;
//static const int STOP_BIT = BIT1;
static const int GOT_DATA_BIT = BIT2;

static void modem_event_handler(void *event_handler_arg, esp_event_base_t event_base, int32_t event_id, void *event_data);
static esp_err_t mqtt_event_handler(esp_mqtt_event_handle_t event);
static void on_ppp_changed(void *arg, esp_event_base_t event_base, int32_t event_id, void *event_data);
static void on_ip_event(void *arg, esp_event_base_t event_base,int32_t event_id, void *event_data);
static void gps_event_handler(void *event_handler_arg, esp_event_base_t event_base, int32_t event_id, void *event_data);
void gps_init();
void gsm_reset();
void gsm_init();
void gsm_deinit();
void mqtt_init();

void make_reconnect(void *pvParameter){
    while(1){
        if(makeReconnect){
            makeReconnect = false;
            gsm_deinit();
            gsm_reset();
            gsm_init();
        }
        vTaskDelay(1);
    }
    vTaskDelete(NULL);
}

static void modem_event_handler(void *event_handler_arg, esp_event_base_t event_base, int32_t event_id, void *event_data)
{
    switch (event_id) {
    case ESP_MODEM_EVENT_PPP_START:
        ESP_LOGI(TAG, "Modem PPP Started");
        break;
    case ESP_MODEM_EVENT_PPP_STOP:
        ESP_LOGI(TAG, "Modem PPP Stopped");
        //xEventGroupSetBits(event_group, STOP_BIT);
        break;
    case ESP_MODEM_EVENT_UNKNOWN:
        ESP_LOGW(TAG, "Unknow line received: %s", (char *)event_data);
        break;
[... 228 lines cut ...]
	};
	mqtt_client = esp_mqtt_client_init(&mqtt_config);
	esp_mqtt_client_start(mqtt_client);
}

void app_main(void) {
    ESP_ERROR_CHECK(esp_event_loop_create_default());
    ESP_ERROR_CHECK(esp_netif_init());
    event_group = xEventGroupCreate();
    ESP_LOGI(TAG, "RESET");
    xTaskCreate(&make_reconnect, "make_reconnect", 4096, NULL, 5, NULL);
    gsm_reset();
    ESP_LOGI(TAG, "INIT");
	gsm_init();

    xEventGroupWaitBits(event_group, CONNECT_BIT, pdTRUE, pdTRUE, portMAX_DELAY);

	//mqtt_init();
    //vTaskDelay(5000 / portTICK_PERIOD_MS);
    //gsm_init();
    //ESP_LOGI(TAG, "INIT");
}
```

All my code, thanks for reply

david-cermak, 2020-07-31:
Thanks for sharing the code. It looks like the race condition is in calling `gsm_deinit()` on event `IP_EVENT_PPP_LOST_IP` from your reconnection thread, since the same event internally triggers `esp-netif` to set the link down (from esp-modem). It then calls `esp_modem_netif_teardown()` which destroys its internal `esp_netif` instance and at the same time as TCP/IP thread trying to set the link down for the same (potentially already destroyed) handle.

This also seems linke an oversight from esp-idf side, as the `esp_modem_netif_teardown()` should be safe and wait internally for all esp-netif unfinished activities before destroying interfaces. Thanks for posting this report, we will have to address this issue.

As a workaround, might I suggest waiting for `esp_netif_is_netif_up(netif)` (for netif that was created in your `gsm_init()` function) until it returns `false` before calling `esp_modem_netif_teardown()`. Or maybe just simply call the `esp_netif_action_disconnected(netif, 0, 0, 0);` before the teardown.

closed 2021-01-05 by espressif-bot