espressif/esp-idf #5635: Add Bluetooth Classic HID device support

NNThomasL, 2020-07-23:
I need Classic HID device support for a project and BLE will not work in my case. There is already Classic host and BLE device support but IDF is missing the Classic device feature. The pull request https://github.com/espressif/esp-idf/pull/3425 was put on hold until the new HID interface was ready and I am curious if the request (or any other implementation) would be good to go at this point.

Wth-Esp, 2020-09-16:
Hi

Please use this [BT_HID](https://github.com/espressif/esp-idf/pull/3425) first if you have some problem to use this please leave your comment in that PR.

Thanks

Wth-Esp, 2021-01-18:
Hi all,

We have almost done the HID Device component based on Classic BT and we decide to have more tests on it including the BQB test case suit. I promise here the feature shall release with IDF v4.4.

Thanks for your patience, we much appreciate it.

closed 2021-09-10 by espressif-bot

Alvin1Zhang, 2021-11-30:
Thanks for reporting, this has been implemented in https://github.com/espressif/esp-idf/commit/1c15c9207c0b161ef2afcb69f917ca40cf63e85e, and will be part of IDF v4.4 release, feel free to reopen.