espressif/esp-idf #5636: Intermittent ASSERT_PARAM BT error

KaeLL, 2020-07-23:
## Environment

- Development Kit:      none
- Module or chip used:  ESP32D0WDQ6 (revision 1)
- IDF version: v4.1-beta2
- Build System:         Make
- Compiler version: esp-2020r1-8.2.0
- Operating System:     Linux
- Using an IDE?: No
- Power Supply:         external 5V

## Problem Description

I keep getting this error in a intermittent, though still frequently, fashion.
The way it generally happens is after initialization of both scan and advertising services, it appears to reach some kind of locked state for a few seconds, after which the ASSERT_PARAM triggers, and after ~3s, the board restarts.
Googling it showed a few other people having this issue in the past, but between the issues dying out, and having fixes already issued apparently because of it, it led to nowhere.
Running `grep` on esp-idf looking for ASSERT_PARAM leads to the `libbtdm_app.a` static library, which I assume is related to esp32's Bt controller.
I [had this issue before](https://github.com/espressif/esp-idf/issues/5358) while trying to debug another Bluetooth issue, but [was instructed to re-issue it](https://github.com/espressif/esp-idf/issues/5358#issuecomment-639327009) due to apparently being unrelated to the main problem I was having at the time. This is it.
An interesting detail I realized from the tests I ran is that apparently the core the controller and Bluedroid are pinned to plays a role in this. If I pin the controller to the APP cpu and the host to PRO, the issue appears within a couple of restarts or so.
<details>
<summary>Expand log</summary>

```
ASSERT_PARAM(-218959118 0), in arch_main.c at line 327
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1). 

Core  1 register dump:
PC      : 0x400879ec  PS      : 0x00060734  A0      : 0x8014ed34  A1      : 0x3ffbfbb0  
0x400879ec: r_assert_param at ??:?

A2      : 0x00000001  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x60008054  
A6      : 0x3ffbdbdc  A7      : 0x60008050  A8      : 0x800879e9  A9      : 0x3ffbfb90  
A10     : 0x00000004  A11     : 0x00000000  A12     : 0x6000804c  A13     : 0xffffffff  
A14     : 0x00000000  A15     : 0xfffffffc  SAR     : 0x00000004  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x40087921  LEND    : 0x40087928  LCOUNT  : 0x00000000  
0x40087921: r_assert_param at ??:?

0x40087928: r_assert_param at ??:?

Core  1 was running in ISR context:
EPC1    : 0x4009ed63  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x400879ec
0x4009ed63: uart_hal_write_txfifo at /esp-idf/components/soc/src/hal/uart_hal_iram.c:35

0x400879ec: r_assert_param at ??:?

Backtrace:0x400879e9:0x3ffbfbb0 0x4014ed31:0x3ffbfbd0 0x40019fb5:0x3ffbfbf0 0x400466c3:0x3ffbfc20 0x4008a22e:0x3ffbfc40 0x40089a95:0x3ffbfc60 0x4008ba4e:0x3ffbfc80 0x4008c927:0x3ffbfca0 0x40085852:0x3ffbfcc0 0x0004001e:0x00000000 |<-CORRUPTED
0x400879e9: r_assert_param at ??:?

0x4014ed31: r_platform_reset at ??:?

0x4008a22e: r_lld_evt_end at ??:?

0x40089a95: r_lld_evt_end_isr at ??:?

0x4008ba4e: r_rwble_isr at ??:?

0x4008c927: r_rwbtdm_isr_wrapper at intc.c:?

0x40085852: _xt_lowint1 at /esp-idf/components/freertos/xtensa/xtensa_vectors.S:1105

Core  0 register dump:
PC      : 0x401c1aca  PS      : 0x00060634  A0      : 0x800d1a0d  A1      : 0x3ffce5a0  
0x401c1aca: esp_pm_impl_waiti at /esp-idf/components/esp32/pm_esp32.c:486

A2      : 0x00000000  A3      : 0x00000001  A4      : 0x00000001  A5      : 0x80000001  
A6      : 0x3ff000e0  A7      : 0x00000001  A8      : 0x80099538  A9      : 0x3ffce580  
A10     : 0x3ffce11c  A11     : 0x40001d48  A12     : 0x00000000  A13     : 0x00060c23  
A14     : 0x00000001  A15     : 0x40086114  SAR     : 0x00000000  EXCCAUSE: 0x00000006  
0x40086114: spi_flash_op_block_func at /esp-idf/components/spi_flash/cache_utils.c:92

EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace:0x401c1ac7:0x3ffce5a0 0x400d1a0a:0x3ffce5c0 0x400996a1:0x3ffce5e0 0x4009b455:0x3ffce600
0x401c1ac7: esp_pm_impl_waiti at /esp-idf/components/esp32/pm_esp32.c:484

0x400d1a0a: esp_vApplicationIdleHook at /esp-idf/components/esp_common/src/freertos_hooks.c:63

0x400996a1: prvIdleTask at /esp-idf/components/freertos/tasks.c:3385 (discriminator 1)

0x4009b455: vPortTaskWrapper at /esp-idf/components/freertos/xtensa/port.c:143

ELF file SHA256: 53e97358cd1ec7f7

Rebooting...
```

</details>

Any thoughts regarding this issue, or the triggered assert?

WCCWCC, 2020-07-24:
Hi， @KaeLL 

   You can now reproduce ASSERT_PARAM(-218959118 0), in arch_main.c at line 327?

  This error means that memory allocation failed during malloc, you need to see if you run out of memory.

KaeLL, 2020-07-24:
Hi @WCCWCC 

It's not that. I mean, it must be, but there seems to be another issue starting a chain of events that leads to that error.
The reason I'm saying that is because, firstly, the app is sitting on at least ~100KB of free heap at all times.
Secondly, moving component initializations around seemed to have done away with this particular issue completely. 
Apparently it's possible for some components to interfere with others during intialization. In this case in particular, it seems to be some interaction between the bluetooth stack and NVS. Don't quote me on that one because I didn't do extensive testing to test this conjecture.
There's something else, though. After the ASSERT_PARAM error stopped, something else came up, in a rather amusing way.
The thing is, the ASSERT_PARAM error only happened at app initialization, so I wrote code for the app to reboot itself after 30s. During initialization, the app would get the restart reason, and send it to me through MQTT. After moving code around, ASSERT_PARAM stopped happening completely, and the error below took its place. In what seems like a joke, the error below would repeat itself at every 28th restart, or every ~14min, with the 28th run panicking as shown below. This pattern kept repeating itself until I turned it off, at which point it had happened 8 times.
Do you have any idea of what might be causing the following error? And by the way, let me know if I should close this issue and open another one, since it's not related to the title.
```
/esp_idf/test_master/esp-idf/components/freertos/queue.c:1448 (xQueueGenericReceive)- assert failed!

abort() was called at PC 0x4009c361 on core 0
0x4009c361: xQueueGenericReceive at /esp_idf/test_master/esp-idf/components/freertos/queue.c:1448 (discriminator 3)

Backtrace:0x4008fc4f:0x3ffe17f0 0x4008f9c5:0x3ffe1810 0x4009d1ce:0x3ffe1830 0x4009c361:0x3ffe18a0 0x400d1910:0x3ffe18e0 0x400d19d2:0x3ffe1900 0x401c4ecf:0x3ffe1920 0x401c4f52:0x3ffe1970 0x401a8c06:0x3ffe19a0 0x401a8c71:0x3ffe19c0 0x401a6857:0x3ffe19f0 0x401a57cf:0x3ffe1a10 0x40125262:0x3ffe1a30 0x4009b529:0x3ffe1aa0
0x4008fc4f: panic_abort at /esp_idf/test_master/esp-idf/components/esp_system/panic.c:330

0x4008f9c5: esp_system_abort at /esp_idf/test_master/esp-idf/components/esp_system/system_api.c:97

0x4009d1ce: abort at /esp_idf/test_master/esp-idf/components/newlib/abort.c:46

0x4009c361: xQueueGenericReceive at /esp_idf/test_master/esp-idf/components/freertos/queue.c:1448 (discriminator 3)

0x400d1910: esp_ipc_call_and_wait at /esp_idf/test_master/esp-idf/components/esp_common/src/ipc.c:128

0x400d19d2: esp_ipc_call_blocking at /esp_idf/test_master/esp-idf/components/esp_common/src/ipc.c:144

0x401c4ecf: gpio_isr_register at /esp_idf/test_master/esp-idf/components/driver/gpio.c:522

0x401c4f52: gpio_install_isr_service at /esp_idf/test_master/esp-idf/components/driver/gpio.c:446
```

WCCWCC, 2020-07-27:
Hi, @KaeLL 

I have a suggestion here, if the function may be executed in the interrupt.

The global variables used need to add DRAM_ATTR, and the function definition adds IRAM_ATTR.

Do not pass string constants in calling function parameters.

```c
DRAM_ATTR int count = 0;
void IRAM_ATTR test_function(){}
```

I will also look for a colleague from the peripheral driver to help look at the problem.

KaeLL, 2020-07-27:
> if the function may be executed in the interrupt.

The only thing the interrupt does is set a flag to 1.
> The global variables used need to add DRAM_ATTR, and the function definition adds IRAM_ATTR.

I'm already doing this.
> Do not pass string constants in calling function parameters.

I don't, but thanks for mentioning.
> I will also look for a colleague from the peripheral driver to help look at the problem.

Thanks.

WCCWCC, 2020-07-28:
Hi, @KaeLL 

It's not like a gpio problem, it might be IPC.

Can you provide the code to help us reproduce your problem？

KaeLL, 2020-07-28:
> Can you provide the code to help us reproduce your problem?

Unfortunately I can't because NDA.
I can only detail what happens.
I proceeded to re-order the initializations, moving the Bluetooth stack down to last and the 2 SPI peripherals up in in the initialization order, and the `gpio_install_isr_service` error stopped happening. But after that, the BLE scan would stop working intermittently. After rebooting, the bluetooth controller, despite initialized, would sometimes not run, like it had not been initialized or the scheduler was simply skipping it.
After that, I tried using an event group in order to hold the initialization until the NVS writes were done, but that wasn't enough, and the issue continued. I added a delay of 5s after the 2nd (and last) NVS write performed during the initialization, and now it seems to be working. I've been testing it for hours, and didn't have a single issue with it.
Sorry I can't provide the code, and making an example would be complicated because I can't pinpoint exactly what is causing all these issues, making the narrowing down of what's related to the problem and what isn't really difficult, meaning I'd basically have to provide the whole app.
I may write a an example in the future and use it as a reference for these situations, but until then, to the extent of my knowledge, this is the furthest I can go.
Feel free to close the issue. If I have these issues again, hopefully I'll have more info to share with you.

Alvin1Zhang, 2020-08-05:
@KaeLL Thanks for reporting, feel free to reopen with more updates. Thanks.

closed 2020-08-05 by Alvin1Zhang