espressif/esp-idf #5640: CMock or other mocking frameworks with ESP-IDF for standardized mocking of components

thehellmaker, 2020-07-25:
**Is your feature request related to a problem? Please describe.**

I am always frustrated to have a codebase which doesn't do automated testing. More so when it can't  be mocked to test all branches.

**Describe the solution you'd like**

It would be great to have a mock framework and to have concise documentation on how to use it so that we can build well tested code on esp32

**Describe alternatives you've considered**

http://www.mark-fink.de/2019-01-03-esp32-cmock-cmake/

**Additional context**

Add any other context or screenshots about the feature request here.

Alvin1Zhang, 2020-07-27:
Thanks for raising this feature request.

0xjakob, 2020-07-27:
Hi @thehellmaker!

Thanks for your feature request. In fact, for exactly the reasons you mentioned, we've recently been looking into this topic and trying to find a solution for this. Currently, we plan a solution enabling mocking on the target as well as on host-based tests.

AshUK, 2020-08-10:
Would also love to a see a proper mocking framework implemented in the IDF. We currently write our own mocks for certain peripherals/drivers. However this approach takes time and we usually shortcut it to our specific usecase needs.

0xjakob, 2020-08-10:
@AshUK @thehellmaker 
Currently, our plan is to integrate cmock into IDF. We first want to make it available for target testing. Host-based testing is, as mentioned above, a more difficult task and is planned to be introduced later.

AshUK, 2020-08-13:
@0xjakob we currently use a combination of FFF, Unity and our own mocks (These are essentially driver implementations that dont do anything). Our preference would be to use CMock but we struggled to get it setup with the latest 4.1 IDF and integrate with our CI.

0xjakob, 2020-08-13:
@AshUK Yeah, cmock puts some burdon onto the build system. But we found a solution and created a PoC. I can't promise that we will have it in 4.3 but we're making progress, I'm quite confident that you'll see cmock in IDF in the near/medium term future. The approach we're currently working on integrates cmock as a component while keeping unity as a different component so far. We set $UNITY_DIR to the unity component to make sure cmock doesn't try to use its own unity submodule (this is mostly for internal reasons). The idea is then to mock on a component-level, e.g. mock the whole NVS component. In the end, we will basically have only the code of the component you want to test, the rest is mocked, as far as possible. Whether a component will be mocked, will depend on some SDK-config definitions (or something else, if we find better solutions).

Does that sound reasonable for your use cases?

0xjakob, 2020-08-23:
There are two things here:
1. If having a component depending on the driver, we aim for mocking the driver component itself and its abstract function calls.
1. If the driver itself needs to be tested, then the HAL-layer, which exists for some components already, can be mocked. However, the HAL layer isn't stable (yet), neither a public API. We are working on this, so this might still change a lot in the future.

Another issue is if we want to mock only some peripherals from the driver component. We haven't looked into this yet.

Our main goal for now is to gradually deploy mocking for IDF components which benefit most from mocking. NVS flash is an example for this since it's used a lot by the network-related components but mostly returns predefined values.

0xjakob, 2020-09-21:
@AshUK CMock has been integrated as a component in the master branch recently. Please take a look. Unit tests using mocks of components will follow soon.

AshUK, 2020-10-01:
Hey @0xjakob  looking really good so far, I just have a couple of questions.

I don't think it would be truly possible for us to offer a complete feedback until its possible to mock/stub other components. This is probably already possible. I'm just unsure how.

The current tests in the IDF now look more like unit tests for each component which is much improved from the previous 'build a test app and run it approach'. I'm guessing the direction in the future will be to continue to set up unity tests for integration testing and then use cmock for unit testing?

How exactly are you running these tests ?

Some of the tests are difficult to read and follow. Would be nice if a consistence pattern were used across the IDF e.g. Arrange, Act and Assert (AAA) pattern or GivenWhenThen etc.

0xjakob, 2020-10-13:
Hi @AshUK, we first plan to tweak our build system so that it mocks a whole component. It's not ready to be merged yet, unfortunately. Below is an example we're currently using to mock the nvs system. You can try to use it in an application which needs nvs mocked. In your application's CMakeLists.txt you only need to set `idf_build_set_property(CONFIG_NVS_MOCK 1)
`, than it should automatically build the nvs mocks instead of building the actual NVS. Note that there are some prerequisites, like ruby, to run cmock.  
[CMakeLists.zip](https://github.com/espressif/esp-idf/files/5370263/CMakeLists.zip)

Of course, we plan to integrate this properly into IDF, it's just a PoC currently.

Hope this helps.

Alvin1Zhang, 2021-11-25:
Thanks for reporting and sorry for slow turnaround, fix on master at https://github.com/espressif/esp-idf/commit/912d831dfaf1e1df435734992c3b4b96b80426a3 and on release/4.4 at https://github.com/espressif/esp-idf/commit/fd89e8dae4917232299a2bc39fe749c57af5b632, feel free to reopen.

closed 2021-11-25 by Alvin1Zhang