espressif/esp-idf #5641: smart_config build fail

iloop2020, 2020-07-25:
Introduction
============
My project(esp32-idf v4.0.1) need the smart_config and some external component, then I base on smart_config example.
When if try to enable some external component by updating the "smart_config/main/CMakeLists.txt",
**idf.py build** show a lot of **mbedtls** error message as below:
**esp-idf/wpa_supplicant/libwpa_supplicant.a(crypto_mbedtls.c.obj):(.literal.ecp_opp+0x0): undefined reference to `mbedtls_ecp_copy'**
it seems when **REQUIRES** is added, the original linker is changed then make mbedtls not linked ? 

### Updated smart_config/main/CMakeLists.txt

```cpp
idf_component_register(SRCS "smartconfig_main.c"    
                    INCLUDE_DIRS "."    
                    REQUIRES nvs_flash wpa_supplicant)    
```

### Original smart_config/main/CMakeLists.txt

```cpp
idf_component_register(SRCS "smartconfig_main.c"
                    INCLUDE_DIRS ".")
```

Thank you.

liuzfesp, 2020-07-27:
Hi @iloop2020 everything looks OK on my side, suggest:

- idf.py clean
- git submodule update -f
- make the original smart_config example pass build
- modify the CMakeLists.txt and make the build pass

iloop2020, 2020-07-27:
Hi @liuzfesp ,

Thank you for your support.

Actually, I did the same procedures, but still fail,
Are you using esp32-idf v4.0.1 ? 

Thank you.

liuzfesp, 2020-07-27:
Yes, it's IDF tagged v4.0.1: 4c81978, no other modifications except the CMakeLists.txt.

tobymurray, 2020-07-28:
I can confirm iloop2020's example with the current head of `release/4.0`: 48ea44f.

iloop2020, 2020-07-29:
Hi @tobymurray ,

I am using Ubuntu mate 18.04 also.

liuzfesp, 2020-07-29:
Hello, could you try following way:

- Start a **new** terminal
- cd $IDF_PATH
- install.sh
- source export.sh

then rebuild the example.

tobymurray, 2020-07-29:
I pulled release/4.0 (now on 7abb8e7) and did as you described with no difference in the result, full output is here: https://gist.github.com/tobymurray/e357bf8f8fa4a2e14361915d7869c567

BrianPugh, 2020-07-30:
just weighing in, I'm experiencing similar mbedtls linking issues from wpa_supplicant using a super clean v4.0.1 esp-idf install. I'm also on Ubuntu. See details at https://github.com/BrianPugh/cookiecutter-esp32-webserver/issues/2

tobymurray, 2020-07-30:
I just tried the just-announced 4.1 RC - I successfully built in the scenario described above (i.e. I was unable to reproduce this issue on the 4.1 RC).

sagb2015, 2020-08-03:
@iloop2020 Can you provide the sdkconfig? We are not able to reproduce the issue.

BrianPugh, 2020-08-07:
what's not reproduceable about this? I get this on v4.0.1 on a clean install, default everything (running Ubuntu 19.04)

tobymurray, 2020-08-07:
Same here except on Ubuntu 18.04. There is no customization required beyond what's mentioned in the original post. To be explicit about it:
- install the ESP-IDF with version 4.0.1
- add `REQUIRES nvs_flash wpa_supplicant` to `esp-idf/examples/wifi/smart_config/main/CMakeLists.txt`
- `idf.py build` that project

@sagb2015 or @Alvin1Zhang - are you saying you have tried this on Ubuntu and it works fine there, or you're unable to reproduce this on another OS? As I noted above, this looks like a platform dependent problem to me. Using exactly the same steps I was successfully able to build on Windows but was not on Ubuntu 18.04.

projectgus, 2020-08-13:
Hi all,

Thanks for being patient while we looked into this.

We were indeed unable to reproduce this on three different developer systems (a mixture of Ubuntu and MacOS) with ESP-IDF v4.0.1. I just now set up a clean VM install of Ubuntu Desktop 18.04.3 and ESP-IDF v4.0.1 and could reproduce it there.

The wpa_supplicant component & mbedtls component libraries have a dependency cycle, via some other component libraries. It seems like CMake can resolve this set of relationships in some situations and not in others - it may be CMake version related (I can't reproduce on CMake 16, Ubuntu 18.04 has CMake 10.2), or it might be some other implementation-specific factor.

A quick fix in ESP-IDF is to add the following line to the very end of `components/wpa_supplicant/CMakeLists.txt`:

```
    set_property(TARGET ${COMPONENT_LIB} APPEND PROPERTY LINK_INTERFACE_MULTIPLICITY 3)
```

(This may also be how we fix it permanently, or we may fix it by cleaning up the dependency cycle.)

closed 2020-11-03 by espressif-bot