espressif/esp-idf #5643: Idf 4.3 cpu_start.c error

pilnikov, 2020-07-25:
After update from 4.2 to 4.3 ш got the following error

`[316/1159] Building C object esp-idf/esp_system/CMakeFiles/__idf_esp_system.dir/port/cpu_start.c.obj
FAILED: esp-idf/esp_system/CMakeFiles/__idf_esp_system.dir/port/cpu_start.c.obj
ccache Z:\espressif\tools\xtensa-esp32-elf\esp-2020r2-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -Iconfig -IZ:/esp-idf/components/esp_system/include -IZ:/esp-idf/components/esp_system/private_include -IZ:/esp-idf/components/esp_system/port/include -IZ:/esp-idf/components/newlib/platform_include -IZ:/esp-idf/components/freertos/include -IZ:/esp-idf/components/freertos/xtensa/include -IZ:/esp-idf/components/heap/include -IZ:/esp-idf/components/log/include -IZ:/esp-idf/components/lwip/include/apps -IZ:/esp-idf/components/lwip/include/apps/sntp -IZ:/esp-idf/components/lwip/lwip/src/include -IZ:/esp-idf/components/lwip/port/esp32/include -IZ:/esp-idf/components/lwip/port/esp32/include/arch -IZ:/esp-idf/components/soc/src/esp32/. -IZ:/esp-idf/components/soc/src/esp32/include -IZ:/esp-idf/components/soc/include -IZ:/esp-idf/components/esp_rom/include -IZ:/esp-idf/components/esp_common/include -IZ:/esp-idf/components/xtensa/include -IZ:/esp-idf/components/xtensa/esp32/include -IZ:/esp-idf/components/esp32/include -IZ:/esp-idf/components/driver/include -IZ:/esp-idf/components/driver/esp32/include -IZ:/esp-idf/components/esp_ringbuf/include -IZ:/esp-idf/components/efuse/include -IZ:/esp-idf/components/efuse/esp32/include -IZ:/esp-idf/components/espcoredump/include -IZ:/esp-idf/components/esp_timer/include -IZ:/esp-idf/components/esp_ipc/include -IZ:/esp-idf/components/soc/soc/esp32/. -IZ:/esp-idf/components/soc/soc/esp32/include -IZ:/esp-idf/components/soc/soc/esp32/../include -IZ:/esp-idf/components/vfs/include -IZ:/esp-idf/components/esp_wifi/include -IZ:/esp-idf/components/esp_wifi/esp32/include -IZ:/esp-idf/components/esp_event/include -IZ:/esp-idf/components/esp_netif/include -IZ:/esp-idf/components/tcpip_adapter/include -IZ:/esp-idf/components/app_trace/include -IZ:/esp-idf/components/spi_flash/include -IZ:/esp-idf/components/app_update/include -IZ:/esp-idf/components/bootloader_support/include -IZ:/esp-idf/components/nvs_flash/include -IZ:/esp-idf/components/mbedtls/port/include -IZ:/esp-idf/components/mbedtls/mbedtls/include -IZ:/esp-idf/components/mbedtls/esp_crt_bundle/include -IZ:/esp-idf/components/pthread/include -mlongcalls -Wno-frame-address   -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -mfix-esp32-psram-cache-issue -mfix-esp32-psram-cache-strategy=memw -Og -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.3-dev-472-gcf056a7d0-dirty\" -DESP_PLATFORM -MD -MT esp-idf/esp_system/CMakeFiles/__idf_esp_system.dir/port/cpu_start.c.obj -MF esp-idf\esp_system\CMakeFiles\__idf_esp_system.dir\port\cpu_start.c.obj.d -o esp-idf/esp_system/CMakeFiles/__idf_esp_system.dir/port/cpu_start.c.obj   -c Z:/esp-idf/components/esp_system/port/cpu_start.c
Z:/esp-idf/components/esp_system/port/cpu_start.c: In function 'call_start_cpu0':
Z:/esp-idf/components/esp_system/port/cpu_start.c:225:9: error: unknown type name 'wdt_hal_context_t'
         wdt_hal_context_t rtc_wdt_ctx = {.inst = WDT_RWDT, .rwdt_dev = &RTCCNTL};
         ^~~~~~~~~~~~~~~~~
Z:/esp-idf/components/esp_system/port/cpu_start.c:225:42: error: field name not in record or union initializer
         wdt_hal_context_t rtc_wdt_ctx = {.inst = WDT_RWDT, .rwdt_dev = &RTCCNTL};
                                          ^
Z:/esp-idf/components/esp_system/port/cpu_start.c:225:42: note: (near initialization for 'rtc_wdt_ctx')
Z:/esp-idf/components/esp_system/port/cpu_start.c:225:50: error: 'WDT_RWDT' undeclared (first use in this function); did you mean 'OWDT_RESET'?
         wdt_hal_context_t rtc_wdt_ctx = {.inst = WDT_RWDT, .rwdt_dev = &RTCCNTL};
                                                  ^~~~~~~~
                                                  OWDT_RESET
Z:/esp-idf/components/esp_system/port/cpu_start.c:225:50: note: each undeclared identifier is reported only once for each function it appears in
Z:/esp-idf/components/esp_system/port/cpu_start.c:225:60: error: field name not in record or union initializer
         wdt_hal_context_t rtc_wdt_ctx = {.inst = WDT_RWDT, .rwdt_dev = &RTCCNTL};
                                                            ^
Z:/esp-idf/components/esp_system/port/cpu_start.c:225:60: note: (near initialization for 'rtc_wdt_ctx')
Z:/esp-idf/components/esp_system/port/cpu_start.c:225:72: warning: excess elements in scalar initializer
         wdt_hal_context_t rtc_wdt_ctx = {.inst = WDT_RWDT, .rwdt_dev = &RTCCNTL};
                                                                        ^
Z:/esp-idf/components/esp_system/port/cpu_start.c:225:72: note: (near initialization for 'rtc_wdt_ctx')
Z:/esp-idf/components/esp_system/port/cpu_start.c:226:9: error: implicit declaration of function 'wdt_hal_write_protect_disable'; did you mean 'xthal_icache_disable'? [-Werror=implicit-function-declaration]
         wdt_hal_write_protect_disable(&rtc_wdt_ctx);
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
         xthal_icache_disable
Z:/esp-idf/components/esp_system/port/cpu_start.c:227:9: error: implicit declaration of function 'wdt_hal_disable'; did you mean 'xthal_int_disable'? [-Werror=implicit-function-declaration]
         wdt_hal_disable(&rtc_wdt_ctx);
         ^~~~~~~~~~~~~~~
         xthal_int_disable
Z:/esp-idf/components/esp_system/port/cpu_start.c:228:9: error: implicit declaration of function 'wdt_hal_write_protect_enable'; did you mean 'xthal_icache_enable'? [-Werror=implicit-function-declaration]
         wdt_hal_write_protect_enable(&rtc_wdt_ctx);
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
         xthal_icache_enable
cc1.exe: some warnings being treated as errors
[321/1159] Performing configure step for 'bootloader'
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.22.0.windows.1")
-- The C compiler identification is GNU 8.2.0
-- The CXX compiler identification is GNU 8.2.0
-- The ASM compiler identification is GNU
-- Found assembler: Z:/espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: Z:/espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: Z:/espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: Z:/espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe
-- Check for working CXX compiler: Z:/espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
-- Adding linker script Z:/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Adding linker script Z:/esp-idf/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script Z:/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script Z:/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script Z:/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script Z:/esp-idf/components/bootloader/subproject/main/ld/esp32/bootloader.ld
-- Adding linker script Z:/esp-idf/components/bootloader/subproject/main/ld/esp32/bootloader.rom.ld
-- Components: bootloader bootloader_support efuse esp32 esp_common esp_rom esptool_py log main micro-ecc partition_table soc spi_flash xtensa
-- Component paths: Z:/esp-idf/components/bootloader Z:/esp-idf/components/bootloader_support Z:/esp-idf/components/efuse Z:/esp-idf/components/esp32 Z:/esp-idf/components/esp_common Z:/esp-idf/components/esp_rom Z:/esp-idf/components/esptool_py Z:/esp-idf/components/log Z:/esp-idf/components/bootloader/subproject/main Z:/esp-idf/components/bootloader/subproject/components/micro-ecc Z:/esp-idf/components/partition_table Z:/esp-idf/components/soc Z:/esp-idf/components/spi_flash Z:/esp-idf/components/xtensa
-- Configuring done
-- Generating done
-- Build files have been written to: Z:/examples/Ka_mod/build/bootloader
ninja: build stopped: subcommand failed.
ninja failed with exit code 1`

I fix it by add in cpu_start.c at line 54 

`#include "hal/wdt_hal.h"`

Alvin1Zhang, 2020-08-06:
Thanks for reporting, we had a fix under internal reviewing. The issue will be closed automatically once the fix is available on GitHub. Thanks.

pilnikov, 2020-08-11:
The new release was released - the error remained. Without adding #include "hal / wdt_hal.h" doesn't work

closed 2020-08-20 by projectgus