espressif/esp-idf #5645: implement "looping until connected" and problem with esp_wifi_connect() return

walidamriou, 2020-07-26:
I want to implement a general WiFi code call it as a function and make the board waiting untile connected to WiFi. I have implemented it but the problem is the work one time.
I have put  "esp_wifi_connect()" function in while loop until return ESP_OK (that what I think) after "SYSTEM_EVENT_STA_START" event, but I see that this function return ESP_OK even the board not connected to the wifi !! so what is ESP_OK means and what the condition that helps me to implement this idea?
The code of the loop part:
...
esp_err_t err;
do{
err = esp_wifi_connect();
} while(err != ESP_OK);
...

I have looked here: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_wifi.html?highlight=esp_wifi_connect#_CPPv416esp_wifi_connectv

igrr, 2020-07-26:
Hi @walidamriou, esp_wifi_connect function is a non-blocking function, it doesn't wait until the connection is established. Wi-Fi driver communicates connection status using WIFI_EVENT_STA_CONNECTED event: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html#wifi-event-sta-connected.
Please have a look at https://github.com/espressif/esp-idf/blob/master/examples/wifi/getting_started/station/main/station_example_main.c example. If you aren't using the master branch, check the version of this example in your copy of IDF. The example doesn't check for WIFI_EVENT_STA_CONNECTED (it waits until the IP address is obtained), but the handling of WIFI_EVENT_STA_CONNECTED is very similar to what is shown in the example.

liuzfesp, 2020-08-04:
Hi @walidamriou, does @igrr's explanation solve your issue?

walidamriou, 2020-08-05:
Hi @liuzfesp I didn't try it yet

Alvin1Zhang, 2020-09-01:
@walidamriou Thanks for reporting, would you please help try and share some updates? Thanks.

Alvin1Zhang, 2020-09-25:
@walidamriou Thanks for reporting, would you please help share if any updates for this issue? Thanks.

walidamriou, 2020-09-29:
@Alvin1Zhang I will come back to it in the future, after complete the current tasks.

Alvin1Zhang, 2020-10-12:
@walidamriou Thanks for sharing the updates.

closed 2021-02-26 by Alvin1Zhang