espressif/esp-idf #5648: Add MQTT_EVENT_ANY = -1 to esp_mqtt_event_id_t enum

benas1989, 2020-07-26:
Hello,
i try to register for all mqtt events(esp_mqtt_client_register_event), but always visual studio show error.

[image]

I added it myself and now it works without problems.

[image]

david-cermak, 2020-07-27:
Hi @benas1989 

Which version of esp-idf or mqtt client are you using? This change has been applied already and referenced to all major IDF releases, please see https://github.com/espressif/esp-mqtt/blob/6bc94add892437d0fd50f26bfabe78c646648c13/include/mqtt_client.h#L40

benas1989, 2020-07-27:
I don’t remember exactly. I downloaded yesterday through a visual studio add-on. Probably 4.0.1 but I'll see later in the evening

benas1989, 2020-07-27:
esp-idf-v4.0.1

benas1989, 2020-07-27:
I feel like something is wrong with the visual studio add-on. Because it shows that it is installing version 4.01 but this update is not available. I copied straight from github and everything is fine now

brianignacio5, 2020-07-31:
Hi @benas1989  when you visual studio add-on, do you mean our [ESP-IDF extension for Visual Studio Code](https://github.com/espressif/vscode-esp-idf-extension) ? If that's so then can you please share with us some steps to reproduce this behavior by posting an issue there or just describing it to us in the [spectrum chat](https://spectrum.chat/espidf-vsc?tab=posts) ?

benas1989, 2020-07-31:
Hi,
I think here is the problem that installs version 4.1. I can't choose version 4.2. They are not on the list. :)
[image]

brianignacio5, 2020-08-03:
v4.2 is the one corresponding to master (development branch)

brianignacio5, 2021-04-20:
Can we close this @benas1989  ?

benas1989, 2021-04-20:
Yes

Alvin1Zhang, 2021-04-20:
Thanks for sharing updates, feel free to reopen.

closed 2021-04-20 by Alvin1Zhang