espressif/esp-idf #5650: > Cmake Parsing Error due to inverted slashes '\'

nabeelkirmani, 2020-07-27:
> 
> 
> This looks like a regression with a problem that was solved previously. In the meantime, can you try adding the following line:
> 
> ```
> set(COMPONENT_DIR ${CMAKE_CURRENT_LIST_DIR})
> ```
> 
> after
> 
> ```
> idf_build_get_property(soc_name IDF_TARGET)
> ```
> 
> for both `components/soc/CMakeLists.txt` and `components/efuse/CMakeLists.txt`

I did the same as instructed above but the issue still remains.

Here is output of _ESP-IDF Command Prompt (cmd.exe)_ :
````powershell

Using Python in C:\Program Files\Python38\
Python 3.8.5
Using Git in C:\Program Files\Git\cmd\
git version 2.27.0.windows.1
Setting IDF_PATH: E:\Source\ESP-IDF\esp-idf

Adding ESP-IDF tools to PATH...
    E:\Program Files\ESP-IDF Tools\.espressif\tools\xtensa-esp32s2-elf\esp-2020r2-8.2.0\xtensa-esp32s2-elf\bin
    E:\Program Files\ESP-IDF Tools\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin
    E:\Program Files\ESP-IDF Tools\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin
    E:\Program Files\ESP-IDF Tools\.espressif\tools\idf-exe\1.0.1\
    E:\Program Files\ESP-IDF Tools\.espressif\tools\ccache\3.7\
    E:\Program Files\ESP-IDF Tools\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
    E:\Program Files\ESP-IDF Tools\.espressif\python_env\idf4.3_py3.8_env\Scripts

Checking if Python packages are up to date...
Python requirements from E:\Source\ESP-IDF\esp-idf\requirements.txt are satisfied.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

E:\Source\ESP-IDF\esp-idf>cd examples\get-started\hello_world

E:\Source\ESP-IDF\esp-idf\examples\get-started\hello_world>idf.py build
Executing action: all (aliases: build)
Running cmake in directory e:\source\esp-idf\esp-idf\examples\get-started\hello_world\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized -DCCACHE_ENABLE=1 e:\source\esp-idf\esp-idf\examples\get-started\hello_world"...
Warn about uninitialized values.
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.27.0.windows.1")
-- IDF_TARGET not set, using default target: esp32
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 8.2.0
-- The CXX compiler identification is GNU 8.2.0
-- The ASM compiler identification is GNU
-- Found assembler: E:/Program Files/ESP-IDF Tools/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: E:/Program Files/ESP-IDF Tools/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: E:/Program Files/ESP-IDF Tools/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: E:/Program Files/ESP-IDF Tools/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe
-- Check for working CXX compiler: E:/Program Files/ESP-IDF Tools/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
CMake Error at E:/Source/ESP-IDF/esp-idf/tools/cmake/component.cmake:224 (message):
  CMake Error at build_properties.temp.cmake:2 (set):

    Syntax error in cmake code at

      E:/Source/ESP-IDF/esp-idf/examples/get-started/hello_world/build/build_properties.temp.cmake:2

    when parsing string

      E:\Program

    Invalid character escape '\P'.

  Call Stack (most recent call first):

    E:/Source/ESP-IDF/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:2 (include)

  CMake Error at build_properties.temp.cmake:2 (set):

    Syntax error in cmake code at

      E:/Source/ESP-IDF/esp-idf/examples/get-started/hello_world/build/build_properties.temp.cmake:2

    when parsing string

      Files\ESP-IDF

    Invalid character escape '\E'.

  Call Stack (most recent call first):

    E:/Source/ESP-IDF/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:2 (include)

  CMake Error at build_properties.temp.cmake:2 (set):

    Syntax error in cmake code at

      E:/Source/ESP-IDF/esp-idf/examples/get-started/hello_world/build/build_properties.temp.cmake:2

    when parsing string

      Tools\.espressif\python_env\idf4.3_py3.8_env\Scripts\python.exe

    Invalid character escape '\p'.

  Call Stack (most recent call first):

    E:/Source/ESP-IDF/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:2 (include)

Call Stack (most recent call first):
  E:/Source/ESP-IDF/esp-idf/tools/cmake/build.cmake:405 (__component_get_requirements)
  E:/Source/ESP-IDF/esp-idf/tools/cmake/project.cmake:395 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also "E:/Source/ESP-IDF/esp-idf/examples/get-started/hello_world/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1

E:\Source\ESP-IDF\esp-idf\examples\get-started\hello_world>

````

and here is the attached CMakeOutput.log file 
[CMakeOutput.log](https://github.com/espressif/esp-idf/files/4982954/CMakeOutput.log)

_Originally posted by @nabeelkirmani in https://github.com/espressif/esp-idf/issues/3771#issuecomment-664477804_

Alvin1Zhang, 2020-07-28:
Thanks for reporting, we will look into.

ankit-thealchemist, 2020-07-28:
> > This looks like a regression with a problem that was solved previously. In the meantime, can you try adding the following line:
> > ```
> > set(COMPONENT_DIR ${CMAKE_CURRENT_LIST_DIR})
> > ```
> > 
> > 
> > after
> > ```
> > idf_build_get_property(soc_name IDF_TARGET)
> > ```
> > 
> > 
> > for both `components/soc/CMakeLists.txt` and `components/efuse/CMakeLists.txt`
> 
> I did the same as instructed above but the issue still remains.
> 
> Here is output of _ESP-IDF Command Prompt (cmd.exe)_ :
> 
> ```powershell
> Using Python in C:\Program Files\Python38\
> Python 3.8.5
> Using Git in C:\Program Files\Git\cmd\
> git version 2.27.0.windows.1
> Setting IDF_PATH: E:\Source\ESP-IDF\esp-idf
> 
> Adding ESP-IDF tools to PATH...
>     E:\Program Files\ESP-IDF Tools\.espressif\tools\xtensa-esp32s2-elf\esp-2020r2-8.2.0\xtensa-esp32s2-elf\bin
>     E:\Program Files\ESP-IDF Tools\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin
>     E:\Program Files\ESP-IDF Tools\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin
>     E:\Program Files\ESP-IDF Tools\.espressif\tools\idf-exe\1.0.1\
>     E:\Program Files\ESP-IDF Tools\.espressif\tools\ccache\3.7\
>     E:\Program Files\ESP-IDF Tools\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
>     E:\Program Files\ESP-IDF Tools\.espressif\python_env\idf4.3_py3.8_env\Scripts
> 
> Checking if Python packages are up to date...
> Python requirements from E:\Source\ESP-IDF\esp-idf\requirements.txt are satisfied.
> 
> Done! You can now compile ESP-IDF projects.
> Go to the project directory and run:
> 
>   idf.py build
> 
> 
> E:\Source\ESP-IDF\esp-idf>cd examples\get-started\hello_world
> 
> E:\Source\ESP-IDF\esp-idf\examples\get-started\hello_world>idf.py build
> Executing action: all (aliases: build)
> Running cmake in directory e:\source\esp-idf\esp-idf\examples\get-started\hello_world\build
> Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized -DCCACHE_ENABLE=1 e:\source\esp-idf\esp-idf\examples\get-started\hello_world"...
> Warn about uninitialized values.
> -- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.27.0.windows.1")
> -- IDF_TARGET not set, using default target: esp32
> -- ccache will be used for faster recompilation
> -- The C compiler identification is GNU 8.2.0
> -- The CXX compiler identification is GNU 8.2.0
> -- The ASM compiler identification is GNU
> -- Found assembler: E:/Program Files/ESP-IDF Tools/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
> -- Check for working C compiler: E:/Program Files/ESP-IDF Tools/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
> -- Check for working C compiler: E:/Program Files/ESP-IDF Tools/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe -- works
> -- Detecting C compiler ABI info
> -- Detecting C compiler ABI info - done
> -- Detecting C compile features
> -- Detecting C compile features - done
> -- Check for working CXX compiler: E:/Program Files/ESP-IDF Tools/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe
> -- Check for working CXX compiler: E:/Program Files/ESP-IDF Tools/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe -- works
> -- Detecting CXX compiler ABI info
> -- Detecting CXX compiler ABI info - done
> -- Detecting CXX compile features
> -- Detecting CXX compile features - done
> -- Building ESP-IDF components for target esp32
[... 84 lines cut ...]
> 
> Call Stack (most recent call first):
>   E:/Source/ESP-IDF/esp-idf/tools/cmake/build.cmake:405 (__component_get_requirements)
>   E:/Source/ESP-IDF/esp-idf/tools/cmake/project.cmake:395 (idf_build_process)
>   CMakeLists.txt:6 (project)
> 
> 
> -- Configuring incomplete, errors occurred!
> See also "E:/Source/ESP-IDF/esp-idf/examples/get-started/hello_world/build/CMakeFiles/CMakeOutput.log".
> cmake failed with exit code 1
> 
> E:\Source\ESP-IDF\esp-idf\examples\get-started\hello_world>
> ```
> 
> and here is the attached CMakeOutput.log file
> [CMakeOutput.log](https://github.com/espressif/esp-idf/files/4982954/CMakeOutput.log)
> 
> And Here is the snapshot of _build_properties.temp.cmake_ file indicating back-slashs which should be forward-slashes.
> [image]
> 
> _Originally posted by @nabeelkirmani in [#3771 (comment)](https://github.com/espressif/esp-idf/issues/3771#issuecomment-664477804)_

Hello, 

Yes I am also experiencing the same while compiling and while also opening the menuconfig.

istokm, 2020-07-30:
#5512

closed 2020-08-20 by nabeelkirmani