espressif/esp-idf #5653: Not able to Erase flash after esp ota update

lakshmisivareddy, 2020-07-28:
Hi i am working with esp-idf v3.3
i am able to do update the esp32 ota successfully 
but after the update i am trying to erase the 4kb of memory stating from this address 0x280000  
i am seeing application crash after calling ease flash function (spi_flash_erase_range)

any help would be appreciated 

Regards
Siva

Alvin1Zhang, 2020-07-28:
@lakshmisivareddy Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

lakshmisivareddy, 2020-07-28:
Hi Alvin1Zhang
please find the below details 
idf -v3.3
tool chain : 5.2
device : esp32-wroom-32
log output : 
abort() was called at PC 0x40111b25 on core 0
0x40111b25: is_safe_write_address at D:/WorkSpace/esp-32/newesp32/esp-at/esp-idf/components/spi_flash/flash_ops.c:126

ELF file SHA256: 7f60046782aaf58883484a8f5cf35ce23e9cdfa31af320121ad502f077fc9530

Backtrace: 0x4008dfe4:0x3fff4590 0x4008e1fd:0x3fff45b0 0x40111b25:0x3fff45d0 0x40086532:0x3fff45f0 0x4010c7d9:0x3fff4620 0x40107acd:0x3fff4650 0x4010c8a1:0x3fff4680 0x4011f3bd:0x3fff46a0 0x401
1f599:0x3fff46d0 0x4011f730:0x3fff4720 0x4011f97c:0x3fff4740 0x4011fa4a:0x3fff4770 0x4010ca98:0x3fff4790 0x400952ad:0x3fff47b0
0x4008dfe4: invoke_abort at D:/WorkSpace/esp-32/newesp32/esp-at/esp-idf/components/esp32/panic.c:715

0x4008e1fd: abort at D:/WorkSpace/esp-32/newesp32/esp-at/esp-idf/components/esp32/panic.c:715

0x40111b25: is_safe_write_address at D:/WorkSpace/esp-32/newesp32/esp-at/esp-idf/components/spi_flash/flash_ops.c:126

0x40086532: spi_flash_erase_range at D:/WorkSpace/esp-32/newesp32/esp-at/esp-idf/components/spi_flash/flash_ops.c:210 (discriminator 2)

mahavirj, 2020-08-03:
@lakshmisivareddy 

`is_safe_write_address` can fail if you try to operate on currently executing partition:

https://github.com/espressif/esp-idf/blob/bf022060964128556b3d3205b65c5d35df9beef6/components/spi_flash/flash_ops.c#L117-L133

Can you please confirm this? If not, then please share your partition table file and console logs.

lakshmisivareddy, 2020-08-04:
Hi mahavirj 
Thanks for the response 
please find the attached files 
the address i am trying to erase is  0x280000
before ota it is working i am able to erase the flash but after the OTA i am not able to erase the flash 
from partition table 
ota_0,app,    ota_0,   0x100000,0x180000
ota_1,app,    ota_1,   0x280000,0x180000

so which section of memory will be free irrespective of OTA Updates 
so that i can use that memory to store certain information 

[IDF_VERSION.txt](https://github.com/espressif/esp-idf/files/5020071/IDF_VERSION.txt)
[partitions_at.txt](https://github.com/espressif/esp-idf/files/5020072/partitions_at.txt)

mahavirj, 2020-08-04:
@lakshmisivareddy 

Based on your partition table, you can not use section of memory from 0x100000 on-wards up-to 4MB, since this range is occupied by `ota_0` and `ota_1` partitions. One way is to reduce size of these partitions from `0x180000` and allocate that much into an independent partition which can then be used to store additional information.

- How much is maximum size of firmware `.bin` image for your application? Is it possible to reduce `ota_0`, `ota_1` partition size?
- Alternatively, `nvs` partition also looks big enough (56K), did you consider that for storing additional information?

lakshmisivareddy, 2020-08-04:
my currently bin size is 1.33 MB 
my understanding this how memory s being used  
esptool.py --chip esp32 --port COM27 --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 
0x10000 ota_data_initial.bin 
0x1000  bootloader.bin 
0x20000 at_customize.bin 
0x24000  server_cert.bin 
0x26000 server_key.bin 
0x28000 server_ca.bin 
0x2a000  client_cert.bin 
0x2c000 client_key.bin 
0x2e000 client_ca.bin 
0x30000 factory_param.bin 
0x70000 fatfs.bin 
0xf000 phy_init_data.bin 
0x100000 esp-at.bin 
0x8000 partitions_at.bin 

i will be strong the firmware for host so it will be more that 56k 
how i can i reduce the partition size ?

mahavirj, 2020-08-05:
@lakshmisivareddy 

I am not sure if understand your last comment but you can try to use following partition table (**update your `partitions.csv` file**):

```
# Name,   Type, SubType, Offset,  Size
phy_init, data, phy,     0xf000,  0x1000
otadata,data,ota,  0x10000,        8K
nvs,data,nvs,   0x12000,        56K
at_customize,0x40,0,0x20000,0xE0000
ota_0,app,    ota_0,   0x100000,0x170000
ota_1,app,    ota_1,   0x270000,0x170000
```
This allows to free up 128KB flash memory at address starting from: 0x3E0000 which can be used for storing additional information. You can also convert this to new partition of type `data`.

Hope this helps.

mahavirj, 2020-08-17:
@lakshmisivareddy Did above suggestion help? Can we close the issue?

mahavirj, 2020-08-19:
Closing due to lack of response. Please feel to reopen in case issue still exist.

closed 2020-08-19 by mahavirj