espressif/esp-idf #5654: DPP Support

Shyamakshi, 2020-07-28:
I see DPP support added in wpa_supplicant. Reference : https://github.com/espressif/esp-idf/commit/ab784bb53aaede8e9085be23ad10f5cadb4f3b31

Is there any followup integration patch for DPP support in the Wi-Fi static library?

sagb2015, 2020-07-28:
Hi @Shyamakshi, The full support is about to merge into master. It should be available in a week or two..

AxelLin, 2020-08-25:
> Hi @Shyamakshi, The full support is about to merge into master. It should be available in a week or two..

@sagb2015 
It has been 4 weeks since your reply, is the full support for DPP available now? seems not found by checking the git log.

sagb2015, 2020-08-26:
Hi @AxelLin, we will try to make this available soon. Our merges are happening slower than expected. Will update here once available.

AxelLin, 2020-10-15:
@sagb2015 
The recent update still does not include this.
If this feature is really *ready* as you said, you probably need to check what is wrong in your process.

AxelLin, 2020-11-04:
Hi @sagb2015 

Your comment on 28 July gave people the expectation that DPP support will be available soon. 
However, it's still not available after waiting for 3 months.
So how is the status now?
Is it possible to post the patch for testing?

closed 2021-02-01 by igrr