espressif/esp-idf #5656: i2s_read returns bogus zero sample exactly every 2044 bytes

aCaB, 2020-07-28:
## Environment

- Development Kit:      [none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it):
    //  v4.3-dev-472-gcf056a7d0
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0
- Operating System:     [Linux]
- Using an IDE?: [No]
- Power Supply:         [USB]

## Problem Description

i2s_read data appear to contain a spurious zero value sample exactly every 2044 bytes

### Expected Behavior

there should be no spurious samples

### Actual Behavior

ADC level is set to a know reference value (0x600)
Here is a dump of channel 0:
```
00000000: 0006 0006 0006 0006 0006 0006 0006 0006  ................
00000010: 0006 0006 0006 0006 0006 0006 0006 0006  ................
00000020: 0006 0006 0006 0006 0006 0006 0106 0006  ................
[ all is well ..]
000007e0: 0006 0006 0006 0006 0006 0006 0006 0006  ................
000007f0: 0006 0006 0006 0006 0006 0006 0000 0006  ................
spurious sample ------------------------^^^^^
00000800: 0006 0006 0006 0006 0006 0006 0006 0006  ................
00000810: 0006 0006 0006 0006 0006 0006 0006 0006  ................
```

Note that a single spurious sample is injected even when sampling multiple channels.
E.g. when sampling chan 0 and 3:
```
00000000: 0030 0006 0030 0006 0030 0006 0030 0006  .0...0...0...0..
00000010: 0030 0006 0030 0006 0030 0006 0030 0006  .0...0...0...0..
[ all good ... ]
000007d0: 0030 0006 0030 0006 0030 0006 0030 0006  .0...0...0...0..
000007e0: 0030 0006 0030 0006 0030 0006 0030 0006  .0...0...0...0..
000007f0: 0030 0006 0030 0006 0030 0006 0000 0030  .0...0...0.....0
[ spurious sample here -----------------^^^^ ]
[ note how the channels are inverted now ]
00000800: 0006 0030 0006 0030 0006 0030 0006 0030  ...0...0...0...0
00000810: 0006 0030 0006 0030 0006 0030 0006 0030  ...0...0...0...0
[ ... all well but then again... ]
00000fd0: 0006 0030 0006 0030 0006 0030 0006 0030  ...0...0...0...0
00000fe0: 0006 0030 0006 0030 0006 0030 0006 0030  ...0...0...0...0
00000ff0: 0006 0030 0006 0030 0006 0000 0030 0006  ...0...0.....0..
[ spurious sample here ------------^^^^ ]
[ and so on... ]
00001000: 0030 0006 0030 0006 0030 0006 0030 0006  .0...0...0...0..
00001010: 0030 0006 0030 0006 0030 0006 0030 0006  .0...0...0...0..

### Steps to reproduce

1. intialise i2s in internal adc mode with 16 buffers of 1024 bytes each
2. call i2s_read asking > 2048 bytes
3. note the spurious sample being injected

### Code to reproduce this issue

```c
// INIT
     i2s_config_t i2s_config = {
         .mode = I2S_MODE_MASTER | I2S_MODE_RX | I2S_MODE_ADC_BUILT_IN,
         .sample_rate =  16000,
         .bits_per_sample = 16,
         .communication_format = I2S_COMM_FORMAT_STAND_MSB,
         .channel_format = I2S_CHANNEL_FMT_ONLY_LEFT,
         .intr_alloc_flags = 0,
         .dma_buf_count = 16,
         .dma_buf_len = 1024,
     };
     i2s_driver_install(I2S_NUM_0, &i2s_config, 0, NULL);
     i2s_set_adc_mode(ADC_UNIT_1, ADC1_CHANNEL_0);

// READ
i2s_adc_enable(I2S_NUM_0);
i2s_read(I2S_NUM_0, my_buff, 4096, &bytes_read, portMAX_DELAY);

// note that my_buff[2044]==0 && my_buff[2045]==00
```

Alvin1Zhang, 2020-07-29:
Thanks for reporting, we will look into.

Ethan-CYQ, 2020-08-05:
Hi, @aCaB 
sorry I could't reproduce your problem. 
Have you try other ADC level? Is it still have spurious sample?

Alvin1Zhang, 2021-08-24:
Thanks for reporting, will close due to short of feedback and can't reproduce locally, feel free to reopen with more details/updates. Thanks.

closed 2021-08-24 by Alvin1Zhang