espressif/esp-idf #5657: Remote branch cf056a7 not found in upstream origin

gvdhoven, 2020-07-28:
The setup guide for mac (https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/index.html#linux-and-macos) refers to branch cf056a7 which cannot be found anymore:

```
git clone -b cf056a7 --recursive https://github.com/espressif/esp-idf.git
Cloning into 'esp-idf'...
fatal: Remote branch cf056a7 not found in upstream origin
```

In addition; the `cd ~/esp` needs to be prepended by `mkdir ~/esp` if you never downloaded the ESP toolchain before.

krzychb, 2020-07-28:
@gvdhoven thank you for reporting this issue. 

The line that returns the error:

```
git clone -b cf056a7 --recursive https://github.com/espressif/esp-idf.git
```
should be replaced with:

```
git clone --recursive https://github.com/espressif/esp-idf.git
```

I am sorry for this issue. Fix is in preparation and will be merged soon.

Alvin1Zhang, 2020-08-12:
Fixed at https://github.com/espressif/esp-idf/commit/28021365952e55fe276771e2e1526d9597b40f2d, feel free to reopen if the issue still happens. Thanks.

closed 2020-08-12 by Alvin1Zhang