espressif/esp-idf #5659: WPS: cpu starvation

boborjan2, 2020-07-28:
## Environment

- Development Kit:      ESP32-LyraT
- Kit version (for WroverKit/PicoKit/DevKitC): v4.3
- Module or chip used:  ESP32-WROVER-B
- IDF version (run ``git describe --tags`` to find it): v4.3-dev-472-gcf056a7d0
- Build System:         Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): crosstool-NG esp-2020r2) 8.2.0
- Operating System:     Linux
- Using an IDE?:  No
- Power Supply:         USB

## Problem Description

WPS task takes full cpu and starves IDLE1 task for at least 2 seconds (right after I start wps). My watchdog task timeout is set to 1s and when I start WPS (push button) via esp API, I get a watchdog reset.
I need to set it above 3s to get my system run WPS. Otherwise WPS is running as expected.
Note: my cpu is running @80MHz.

### Expected Behavior

WPS task should let other low priority tasks run every now and then (using e.g. small delays - vTaskDelay(1) etc)

### Actual Behavior

WPS tasks takes whole cpu for seconds.

### Steps to reproduce

1) start any example system with task watchdog enabled.
2) start WPS

### Code to reproduce this issue

```cpp
		esp_wps_config_t config = WPS_CONFIG_INIT_DEFAULT(WPS_TYPE_PBC);

	    esp_wifi_wps_enable(&config);
	    esp_wifi_wps_start(0);

```

Alvin1Zhang, 2020-07-29:
Thanks for reporting, we will look into.

nachiketkukade, 2020-08-04:
@boborjan2 can you check in your sdkconfig if below two flags are set?
CONFIG_MBEDTLS_HARDWARE_MPI
CONFIG_MBEDTLS_HARDWARE_SHA

boborjan2, 2020-08-10:
Hi,

CONFIG_MBEDTLS_HARDWARE_SHA=y
# CONFIG_MBEDTLS_HARDWARE_MPI is not set

I am going to redo the test with CONFIG_MBEDTLS_HARDWARE_MPI enabled.

Thx
Viktor

nachiketkukade, 2020-08-10:
Sure, these two flags are enabled by default since end of 2019. That should cut down the time for WPS start up and not trigger the Watchdog. Let me know your test result.

boborjan2, 2020-08-17:
I confirm that enabling CONFIG_MBEDTLS_HARDWARE_MPI  does help: I can start WPS with watchdog timeout == 1s.
However, when the connection and handshake is performed with the AP, the process needs at least 2s watchdog timeout. Using 1s, watchdog reset is triggered. I can live now with 2s but it would be nice if CPU was not starved for huge periods like 1s in any scenario.

boborjan2, 2020-08-17:
btw, here is the backtrace when watchdog resets:
0x40087e28 T panic_abort
0x40088498 T esp_system_abort
0x4008edec T abort
0x40100a14 t task_wdt_isr
0x40082060 t _xt_lowint1
0x4000bfdc A _xtos_set_intlevel
0x4008a4ec T vPortExitCritical
0x40145954 t esp_sha_lock_engine_common
0x401459c8 T esp_sha_try_lock_engine
0x40146ec8 T mbedtls_internal_sha1_process
0x4014800c t mbedtls_sha1_update_ret$part$0
0x40148080 T mbedtls_sha1_update_ret
0x4014612c t sha1_update_wrap
0x40144974 T mbedtls_md_hmac_starts
0x4017127c T mbedtls_pkcs5_pbkdf2_hmac
0x4015e6cc T pbkdf2_sha1
0x40166a00 T wpa_set_passphrase
0x40166aa8 T wpa_set_bss
0x4016068c T wpa_config_bss
0x401606c4 T wpa_sta_connect
0x40117b0c T cnx_sta_connect_led_timer_cb
0x40117b0c T cnx_sta_connect_led_timer_cb
0x40118e18 T cnx_remove_all_rc
0x401191ec T cnx_start_handoff_cb
0x40113d80 T clear_bss_queue
0x4011421c T scan_inter_channel_timeout_process
0x401174ec T chm_end_op
0x40117520 T chm_end_op_timeout_process
0x40116f7c T ieee80211w_get_igtk_from_keyidx
0x401172e8 T ieee80211_timer_do_process
0x40090b10 T ppTask

nachiketkukade, 2020-10-16:
> I confirm that enabling CONFIG_MBEDTLS_HARDWARE_MPI does help: I can start WPS with watchdog timeout == 1s.
> However, when the connection and handshake is performed with the AP, the process needs at [...]

Just to close off, I agree that 1 second is a huge period. this amount of time is mainly due to passphrase generation for the WPA2-PSK which requires 4096 hash operations, seen as pbkdf2_sha1 in the dump stack. Unfortunately the SHA1 Hardware acceleration on ESP32 is just not fast enough, but you can still make it within a second if you bump up the CPU clock speed above 80MHz.
@Alvin1Zhang please close the issue.

closed 2020-10-16 by nachiketkukade