espressif/esp-idf #5661: BLE_MESH: bt_mesh_app_key_get, Failed to get AppKey after reset

dsoon, 2020-07-29:
[onoff_client.zip](https://github.com/espressif/esp-idf/files/4994733/onoff_client.zip)

## Environment

- Development Kit :  LONLIN32

- Kit version : V1.0.0

- Module or chip used:  ESP32-WROOM-32

- IDF version (run ``git describe --tags`` to find it): 4.0.1

- Build System:         idf.py

- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0

- Operating System:     Windows10

- (Windows only) environment type: PowerShell

- Using an IDE?: Yes : VCS Code

- Power Supply:  USB

## Problem Description

Used esp_ble_mesh\ble_mesh_node\onoff_client and onoff_server examples
Used nRF App on Android to provision both devices
When newly provisioned both applications work as expected - button on client can switch the LED on server on and off
When the onoff_client LOLIN32 board is reset, the client is unable to send 

### Expected Behaviour

Client button controls Server LED even after board reset
All provisioned data should be persisted in NVS (enabled in menuconfig)

### Actual Behaviour

Client button unable to send BLE message to client to control LED after client board reset.

### Steps to reproduce

1.  Erase flash, flash client and server code
2. .Provision with nRF app
3. Applications works as expected
4. Press reset button on onoff_client, error is produced on monitor when extern switch is pressed.

// If possible, attach a picture of your setup/wiring here.

Used 2 separate LOLIN32 boards, one for onoff_server and one for onoff_client
onoff_server used a the on board blue led on gpio5 (modified board.c/h)
board.h
#define LED_R GPIO_NUM_5
#define LED_G GPIO_NUM_5
#define LED_B GPIO_NUM_5

onoff_client used an external push to make switch on cpio15 pulled to ground.
board.c 
#define BUTTON_IO_NUM           15
#define BUTTON_ACTIVE_LEVEL     1 

### Monitor Log - refer to attatched files

### Code to reproduce this issue - no changes to original example code

Refer to:
esp_ble_mesh\ble_mesh_node\onoff_server\main\ble_mesh_demo_main.c
esp_ble_mesh\ble_mesh_node\onoff_client\main\ble_mesh_demo_main.c

## Other items if possible

- [ x ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ x ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ x ] Monitor log
- [ ] coredump (This provides stacks of tasks.)

chegewara, 2020-07-29:
Did you enable persistent provisioning in menuconfig?

dsoon, 2020-07-29:
> Did you enable persistent provisioning in menuconfig?

I enabled "Store BLE Mesh configuration persistently" in menuconfig; 

I assume provisioning data will be be saved with this option?

dsoon, 2020-07-29:
When I review the monitor when I press the switch I see the following:

```
I (95734) BOARD: tap cb (RELEASE)
I (95734) BLE_MESH: net_idx 0x0000 app_idx 0x0000 dst 0xffff
I (95734) BLE_MESH: len 4: 82030101
```

After I reset the onoff_client and it has read the provisioned settings from NVS and I press the switch, the log looks like this:

```
I (16654) BOARD: tap cb (RELEASE)
I (16654) BLE_MESH: net_idx 0x0000 app_idx 0xffff dst 0xffff
I (16654) BLE_MESH: len 4: 82030101
D (16664) BLE_MESH: net_idx 0x0000 app_idx 0xffff dst 0xffff
D (16664) BLE_MESH: len 4: 82030101
E (16674) BLE_MESH: bt_mesh_app_key_get, Failed to get AppKey
E (16674) BLE_MESH: gen_set_state, Failed to send Generic Set message (err -2)
```

It seems that the app_idx has changed from 0x0000 to 0xffff after a reset.

Alvin1Zhang, 2020-07-30:
@chegewara Thanks for the notes. @dsoon Thanks for reporting, we will look into.

Campou, 2020-07-30:
Hi @dsoon 

Please have a try with the [latest release/v4.0 branch](https://github.com/espressif/esp-idf/tree/release/v4.0). In idf v4.0.1, the `onoff_client` example doesn't support storing the app_idx into NVS.

Thanks.

dsoon, 2020-07-30:
Hi @campou,

Thanks for your message. I have cloned the main branch, but having problems running idf.py menuconfig it as I’m getting cmake errors. Please see below. It seems that the Python interpretor path is coded using C:\esp\tools\.espressif\python_env\idf4.3_py3.7_env\Scripts\python.exe and not C:/esp/tools/.espressif/python_env/idf4.3_py3.7_env/Scripts/python.exe.

Can you tell me where I can fix this?

Thanks,
Dsoon

Executing action: menuconfig
Running cmake in directory c:\users\david\esp\esp-idf\examples\bluetooth\esp_ble_mesh\ble_mesh_node\onoff_server\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized -DCCACHE_ENABLE=1 c:\users\david\esp\esp-idf\examples\bluetooth\esp_ble_mesh\ble_mesh_node\onoff_server"...
Warn about uninitialized values.
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.28.0.windows.1")
-- IDF_TARGET not set, using default target: esp32
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 8.2.0
-- The CXX compiler identification is GNU 8.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/esp/tools/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/esp/tools/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/esp/tools/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/esp/tools/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe
-- Check for working CXX compiler: C:/esp/tools/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
CMake Error at C:/Users/david/esp/esp-idf/tools/cmake/component.cmake:224 (message):
  CMake Error at build_properties.temp.cmake:2 (set):

    Syntax error in cmake code at

      C:/Users/david/esp/esp-idf/examples/bluetooth/esp_ble_mesh/ble_mesh_node/onoff_server/build/build_properties.temp.cmake:2

    when parsing string

      C:\esp\tools\.espressif\python_env\idf4.3_py3.7_env\Scripts\python.exe

    Invalid character escape '\e'.

  Call Stack (most recent call first):

    C:/Users/david/esp/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:2 (include)

Call Stack (most recent call first):
  C:/Users/david/esp/esp-idf/tools/cmake/build.cmake:405 (__component_get_requirements)
  C:/Users/david/esp/esp-idf/tools/cmake/project.cmake:395 (idf_build_process)
  CMakeLists.txt:8 (project)

-- Configuring incomplete, errors occurred!
See also "C:/Users/david/esp/esp-idf/examples/bluetooth/esp_ble_mesh/ble_mesh_node/onoff_server/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1

dsoon, 2020-07-30:
Found a work around for this issue :- https://github.com/espressif/esp-idf/issues/5512

dsoon, 2020-07-30:
@Campou 

I have tried to use the code from the main branch as you suggested, but now get the following errors when resetting the board. Any ideas?

```

I (680) gpio: GPIO[15]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3
I (710) BTDM_INIT: BT controller compile version [8fafaa4]
I (710) system_api: Base MAC address is not set
I (710) system_api: read default base MAC address from EFUSE
I (820) phy: phy_version: 4182, f1ba940, Jun  4 2020, 19:40:07, 0, 0
I (1090) EXAMPLE_NVS: Open namespace done, name "mesh_example"
I (1180) EXAMPLE: ESP_BLE_MESH_PROV_REGISTER_COMP_EVT, err_code 0
I (1180) EXAMPLE_NVS: Restore, key "onoff_client", length 6
I (1190) EXAMPLE_NVS: Restore, data: 00 00 00 00 01 03
I (1190) EXAMPLE: Restore, net_idx 0x0000, app_idx 0x0000, onoff 1, tid 0x03
I (1210) EXAMPLE: ESP_BLE_MESH_NODE_PROV_ENABLE_COMP_EVT, err_code 0
I (1210) EXAMPLE: BLE Mesh Node initialized
I (8890) BOARD: tap cb (RELEASE)
I (8900) EXAMPLE_NVS: Store, key "onoff_client", length 6
I (8900) EXAMPLE_NVS: Store, data: 00 00 00 00 00 04
E (8900) BLE_MESH: bt_mesh_model_send, Failed to get subnet
E (8900) BLE_MESH: gen_set_state, Failed to send Generic Set message (err -22)
I (8910) EXAMPLE: Generic client, event 1, error code -22, opcode is 0x8203
I (8920) EXAMPLE: ESP_BLE_MESH_GENERIC_CLIENT_SET_STATE_EVT
I (10170) BOARD: tap cb (RELEASE)
I (10170) EXAMPLE_NVS: Store, key "onoff_client", length 6
I (10170) EXAMPLE_NVS: Store, data: 00 00 00 00 01 05
E (10170) BLE_MESH: bt_mesh_model_send, Failed to get subnet
E (10180) BLE_MESH: gen_set_state, Failed to send Generic Set message (err -22)

```

closed 2020-08-04 by dsoon