espressif/esp-idf #5662: Fail esp_http_client_perform on esp-idf version 4.2

zavovi, 2020-07-29:
Hello,
please, I have some issues with my code after upgrading the esp-idf to version 4.2
I have the JSON on my server, which have authentication and certificate. I am using this code:

```
esp_http_client_config_t config =
	{
		.url = url,
		.event_handler = _http_event_handler,
		.cert_pem = server_root_cert_pem_start,
		.timeout_ms = 2000,
        .auth_type = HTTP_AUTH_TYPE_BASIC,
		.max_redirection_count = 2
	};

	client = esp_http_client_init(&config);

    esp_http_client_set_method(client, HTTP_METHOD_POST);
    esp_http_client_set_post_field(client, post_data, tcs_strlen(post_data));

	esp_err_t esp_err = esp_http_client_perform(client);
	if (esp_err != ESP_OK)
	{
		TCS_ERROR_PRINTF(LOG_CATEGORY, "Error perform HTTP request %s", esp_err_to_name(err));
		err = TCS_ERR_FWUPDATE_HTTP;
	}
```

And this worked on the version 4.1 and now it crashes with this:

```
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40090085  PS      : 0x00060133  A0      : 0x80091a24  A1      : 0x3ffe09a0
0x40090085: compare_and_set_native at C:/Users/ZAVAVOV/ZAVAVOV/10_TCS_controller/esp-idf/components/soc/include/soc/compare_set.h:28

 (inlined by) spinlock_acquire at C:/Users/ZAVAVOV/ZAVAVOV/10_TCS_controller/esp-idf/components/soc/include/soc/spinlock.h:97
 (inlined by) vPortCPUAcquireMutex at C:/Users/ZAVAVOV/ZAVAVOV/10_TCS_controller/esp-idf/components/freertos/xtensa/include/freertos
/portmacro.h:202
 (inlined by) vPortEnterCritical at C:/Users/ZAVAVOV/ZAVAVOV/10_TCS_controller/esp-idf/components/freertos/xtensa/port.c:394

A2      : 0xb33fffff  A3      : 0x0000abab  A4      : 0x00060123  A5      : 0x00060120
A6      : 0x0000cdcd  A7      : 0xb33fffff  A8      : 0x0000abab  A9      : 0x3ffe09a0
A10     : 0x00000003  A11     : 0x00060123  A12     : 0x00060120  A13     : 0x3ffe23f0
A14     : 0x258a12b6  A15     : 0x00000000  SAR     : 0x00000018  EXCCAUSE: 0x0000001d
EXCVADDR: 0xb33fffff  LBEG    : 0x4008b25a  LEND    : 0x4008b265  LCOUNT  : 0x00000000
0x4008b25a: memset at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/m
emset.S:150

0x4008b265: memset at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/m
emset.S:160

Backtrace:0x40090082:0x3ffe09a0 0x40091a21:0x3ffe09d0 0x4008493d:0x3ffe09f0 0x40084c56:0x3ffe0a10 0x4008a0d4:0x3ffe0a30 0x40192d19:0
x3ffe0a50 0x4019a755:0x3ffe0a70 0x4019a579:0x3ffe0a90 0x401b94bc:0x3ffe0ab0 0x401c0839:0x3ffe0c70 0x401c0be4:0x3ffe0ce0 0x401b9c79:0
x3ffe0d00 0x401b9cb5:0x3ffe0d20 0x401ae9a1:0x3ffe0d40 0x401ae251:0x3ffe0d60 0x401ae45f:0x3ffe0d80 0x401ae54f:0x3ffe0db0 0x401a4344:0
x3ffe0de0 0x401c6901:0x3ffe0e00 0x40127069:0x3ffe0e20 0x40127495:0x3ffe0e40 0x400d9ae1:0x3ffe0e60 0x4008ff15:0x3ffe0f90
0x40090082: compare_and_set_native at C:/Users/ZAVAVOV/ZAVAVOV/10_TCS_controller/esp-idf/components/soc/include/soc/compare_set.h:28

 (inlined by) spinlock_acquire at C:/Users/ZAVAVOV/ZAVAVOV/10_TCS_controller/esp-idf/components/soc/include/soc/spinlock.h:97
 (inlined by) vPortCPUAcquireMutex at C:/Users/ZAVAVOV/ZAVAVOV/10_TCS_controller/esp-idf/components/freertos/xtensa/include/freertos
/portmacro.h:202
 (inlined by) vPortEnterCritical at C:/Users/ZAVAVOV/ZAVAVOV/10_TCS_controller/esp-idf/components/freertos/xtensa/port.c:394

0x40091a21: multi_heap_internal_lock at C:/Users/ZAVAVOV/ZAVAVOV/10_TCS_controller/esp-idf/components/heap/multi_heap.c:382
 (inlined by) multi_heap_malloc_impl at C:/Users/ZAVAVOV/ZAVAVOV/10_TCS_controller/esp-idf/components/heap/multi_heap.c:423

0x4008493d: heap_caps_malloc at C:/Users/ZAVAVOV/ZAVAVOV/10_TCS_controller/esp-idf/components/heap/heap_caps.c:145

0x40084c56: heap_caps_calloc at C:/Users/ZAVAVOV/ZAVAVOV/10_TCS_controller/esp-idf/components/heap/heap_caps.c:395

0x4008a0d4: esp_mbedtls_mem_calloc at C:/Users/ZAVAVOV/ZAVAVOV/10_TCS_controller/esp-idf/components/mbedtls/port/esp_mem.c:25

0x40192d19: mbedtls_calloc at C:/Users/ZAVAVOV/ZAVAVOV/10_TCS_controller/esp-idf/components/mbedtls/mbedtls/library/platform.c:66

0x4019a755: gcm_ctx_alloc at C:/Users/ZAVAVOV/ZAVAVOV/10_TCS_controller/esp-idf/components/mbedtls/mbedtls/library/cipher_wrap.c:92

0x4019a579: mbedtls_cipher_setup at C:/Users/ZAVAVOV/ZAVAVOV/10_TCS_controller/esp-idf/components/mbedtls/mbedtls/library/cipher.c:1
90

0x401b94bc: mbedtls_ssl_derive_keys at C:/Users/ZAVAVOV/ZAVAVOV/10_TCS_controller/esp-idf/components/mbedtls/mbedtls/library/ssl_tls
.c:1033

0x401c0839: ssl_write_certificate_verify at C:/Users/ZAVAVOV/ZAVAVOV/10_TCS_controller/esp-idf/components/mbedtls/mbedtls/library/ss
l_cli.c:3234

0x401c0be4: mbedtls_ssl_handshake_client_step at C:/Users/ZAVAVOV/ZAVAVOV/10_TCS_controller/esp-idf/components/mbedtls/mbedtls/libra
ry/ssl_cli.c:3590

0x401b9c79: mbedtls_ssl_handshake_step at C:/Users/ZAVAVOV/ZAVAVOV/10_TCS_controller/esp-idf/components/mbedtls/mbedtls/library/ssl_
tls.c:8064

0x401b9cb5: mbedtls_ssl_handshake at C:/Users/ZAVAVOV/ZAVAVOV/10_TCS_controller/esp-idf/components/mbedtls/mbedtls/library/ssl_tls.c
:8088

0x401ae9a1: esp_mbedtls_handshake at C:/Users/ZAVAVOV/ZAVAVOV/10_TCS_controller/esp-idf/components/esp-tls/esp_tls_mbedtls.c:125

0x401ae251: esp_tls_handshake at C:/Users/ZAVAVOV/ZAVAVOV/10_TCS_controller/esp-idf/components/esp-tls/esp_tls.c:82

0x401ae45f: esp_tls_low_level_conn at C:/Users/ZAVAVOV/ZAVAVOV/10_TCS_controller/esp-idf/components/esp-tls/esp_tls.c:308

0x401ae54f: esp_tls_conn_new_sync at C:/Users/ZAVAVOV/ZAVAVOV/10_TCS_controller/esp-idf/components/esp-tls/esp_tls.c:359

0x401a4344: ssl_connect at C:/Users/ZAVAVOV/ZAVAVOV/10_TCS_controller/esp-idf/components/tcp_transport/transport_ssl.c:74

0x401c6901: esp_transport_connect at C:/Users/ZAVAVOV/ZAVAVOV/10_TCS_controller/esp-idf/components/tcp_transport/transport.c:165

0x40127069: esp_http_client_connect at C:/Users/ZAVAVOV/ZAVAVOV/10_TCS_controller/esp-idf/components/esp_http_client/esp_http_client
.c:1038

0x40127495: esp_http_client_perform at C:/Users/ZAVAVOV/ZAVAVOV/10_TCS_controller/esp-idf/components/esp_http_client/esp_http_client
.c:912

0x400d9ae1: _app_otafw_https_check at C:/Users/ZAVAVOV/ZAVAVOV/10_TCS_controller/project/main/application/app_ota_fw.c:691

0x4008ff15: vPortTaskWrapper at C:/Users/ZAVAVOV/ZAVAVOV/10_TCS_controller/esp-idf/components/freertos/xtensa/port.c:143

```

Please, could you help me?

Thank you very much.

Alvin1Zhang, 2020-07-30:
Thanks for reporting and letting us know. We will look into.

shubhamkulkarni97, 2020-08-13:
Hello @zavovi,

I tried to reproduce this issue by hosting a HTTPS server with basic auth enabled, but I was not able to reproduce this issue.

Can you please verify that you are observing this issue even with [esp_http_client](https://github.com/espressif/esp-idf/tree/master/examples/protocols/esp_http_client) example?

Can you share from where do you call `esp_http_client_perform` or possibly entire code to reproduce the issue?

Thanks,
Shubham

zavovi, 2020-08-13:
Hello @shubhamkulkarni97 ,
Thank you for your reply and that you trying reproduce the issue. 
I have this function, for reading JSON from web server:

```
static void _app_otafw_https_check(void *pvParameters)
{
	tcs_err_t err = TCS_OK;
	esp_http_client_handle_t client = NULL;
	char post_data[30];
	char url[150];

	/* Show wait window */
	if(!app_fwupdate_ctx.silence && app_fwupdate_ctx.screen.wait_window == NULL)
		app_fwupdate_ctx.screen.wait_window = app_screen_msg_wait(APP_LANG_NODE_CODE_MSG_FW_LOOKINGFOR, NULL, NULL);

	/* WiFi must be connected */
	if(!tcs_low_wifi_is_connected())
	{
		err = TCS_ERR_WIFI_NOTFOUND;
		goto ERROR;
	}

	tcs_snprintf(url, sizeof(url), "%s%s@%s", APP_OTAFW_HTTP, APP_OTAFW_WEB_AUTH, APP_OTAFW_CHECK_URL);

	esp_http_client_config_t config =
	{
		.url = url,
		.event_handler = _http_event_handler,
		.cert_pem = server_root_cert_pem_start,
        .auth_type = HTTP_AUTH_TYPE_BASIC,
		.max_redirection_count = 2
	};

	client = esp_http_client_init(&config);

	tcs_snprintf(post_data, sizeof(post_data), "UID=%s", tcs_get_app_uid());
    esp_http_client_set_method(client, HTTP_METHOD_POST);
    esp_http_client_set_post_field(client, post_data, tcs_strlen(post_data));

	esp_err_t esp_err = esp_http_client_perform(client);
	if (esp_err != ESP_OK)
	{
		TCS_ERROR_PRINTF(LOG_CATEGORY, "Error perform HTTP request %s", esp_err_to_name(err));
		err = TCS_ERR_FWUPDATE_HTTP;
	}

ERROR:
	if(TCS_FAILED(err))
	{
		/* Hide wait window */
		if(app_fwupdate_ctx.screen.wait_window)
		{
			app_screen_msg_hide(app_fwupdate_ctx.screen.wait_window);
			app_fwupdate_ctx.screen.wait_window = NULL;
		}

		if(!app_fwupdate_ctx.silence)
			app_screen_msg_error(err, app_language_get_string(APP_LANG_NODE_CODE_BUTTON_OK), NULL, TCS_TRUE, NULL, NULL);

		TCS_ERROR_PRINTF(LOG_CATEGORY, "Error when checking FW update - http check(%d)!", err);
	}

	if(client)
	{
		esp_http_client_close(client);
		esp_http_client_cleanup(client);
	}

    vTaskDelay(1000 / portTICK_PERIOD_MS);

    app_fwupdate_ctx.check_task = NULL;

    /* Close task */
    vTaskDelete( NULL );
}
```

Is is new task, initialized here:
```
app_fwupdate_ctx.silence = silence;
	xTaskCreate(&_app_otafw_https_check, "app_otafw_check", 4000, NULL, APP_TASK_PRIORITY_FWUPDATE, &app_fwupdate_ctx.check_task);
```

Here are some defines:

```
#define APP_OTAFW_HTTP 		 	"https://"
#define APP_OTAFW_CHECK_URL  	"my_web_server/neco.php"
#define APP_OTAFW_WEB_AUTH 		"USER:password"
```

And my HTTPS handler:

```
static esp_err_t _http_event_handler(esp_http_client_event_t *evt)
{
	tcs_err_t err = TCS_OK;

	if(evt && evt->event_id == HTTP_EVENT_ERROR)
	{
		err = TCS_ERR_FWUPDATE_HTTP;
	}
	else if(evt && evt->event_id == HTTP_EVENT_ON_DATA && evt->data_len > 0)
	{
		err = _app_otafw_https_parse_json(evt->data);
	}

	if(TCS_FAILED(err))
	{
		/* Hide wait window */
		if(app_fwupdate_ctx.screen.wait_window)
		{
			app_screen_msg_hide(app_fwupdate_ctx.screen.wait_window);
			app_fwupdate_ctx.screen.wait_window = NULL;
		}

		if(!app_fwupdate_ctx.silence)
			app_screen_msg_error(err, app_language_get_string(APP_LANG_NODE_CODE_BUTTON_OK), NULL, TCS_TRUE, NULL, NULL);

		TCS_ERROR_PRINTF(LOG_CATEGORY, "Error when checking FW update - event handler(%d)!", err);
	}

    return ESP_OK;
}
```

I know, there is lot of references to my own code, I hope, that it helps. I had to go back to version 4.1, because, I had more issues with 4.2 and I need release my device early. With version 4.1 it is working properly. 

Thank you very much.
Best regards,
Vilem

shubhamkulkarni97, 2020-08-13:
@zavovi Can you try suggestions mentioned in [above comment](https://github.com/espressif/esp-idf/issues/5662#issuecomment-673291788)? It will be helpful to debug the issue if you are facing the same with `esp_http_client` example as well.

Also try increasing stack size for the task from 4000 to 8192.

zavovi, 2020-08-13:
Hi @shubhamkulkarni97 ,
when I increased stack size, it seems, that it is working! I am sorry about this issue and wasting your time. 

Many thanks.

closed 2020-08-13 by zavovi