espressif/esp-idf #5663: UDP Client example issue.

nvannote, 2020-07-30:
## Environment

ESP-IDF UDP client example.

## Problem Description

Typo / Bug

File: esp-idf/examples/protocols/sockets/udp_client/main/udp_client.c
Line: 87

The existing code has the following declaration.

```c
struct sockaddr_in source_addr; // Large enough for both IPv4 or IPv6
```

Judging from the other examples, this was probably intended to be...

```c
struct sockaddr_in6 source_addr; // Large enough for both IPv4 or IPv6
```

In a perfect world, all of the combined IPv4/IPv6 examples should probably be using struct sockaddr_storage for these situations;  But I point this single example out as it is an immediate potential issue.

Best Regards

liuzfesp, 2020-07-31:
Please @freakyxue help to fix it.

freakyxue, 2020-08-13:
hi @nvannote 
This is really a problem, we should use struct SOCKADDR_ Storage to contain the IPv4 and IPv6 addresses
I will fix this issue later.

closed 2020-12-13 by espressif-bot