espressif/esp-idf #5664: ESP32 + NimBLE's Coded PHY support?

huming2207, 2020-07-30:
**Is your feature request related to a problem? Please describe.**

In file `esp_nimble_cfg.h`, it seems like the BLE 5 Coded PHY feature (the "long range mode") is [hardcoded to be 0](https://github.com/espressif/esp-idf/blob/master/components/bt/host/nimble/port/include/esp_nimble_cfg.h#L132). Is that means ESP32 does not have Coded PHY feature? Or did I misread the code and that feature is re-defined elsewhere? If it is re-defined, then how can I enable it?

**Describe the solution you'd like**

If this feature is intended to be disabled, I would like to request you to implement the low-level functionality (if any) and leave an option in `Kconfig`. Thanks!

**Describe alternatives you've considered**

ESP-NOW has similar functionality but it's not an open standard. It probably also has some overhead on 802.11 stacks and WiFi stacks.

**Additional context**

N/A

dhrishi, 2020-07-30:
@huming2207 The option you point to is the NimBLE controller option (`MYNEWT_VAL_BLE_LL_*` are all controller options) We do not use NimBLE controller and hence that option is not used.

To answer your question about Coded PHY, it is a v5.0 feature and ESP32 Bluetooth controller being v4.2, that is not supported

huming2207, 2020-07-30:
@dhrishi Thanks for your reply. Yea I didn't carefully read this article and I thought all BLE 5.0 features are implemented: https://www.espressif.com/en/news/BLE_5.0_Certification

Anyway, I will consider switching to ESP-NOW or something else.

closed 2020-07-30 by huming2207