espressif/esp-idf #5665: Infinity build loop on ESP-IDF v4.1-rc

thecashit, 2020-07-30:
Hello!

Thanks for 4.1 rc, but all my attempts to build it on OSX failed.
OSX 10.15.6 (19G73)

Steps to install ESP-IDF:
1) Clean previous ESP-IDF dir completely
2) Clean Tools dir completely
3) Delete build dir of project
4) Download archive with v4.1-rc, install and export it
5) reboot system

After that i can't build any project on new version.
All attempts end in an endless loop.
I can find in console follow messages 
"- Configuring done
- Generating done
- Build files have been written to: /Users/thecashit/Documents/LOOK.in/Firmware/blehr/build
[0/1] Re-running CMake ..."

And all starts from the beginning.

I also try to build one of examples - the behavior is the same.
I am attaching a log from the blehr example, maybe it will help
<SOME_PATH> is omitted path, its real in console.

```
idf.py build

Executing action: all (aliases: build)
Running cmake in directory /<SOME_PATH>/blehr/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized -DCCACHE_ENABLE=0 /<SOME_PATH>/blehr"...
Warn about uninitialized values.
-- Found Git: /usr/bin/git (found version "2.21.0 (Apple Git-122.2)") 
-- IDF_TARGET not set, using default target: esp32
-- The C compiler identification is GNU 8.2.0
-- The CXX compiler identification is GNU 8.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /<SOME_PATH>/Toolchain/tools/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /<SOME_PATH>/Toolchain/tools/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /<SOME_PATH>/Toolchain/tools/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /<SOME_PATH>/Toolchain/tools/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++
-- Check for working CXX compiler: /<SOME_PATH>/Toolchain/tools/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Project version: 1
-- Building ESP-IDF components for target esp32
Loading defaults file /<SOME_PATH>/blehr/sdkconfig.defaults...
-- Found PythonInterp: /<SOME_PATH>/Toolchain/tools/python_env/idf4.1_py2.7_env/bin/python (found version "2.7.16") 
-- Found Perl: /usr/bin/perl (found version "5.18.4") 
-- Adding linker script /<SOME_PATH>/blehr/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script /<SOME_PATH>/Toolchain/esp-idf/components/esp32/ld/esp32.project.ld.in
-- Adding linker script /<SOME_PATH>/Toolchain/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Adding linker script /<SOME_PATH>/Toolchain/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-time.ld
-- Adding linker script /<SOME_PATH>/Toolchain/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /<SOME_PATH>/Toolchain/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /<SOME_PATH>/Toolchain/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script /<SOME_PATH>/Toolchain/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script /<SOME_PATH>/Toolchain/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Components: app_trace app_update asio bootloader bootloader_support bt cbor coap console cxx driver efuse esp-tls esp32 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_http_client esp_http_server esp_https_ota esp_https_server esp_local_ctrl esp_netif esp_ringbuf esp_rom esp_serial_slave_link esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos heap idf_test jsmn json libsodium log lwip main mbedtls mdns mqtt newlib nghttp nvs_flash openssl partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter ulp unity vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: /<SOME_PATH>/Toolchain/esp-idf/components/app_trace /<SOME_PATH>/Toolchain/esp-idf/components/app_update /<SOME_PATH>/Toolchain/esp-idf/components/asio /<SOME_PATH>/Toolchain/esp-idf/components/bootloader /<SOME_PATH>/Toolchain/esp-idf/components/bootloader_support /<SOME_PATH>/Toolchain/esp-idf/components/bt /<SOME_PATH>/Toolchain/esp-idf/components/cbor /<SOME_PATH>/Toolchain/esp-idf/components/coap /<SOME_PATH>/Toolchain/esp-idf/components/console /<SOME_PATH>/Toolchain/esp-idf/components/cxx /<SOME_PATH>/Toolchain/esp-idf/components/driver /<SOME_PATH>/Toolchain/esp-idf/components/efuse /<SOME_PATH>/Toolchain/esp-idf/components/esp-tls /<SOME_PATH>/Toolchain/esp-idf/components/esp32 /<SOME_PATH>/Toolchain/esp-idf/components/esp_adc_cal /<SOME_PATH>/Toolchain/esp-idf/components/esp_common /<SOME_PATH>/Toolchain/esp-idf/components/esp_eth /<SOME_PATH>/Toolchain/esp-idf/components/esp_event /<SOME_PATH>/Toolchain/esp-idf/components/esp_gdbstub /<SOME_PATH>/Toolchain/esp-idf/components/esp_http_client /<SOME_PATH>/Toolchain/esp-idf/components/esp_http_server /<SOME_PATH>/Toolchain/esp-idf/components/esp_https_ota /<SOME_PATH>/Toolchain/esp-idf/components/esp_https_server /<SOME_PATH>/Toolchain/esp-idf/components/esp_local_ctrl /<SOME_PATH>/Toolchain/esp-idf/components/esp_netif /<SOME_PATH>/Toolchain/esp-idf/components/esp_ringbuf /<SOME_PATH>/Toolchain/esp-idf/components/esp_rom /<SOME_PATH>/Toolchain/esp-idf/components/esp_serial_slave_link /<SOME_PATH>/Toolchain/esp-idf/components/esp_websocket_client /<SOME_PATH>/Toolchain/esp-idf/components/esp_wifi /<SOME_PATH>/Toolchain/esp-idf/components/espcoredump /<SOME_PATH>/Toolchain/esp-idf/components/esptool_py /<SOME_PATH>/Toolchain/esp-idf/components/expat /<SOME_PATH>/Toolchain/esp-idf/components/fatfs /<SOME_PATH>/Toolchain/esp-idf/components/freemodbus /<SOME_PATH>/Toolchain/esp-idf/components/freertos /<SOME_PATH>/Toolchain/esp-idf/components/heap /<SOME_PATH>/Toolchain/esp-idf/components/idf_test /<SOME_PATH>/Toolchain/esp-idf/components/jsmn /<SOME_PATH>/Toolchain/esp-idf/components/json /<SOME_PATH>/Toolchain/esp-idf/components/libsodium /<SOME_PATH>/Toolchain/esp-idf/components/log /<SOME_PATH>/Toolchain/esp-idf/components/lwip /<SOME_PATH>/blehr/main /<SOME_PATH>/Toolchain/esp-idf/components/mbedtls /<SOME_PATH>/Toolchain/esp-idf/components/mdns /<SOME_PATH>/Toolchain/esp-idf/components/mqtt /<SOME_PATH>/Toolchain/esp-idf/components/newlib /<SOME_PATH>/Toolchain/esp-idf/components/nghttp /<SOME_PATH>/Toolchain/esp-idf/components/nvs_flash /<SOME_PATH>/Toolchain/esp-idf/components/openssl /<SOME_PATH>/Toolchain/esp-idf/components/partition_table /<SOME_PATH>/Toolchain/esp-idf/components/perfmon /<SOME_PATH>/Toolchain/esp-idf/components/protobuf-c /<SOME_PATH>/Toolchain/esp-idf/components/protocomm /<SOME_PATH>/Toolchain/esp-idf/components/pthread /<SOME_PATH>/Toolchain/esp-idf/components/sdmmc /<SOME_PATH>/Toolchain/esp-idf/components/soc /<SOME_PATH>/Toolchain/esp-idf/components/spi_flash /<SOME_PATH>/Toolchain/esp-idf/components/spiffs /<SOME_PATH>/Toolchain/esp-idf/components/tcp_transport /<SOME_PATH>/Toolchain/esp-idf/components/tcpip_adapter /<SOME_PATH>/Toolchain/esp-idf/components/ulp /<SOME_PATH>/Toolchain/esp-idf/components/unity /<SOME_PATH>/Toolchain/esp-idf/components/vfs /<SOME_PATH>/Toolchain/esp-idf/components/wear_levelling /<SOME_PATH>/Toolchain/esp-idf/components/wifi_provisioning /<SOME_PATH>/Toolchain/esp-idf/components/wpa_supplicant /<SOME_PATH>/Toolchain/esp-idf/components/xtensa
-- Configuring done
-- Generating done
[... 26 lines cut ...]
Loading defaults file /<SOME_PATH>/blehr/sdkconfig.defaults...
-- Adding linker script /<SOME_PATH>/blehr/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script /<SOME_PATH>/Toolchain/esp-idf/components/esp32/ld/esp32.project.ld.in
-- Adding linker script /<SOME_PATH>/Toolchain/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Adding linker script /<SOME_PATH>/Toolchain/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-time.ld
-- Adding linker script /<SOME_PATH>/Toolchain/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
[... cut ...] /Toolchain/esp-idf/components/wifi_provisioning /<SOME_PATH>/Toolchain/esp-idf/components/wpa_supplicant /<SOME_PATH>/Toolchain/esp-idf/components/xtensa
-- Configuring done
-- Generating done
-- Build files have been written to: /<SOME_PATH>/blehr/build
[0/1] Re-running CMake...
 

.....
ninja: error: manifest 'build.ninja' still dirty after 100 tries
```
All repeated in infinity loop

Could you please fix that or point me at right direction?
Clean, fullclean, and removing build dir didn't help

thecashit, 2020-07-30:
Works after downgrade to xtensa installed from 4.1-beta2 version

closed 2020-07-30 by thecashit

projectgus, 2020-07-31:
Hi @thecashit,

I'm glad to hear that you found a solution to the problem. We'd like to understand if there's still something here that we need to fix in ESP-IDF.

* Do I understand correctly that this works with `esp-2019r2` toolchain but not with the `esp-2020r2` toolchain? This is a little unexpected as this part of the toolchain does not have much overlap with the build system.
* Are you able to post the output from `idf.py reconfigure` when using the working toolchain, please? (should look similar to the output above.)
* Have you tried upgrading back to `esp-2020r2` again? Does the problem immediately come back?

Thanks,

Angus

thecashit, 2020-07-31:
@projectgus Thnx for answer
1. Correct, when i use esp-2020r2 it was infinity loop on configuration. esp-2019r2 works well
2. Please see attached output

```
idf.py reconfigure

Executing action: reconfigure
Running cmake in directory /<SOME_PATH>/blehr/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized -DCCACHE_ENABLE=0 /<SOME_PATH>/blehr"...
Warn about uninitialized values.
-- Unexpected file in components directory: /<SOME_PATH>/Toolchain/esp-idf/components/.DS_Store
-- Component directory /<SOME_PATH>/Toolchain/esp-idf/components/tinyusb does not contain a CMakeLists.txt file. No component will be added
CMake Warning at /<SOME_PATH>/Toolchain/esp-idf/tools/cmake/git_submodules.cmake:52 (message):
  Git submodule components/mbedtls/mbedtls is out of date.  Run 'git
  submodule update --init --recursive' to fix.
Call Stack (most recent call first):
  /<SOME_PATH>/Toolchain/esp-idf/tools/cmake/build.cmake:78 (git_submodule_check)
  /<SOME_PATH>/Toolchain/esp-idf/tools/cmake/build.cmake:160 (__build_get_idf_git_revision)
  /<SOME_PATH>/Toolchain/esp-idf/tools/cmake/idf.cmake:48 (__build_init)
  /<SOME_PATH>/Toolchain/esp-idf/tools/cmake/project.cmake:7 (include)
  CMakeLists.txt:5 (include)

-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Project version: 1
-- Building ESP-IDF components for target esp32
Loading defaults file /<SOME_PATH>/blehr/sdkconfig.defaults...
-- Adding linker script /<SOME_PATH>/blehr/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script /<SOME_PATH>/Toolchain/esp-idf/components/esp32/ld/esp32.project.ld.in
-- Adding linker script /<SOME_PATH>/Toolchain/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Adding linker script /<SOME_PATH>/Toolchain/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-time.ld
-- Adding linker script /<SOME_PATH>/Toolchain/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /<SOME_PATH>/Toolchain/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /<SOME_PATH>/Toolchain/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script /<SOME_PATH>/Toolchain/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script /<SOME_PATH>/Toolchain/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Components: app_trace app_update asio bootloader bootloader_support bt cbor coap console cxx driver efuse esp-tls esp32 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_http_client esp_http_server esp_https_ota esp_https_server esp_local_ctrl esp_netif esp_ringbuf esp_rom esp_serial_slave_link esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos heap idf_test jsmn json libsodium log lwip main mbedtls mdns mqtt newlib nghttp nvs_flash openssl partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter ulp unity vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: /<SOME_PATH>/Toolchain/esp-idf/components/app_trace /<SOME_PATH>/Toolchain/esp-idf/components/app_update /<SOME_PATH>/Toolchain/esp-idf/components/asio /<SOME_PATH>/Toolchain/esp-idf/components/bootloader /<SOME_PATH>/Toolchain/esp-idf/components/bootloader_support /<SOME_PATH>/Toolchain/esp-idf/components/bt /<SOME_PATH>/Toolchain/esp-idf/components/cbor /<SOME_PATH>/Toolchain/esp-idf/components/coap /<SOME_PATH>/Toolchain/esp-idf/components/console /<SOME_PATH>/Toolchain/esp-idf/components/cxx /<SOME_PATH>/Toolchain/esp-idf/components/driver /<SOME_PATH>/Toolchain/esp-idf/components/efuse /<SOME_PATH>/Toolchain/esp-idf/components/esp-tls /<SOME_PATH>/Toolchain/esp-idf/components/esp32 /<SOME_PATH>/Toolchain/esp-idf/components/esp_adc_cal /<SOME_PATH>/Toolchain/esp-idf/components/esp_common /<SOME_PATH>/Toolchain/esp-idf/components/esp_eth /<SOME_PATH>/Toolchain/esp-idf/components/esp_event /<SOME_PATH>/Toolchain/esp-idf/components/esp_gdbstub /<SOME_PATH>/Toolchain/esp-idf/components/esp_http_client /<SOME_PATH>/Toolchain/esp-idf/components/esp_http_server /<SOME_PATH>/Toolchain/esp-idf/components/esp_https_ota /<SOME_PATH>/Toolchain/esp-idf/components/esp_https_server /<SOME_PATH>/Toolchain/esp-idf/components/esp_local_ctrl /<SOME_PATH>/Toolchain/esp-idf/components/esp_netif /<SOME_PATH>/Toolchain/esp-idf/components/esp_ringbuf /<SOME_PATH>/Toolchain/esp-idf/components/esp_rom /<SOME_PATH>/Toolchain/esp-idf/components/esp_serial_slave_link /<SOME_PATH>/Toolchain/esp-idf/components/esp_websocket_client /<SOME_PATH>/Toolchain/esp-idf/components/esp_wifi /<SOME_PATH>/Toolchain/esp-idf/components/espcoredump /<SOME_PATH>/Toolchain/esp-idf/components/esptool_py /<SOME_PATH>/Toolchain/esp-idf/components/expat /<SOME_PATH>/Toolchain/esp-idf/components/fatfs /<SOME_PATH>/Toolchain/esp-idf/components/freemodbus /<SOME_PATH>/Toolchain/esp-idf/components/freertos /<SOME_PATH>/Toolchain/esp-idf/components/heap /<SOME_PATH>/Toolchain/esp-idf/components/idf_test /<SOME_PATH>/Toolchain/esp-idf/components/jsmn /<SOME_PATH>/Toolchain/esp-idf/components/json /<SOME_PATH>/Toolchain/esp-idf/components/libsodium /<SOME_PATH>/Toolchain/esp-idf/components/log /<SOME_PATH>/Toolchain/esp-idf/components/lwip /<SOME_PATH>/blehr/main /<SOME_PATH>/Toolchain/esp-idf/components/mbedtls /<SOME_PATH>/Toolchain/esp-idf/components/mdns /<SOME_PATH>/Toolchain/esp-idf/components/mqtt /<SOME_ [... cut ...] OME_PATH>/Toolchain/esp-idf/components/xtensa
-- Configuring done
-- Generating done
-- Build files have been written to: /<SOME_PATH>/blehr/build
```

3. That's pretty strange but i update toolchain to esp-2020r2 and now it works well.
So, i think the problem MAYBE appear in fresh install but i made fresh install again in attempts to make it works and it didn't help.

At this moment seems like problem on our side. If it will appears again i will create issue or reopen this,

Thx for fast answer

projectgus, 2020-07-31:
@thecashit Thanks for the update. Yes, please reopen if it reappears.

However I just noticed something, that you were using the archive ZIP file.

You must have downloaded the esp-idf-v4.1-rc.zip file less than two hours after it was released, and we have a known issue (#3265) that CMake can get into an infinite build loop if some files are newer than the current system time. (This seems like something in CMake, so far it's proven hard for us to fix.)

The timestamps in the esp-idf-v4.1-rc.zip are all around `2020-07-30 19:15` - ZIP format doesn't include any time zone information.

What zip extraction tool did you use on MacOS, and is it possible that the local time when v4.1-rc was not working was before this time (locally) and then it started working after this time?

thecashit, 2020-07-31:
@projectgus 
So, we have a problem with BLE & Wi-Fi coexistence([issue](https://github.com/espressif/esp-idf/issues/5427) , sadly not solved) and it its blocker for us so i downloaded 4.1-rc immediately after getting email about new version.
Out GMT offset is +5 so it it most 

It looks like this is exactly the behavior that you describe. The next attempts were about 5 hours later and could just coincide with the Toolchain downgrade

projectgus, 2020-07-31:
Hi @thecashit ,

Sorry to hear that the coexistence issue wasn't resolved, but glad to have figured this out.

Will see what we can do about creating the zip with overriden timestamps or timezone so they're in the past for everyone.

thecashit, 2020-07-31:
@projectgus Thnx for you work!

projectgus, 2020-08-06:
Quick update, I'm afraid it doesn't look like the zip tools we have available (info-zip and 7zip) give us a way to override the timestamp when making the zip.

So this is effectively a duplicate of the build system timestamp problem #3265 . Will have to look again for a solution to that issue.

lambrou, 2021-03-06:
Can someone explain how to downgrade to xtensa installed from 4.1-beta2 version? This seems to be the current workaround to the issue. I am getting this same issue when trying to build the LCD example on the ESP32-S2 Kaluga board.

projectgus, 2021-03-07:
Hi @lambrou,

If your system has the correct time set (i.e. not months in the past) then the issue mentioned here should only happen if you grab an ESP-IDF release zip file within 24 hours of when it was made, and should fix itself within 24 hours once the timestamps for the files in the zip move to the "past".

We haven't released anything in the past few days, so it seems likely the issue you're looking at is different. Could you please open a new ticket and supply as much information as possible about versions, build output, etc?

vlaskz, 2021-10-04:
> @thecashit Thanks for the update. Yes, please reopen if it reappears.
>
> However I just noticed something, that you were using the archive ZIP file.
>
> You must have downloaded the esp-idf-v4.1-rc.zip [...]

This, in sum of a new instalation problem because of the certificates, had the problem of the whole thing. Thank you for this tip. The creation of an object can't be later than its build.

mj-sakellaropoulos, 2024-11-28:
Seems like this is still an issue in latest stable when using IDF example extracted from a zip file. Simple solution is to run `touch -d "2 days ago" *` on all project files.

F1RST1, 2025-01-20:
FYI: Branch name with Chinese character in git could result in infinite build loop too.