espressif/esp-idf #5666: target_add_binary_data() does not depend of the file

galah92, 2020-07-30:
Using ESP-IDF v4.1

Updating the file embedded with `target_add_binary_data()` on disk doesn't trigger a rebuild of the relevant component.

Alvin1Zhang, 2020-07-31:
Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, ha

galah92, 2020-08-02:
Hi @Alvin1Zhang, the original issue was updated.

Alvin1Zhang, 2020-08-03:
@galah92 Thanks for the updates, we will look into.

projectgus, 2020-08-06:
Hi @galah92 ,

Thanks for providing the extra details.

I'm not able to reproduce this in a simple example under Linux with ESP-IDF v4.0.1 or on master branch. Building with `idf.py build` or `cmake --build build` both seem to pick up the changed file and trigger the necessary generation, compile & linking steps.

> the relevant binary file which is generated from file.txt and then attached is not updated.

There are four steps where this might be failing: first `file.txt` is processed into an assembly file `file.txt.S`, then assembled to `file.txt.S.obj`, then saved into the component static library, and finally linked into the executable.

I am guessing it's most likely the first step is not happening because CMake handles the rest in a fairly standard way. But to rule it out, could you please check if either/both of `file.txt.S` and `file.txt.S.obj` are being updated in the build directory?

One thing I did notice is that `file.txt.S` is generated in the top-level build directory, so there is a potential for collisions if a file with the same name is embedded into multiple components.

The only other possibilities I can think of are:

* Some unexpected difference between CMake versions. I'm using 3.16.3.
* Some other condition in the component or the project has to be true to trigger the bug.

Finally, if it doesn't contain any sensitive information then are you able to attach your generated build.ninja or Makefile?

Thanks.

Alvin1Zhang, 2020-11-12:
@galah92 Thanks for reporting, would you please help share if any updates for this issue? Thanks.

galah92, 2020-11-12:
> @galah92 Thanks for reporting, would you please help share if any updates for this issue? Thanks.

Hi, yes - thanks for raising that up. The solution was to add `enable_language(C CXX ASM)` before calling `project(..)`. I didn't get into the details, but a think a call of such is required somewhere under `idf.cmake` (or in a nested CMake file).

closed 2022-02-24 by igrr