espressif/esp-idf #5667: Installation: Win32 Python venv binary folder is 'bin' when using MSYS2 MinGW64 python

prwang, 2020-07-30:
https://github.com/espressif/esp-idf/blob/7abb8e7fc34085360cbae87c8e2a19c339228b12/tools/idf_tools.py#L876

My installation failed with the following error
```
created virtual environment CPython3.8.3.final.0-64 in 1068ms
  creator Venv(dest=C:\Users\prwang\.espressif\python_env\idf4.0_py3.8_env, clear=False, global=False, describe=CPython3Windows)
  seeder FromAppData(download=False, pip=bundle, setuptools=bundle, wheel=bundle, via=copy, app_data_dir=C:\Users\prwang\AppData\Local\pypa\virtualenv)
    added seed packages: pip==20.1.1, setuptools==49.2.0, wheel==0.34.2
  activators BashActivator,BatchActivator,FishActivator,PowerShellActivator,PythonActivator,XonshActivator
venv path is C:\Users\prwang\.espressif\python_env\idf4.0_py3.8_env\Scripts\python.exe

Installing Python packages from H:\esp-idf-v4.0.1\requirements.txt
Traceback (most recent call last):
  File "H:\esp-idf-v4.0.1\tools\idf_tools.py", line 1388, in <module>
    main(sys.argv[1:])
  File "H:\esp-idf-v4.0.1\tools\idf_tools.py", line 1384, in main
    action_func(args)
  File "H:\esp-idf-v4.0.1\tools\idf_tools.py", line 1197, in action_install_python_env
    subprocess.check_call(run_args, stdout=sys.stdout, stderr=sys.stderr)
  File "H:\msys64\mingw64\lib\python3.8\subprocess.py", line 359, in check_call
    retcode = call(*popenargs, **kwargs)
  File "H:\msys64\mingw64\lib\python3.8\subprocess.py", line 340, in call
    with Popen(*popenargs, **kwargs) as p:
  File "H:\msys64\mingw64\lib\python3.8\subprocess.py", line 854, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "H:\msys64\mingw64\lib\python3.8\subprocess.py", line 1307, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
FileNotFoundError: [WinError 2] The system cannot find the file specified
```

It turns out that the venv has the python.exe in bin instead of Scripts.
```
C:\Users\prwang\.espressif\python_env\idf4.0_py3.8_env>dir
 Volume in drive C has no label.
 Volume Serial Number is F4AF-648E

 Directory of C:\Users\prwang\.espressif\python_env\idf4.0_py3.8_env

2020/07/30  23:46    <DIR>          .
2020/07/30  23:46    <DIR>          ..
2020/07/30  23:46                42 .gitignore
2020/07/30  23:54    <DIR>          bin
2020/07/30  23:46    <DIR>          include
2020/07/30  23:46    <DIR>          lib
2020/07/30  23:52               163 pyvenv.cfg
               2 File(s)            205 bytes
               5 Dir(s)  32,183,144,448 bytes free

C:\Users\prwang\.espressif\python_env\idf4.0_py3.8_env>
```

My python is installed from MSYS2 pacman package mingw-w64-x86_64-python.

dobairoland, 2020-07-30:
Hi @prwang,

> My python is installed from MSYS2 pacman package mingw-w64-x86_64-python.

You are not supposed to do that. To the best of my knowlege, MinGW64 newer was supported.

Please install a regular Windows Python and don't try invoke from MSYS. You can also use the Windows tool installer to setup your environment. The recommended way is to use the Windows command line instead of MinGW32 since IDF v4.0.

prwang, 2020-08-02:
Thanks @dobairoland 
> Please install a regular Windows Python and don't try invoke from MSYS. You can also use the Windows tool installer to setup your environment. The recommended way is to use the Windows command line instead of MinGW32 since IDF v4.0.

The installation succeeded later after changing the line 876 to 'bin'. I usually avoid installing regular windows python because I heavily rely on MinGW64 toolchain to develop and test lightweight console apps and put them in my PATH, and the mingw64-python3 is a dependency of mingw64-gdb and also in my PATH, so it would be a disaster to have another conflicting python3 installed...  My point is that your code should adapt to a broader situation instead of dictating user how to install python.

dobairoland, 2020-08-03:
I'm happy that you've found a workaround for your problem and I'm sorry that your desired combination of tools is not supported officially.

closed 2020-08-03 by dobairoland