espressif/esp-idf #5672: ESP32 Rev3 esp_efuse_enable_rom_secure_download_mode not compiled in

sicanins, 2020-07-31:
## Environment

- Module or chip used: ESP32-WROVER-E
- IDF version (run ``git describe --tags`` to find it): v4.3-dev-472-cf056a7d0 (latest from master)
- Build System:    cmake
- Operating System:     Windows

## Problem Description

I am using the new Rev3 ESP32. Secure boot and flash encryption enabled. But there seemst to be a bug in the code, the function esp_err_t esp_efuse_enable_rom_secure_download_mode(void) is not compiled in. 
The headers are also disabled. The #ifdef SOC_SUPPORTS_SECURE_DL_MODE is false.
After forcing the header to enable, still the object is not compiled, but there is no #ifdef, I dont know how to force it.

sdkconfig security configuration:

`
CONFIG_SECURE_SIGNED_ON_BOOT=y
CONFIG_SECURE_SIGNED_ON_UPDATE=y
CONFIG_SECURE_SIGNED_APPS=y
CONFIG_SECURE_SIGNED_APPS_RSA_SCHEME=y
CONFIG_SECURE_BOOT=y
# CONFIG_SECURE_BOOT_V1_ENABLED is not set
CONFIG_SECURE_BOOT_V2_ENABLED=y
CONFIG_SECURE_BOOT_BUILD_SIGNED_BINARIES=y
CONFIG_SECURE_BOOT_SIGNING_KEY="main/xxx.pem"
# CONFIG_SECURE_BOOT_INSECURE is not set
CONFIG_SECURE_FLASH_ENC_ENABLED=y
# CONFIG_SECURE_FLASH_ENCRYPTION_MODE_DEVELOPMENT is not set
CONFIG_SECURE_FLASH_ENCRYPTION_MODE_RELEASE=y
`
Min revision is set to "Rev 3".

negativekelvin, 2020-08-02:
Only works for esp32s2

sicanins, 2020-08-02:
I see. So the problem seems to be in the Kconfig.projbuild:

```
    choice SECURE_FLASH_ENCRYPTION_MODE
        bool "Enable usage mode"
        depends on SECURE_FLASH_ENC_ENABLED
        default SECURE_FLASH_ENCRYPTION_MODE_DEVELOPMENT
        help
            By default Development mode is enabled which allows UART bootloader to perform flash encryption operations

            Select Release mode only for production or manufacturing. Once enabled you can not reflash using UART
            bootloader

            Refer to the Secure Boot section of the ESP-IDF Programmer's Guide for this version and
            https://docs.espressif.com/projects/esp-idf/en/latest/security/flash-encryption.html for details.

        config SECURE_FLASH_ENCRYPTION_MODE_DEVELOPMENT
            bool "Development(NOT SECURE)"
            select SECURE_FLASH_UART_BOOTLOADER_ALLOW_ENC

        config SECURE_FLASH_ENCRYPTION_MODE_RELEASE
            bool "Release"
           # select SECURE_ENABLE_SECURE_ROM_DL_MODE

    endchoice
```

After commenting out the `select SECURE_ENABLE_SECURE_ROM_DL_MODE` the code compiled again!

Thanks

KonstantinKondrashov, 2020-08-07:
Hi @sicanins! It was fixed in eff6a1e. Please update your IDF version.

closed 2020-08-12 by Alvin1Zhang