espressif/esp-idf #5673: 'esp_event_handler_instance_register()' not in v4.1-rc but was in v4.1-beta2

bittailor, 2020-07-31:
I started a new project a couple of weeks ago with ESP-IDF **[v4.1-beta2](https://github.com/espressif/esp-idf/releases/tag/v4.1-beta2)**. I today updated to **[v4.1-rc](https://github.com/espressif/esp-idf/releases/tag/v4.1-rc)** is it was just released yesterday.

I now noticed that my project did not compile anymore since `esp_event_handler_instance_register` is missing in **[v4.1-rc](https://github.com/espressif/esp-idf/releases/tag/v4.1-rc)** but was present in **[v4.1-beta2](https://github.com/espressif/esp-idf/releases/tag/v4.1-beta2)**. I used this function to register a event handler as the comment/documentation mentioned that this is the way forward and the _old_ `esp_event_handler_register` will be deprecated soon.

https://github.com/espressif/esp-idf/blob/cd4fa46f5372a552a67452313d246d4d09d2b39f/components/esp_event/include/esp_event.h#L119-L120

So I wonder if there was a mistake in adding it to **[v4.1-beta2](https://github.com/espressif/esp-idf/releases/tag/v4.1-beta2)** or removing it from **[v4.1-rc](https://github.com/espressif/esp-idf/releases/tag/v4.1-rc)**.

So should I 

- change my code to use the _old_ `esp_event_handler_register` as the final **[v4.1]** will not include `esp_event_handler_instance_register`, as 9b4815e3c8c9d0afec257273e1e831d5dd7445a1 was by mistake added? 
- or should I downgarde back to **[v4.1-beta2](https://github.com/espressif/esp-idf/releases/tag/v4.1-beta2)** and keep `esp_event_handler_instance_register` as **[v4.1]** will include it, as 9b4815e3c8c9d0afec257273e1e831d5dd7445a1 was by mistake removed?

Thanks for some insights.

projectgus, 2020-08-03:
Hi @bittailor ,

I'm really sorry for the inconvenience caused by this mistake.

It seems that some time after `v4.1-beta2` was tagged, a branch also named v4.1-beta2 was pushed to GitHub with the head commit cd4fa46f53. This commit is a development version of ESP-IDF v4.2, not v4.1, so it contained a number of features added in ESP-IDF v4.2. I've just now deleted this invalid branch, which should never have existed.

Simultaneously the tag `v4.1-beta2` existed and pointed to the correct commit on the `release/v4.1` branch.

You can check which version you have by running `git describe --tags --all` in the IDF_PATH directory. If it prints `v4.1-beta2`, then you have the `v4.1-beta2` pre-release. If it prints `4.2-dev-1303-gcd4fa46f53` then this was the `v4.1-beta2` branch which should never have existed.

Your options are:

* To keep the new `esp_event_handler_instance_register` function, update to the `release/v4.2` branch (currently pre-beta1).
* Change the code to use the old `esp_event_handler_register` function which will work in v4.1 (and continue to work in newer v4.x).

We're still investigating internally to see how this mistake happened. Our internal CI system (which automatically pushed the "real" v4.1-beta2 tag) did not seem to create the invalid branch. Unfortunately the GitHub Events API only returns up to 300 events, and it doesn't seem we can retrieve the exact event that shows this branch being created.

bittailor, 2020-08-03:
@projectgus, thanks for your replay. I actually had the _invalid_ branch `v4.1-beta2` and not the tag as `git clone -b v4.1-beta2 --recursive https://github.com/espressif/esp-idf.git` _favored_ the branch over the tag. For now, I updated to the release/v4.2 branch to keep my code.

closed 2020-08-03 by bittailor

projectgus, 2020-08-03:
@bittailor, thanks for understanding and sorry again for the hassle.