espressif/esp-idf #5674: wifi:bcn_timout,ap_probe_send_start after esp_http_client use?

ZweiEuro, 2020-07-31:
## Environment

- Development Kit:     ESP32 Devkit V1 -Doit (I think, it's soldered i can't look at it anymore) (here)[https://docs.espressif.com/projects/esp-idf/en/latest/esp32/hw-reference/modules-and-boards.html#esp32-wroom-32]
- Kit version -
- Module or chip used:  ESP32-WROOM-32

- IDF version (run ``git describe --tags`` to find it): framework-espidf 3.40001.200521 (4.0.1)  (PIO)
- Build System:         platformio
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0

- Operating System:   Linux

- Using an IDE?:  Vscode platformio
- Power Supply:         USB

## Problem Description
 I get this error a lot:
```
 (46748) wifi:bcn_timout,ap_probe_send_start
I (55868) wifi:bcn_timout,ap_probe_send_start
I (64978) wifi:bcn_timout,ap_probe_send_start
I (79538) wifi:bcn_timout,ap_probe_send_start
I (88938) wifi:bcn_timout,ap_probe_send_start
```

but only every other or so reboot, and it seems to only happen if i use the esp_http_client.
The rest of the system is rather big but all in all i am just using the [example](https://github.com/espressif/esp-idf/blob/master/examples/protocols/esp_http_client/main/esp_http_client_example.c), with my data put in and some output at the end.

The WiFi event_handler doesn'T seem to get informed of this printout either, and i am sure the wifi itself has no issues.

What causes this or rather what information is needed to determine the cause? I have seen some posts on the espressif forums but they all data back and seem to get a disconnect reason, which i don't.

As a result i can't use wifi at all anymore (another task runs a tcp server).

Also seems to only happen after the first restart?
I just uploaded, see the http requ go through, nothing happens, restart retry, after the http request succeeds the bcn times out.. it takes some time though.

### Expected Behavior

Not failing wifi

### Actual Behavior

Seems that esphttp_client does something weird with the wifi

### Steps to reproduce
Not 100% sure if the example code is enough to reproduce this.

I will try this later but if someone knows something about this issue that could help me narrow this down i'd greatly appreciate it

### Code to reproduce this issue

Add the {...} into the example and that's my code.
Additionally i have connect.c in, so the example connect works with my WiFi.

## Debug Logs

## Other items if possible

- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] coredump (This provides stacks of tasks.)

ZweiEuro, 2020-07-31:
After playing around a bit I wanted to know if [this](https://github.com/espressif/esp-idf/blob/release/v4.0/examples/protocols/http_request/main/http_request_example_main.c) causes the same error.
And the error was actually there as well ! 
```
␛[0;32mI (10635) mg_net_services: ... socket send success␛[0m
␛[0;32mI (10645) mg_net_services: ... set socket receiving timeout success␛[0m
D (10715) mg_net_services: DATA, HTTP/1.0 200 OK
Connection: close
X-Powered-By: PHP/7.0.33
Content-Type: application/json
Access-Control-Allow-Methods: POST
Access-Control-Max-Age: 3600
Access-Control-Allow-Headers: Content-Type, Access-Control-Allow-Headers, Authorization, X-Requested-With
Content-Length: 48
Date: Fri, 31 Jul 2020 19:47:31 GMT
Server: LiteSpeed

 {"type":"reboot","time":"1596224851","dst":"1"}␛[0m
D (10745) mg_net_services: >{"type":"reboot","time":"1596224851","dst":"1"}<␛[0m
␛[0;32mI (10755) time-mod: Time updated to : 424051␛[0m
␛[0;32mI (10755) mg_net_services: ... done reading from socket. Last read return=393 errno=0.␛[0m
I (54095) wifi:bcn_timout,ap_probe_send_start
I (63805) wifi:bcn_timout,ap_probe_send_start
I (78595) wifi:bcn_timout,ap_probe_send_start
I (87805) wifi:bcn_timout,ap_probe_send_start
I (96825) wifi:bcn_timout,ap_probe_send_start
I (106045) wifi:bcn_timout,ap_probe_send_start
```

After a bit of playing i noticed it only seems to happen after calling 'read', I have no idea what that could be causing internally.

ZweiEuro, 2020-08-01:
happens no matter if BT is enabled or not.

ZweiEuro, 2020-08-01:
Upgraded to v4.1 and the error still persists.
```
I (7081) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (7091) wifi station: WIFI_EVENT_STA_CONNECTED
I (7091) wifi station: anything
I (17131) esp_netif_handlers: sta ip: 192.168.0.14, mask: 255.255.255.0, gw: 192.168.0.1
I (17131) wifi station: IP_EVENT_STA_GOT_IP Wifi ready
I (17131) wifi station: got ip:192.168.0.14
I (17191) [TCP]: WiFi init tcp server
I (17191) [TCP]: WIFI BIT SET: 13
I (17191) [TCP]: creating TCP server
I (17201) wifi station: anything
I (17201) [UDP]: WiFi init udp server
I (17201) [TCP]: Socket created
I (17211) [UDP]: Socket created
I (17211) [TCP]: Socket bound, port 60583
I (17221) [UDP]: Socket bound, port 60582
I (17221) [TCP]: starting creating TCP server
I (17231) [UDP]: Waiting 
I (17231) [TCP]: Socket listening
E (17201) mg_net_services: service_get_time_task started 

I (17841) time-mod: Time updated to : 500734
I (17841) mg_net_services: HTTP POST Status = 200, content_length = 48
I (17841) mg_net_services: HTTP_EVENT_DISCONNECTED
I (50741) wifi:bcn_timout,ap_probe_send_start
I (60011) wifi:bcn_timout,ap_probe_send_start
```

gadget-man, 2020-08-01:
I'm seeing similar, but on 3.3-r4. One thing I've noticed on my devices is that UDP stays connected (I'm seeing the logs remotely over UDP), but TCP fails and is unable to reconnect until after a reboot. I'm not sure whether my issue is linked to esp_http_client.

ZweiEuro, 2020-08-01:
Same behaviour on 4.2... (once set up it was easy to switch to the version of release/v4.2
```
I (10149) [TCP]: Socket created
I (10149) [UDP]: Waiting 
E (10129) mg_net_services: service_get_time_task started 

I (10149) [TCP]: Socket bound, port 60583
I (10159) [TCP]: starting creating TCP server
I (10169) [TCP]: Socket listening
I (10649) time-mod: Time updated to : 501255
I (10649) mg_net_services: HTTP POST Status = 200, content_length = 48
I (10649) mg_net_services: HTTP_EVENT_DISCONNECTED
I (53859) wifi:bcn_timout,ap_probe_send_start
I (63019) wifi:bcn_timout,ap_probe_send_start
I (72139) wifi:bcn_timout,ap_probe_send_start
I (81819) wifi:bcn_timout,ap_probe_send_start
```
@gadget-man  it may be some kind of socket problem since internall esp_http_client uses the same things as tcp (read / socket / listen / bind  etc)

gadget-man, 2020-08-01:
Can you increase wifi logging to Debug to see if it gives any clues?

ZweiEuro, 2020-08-01:
I incresed it to verbose in the settings, no change sadly @gadget-man

gadget-man, 2020-08-01:
That's very odd, I'd expect you to at least see a load of `I (3496766) wifi: rsn valid: gcipher=3 ucipher=3 akm=5` before the disconnect event starts.

ZweiEuro, 2020-08-01:
I can see that a bunch before though:
```
D (7229) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

D (7229) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

D (7229) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

D (7239) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

D (7239) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

D (7249) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

D (7249) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

D (7259) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

D (7259) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

D (7269) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

D (7269) wifi:recv auth: seq=2, status=0
I (7279) wifi:state: auth -> assoc (0)
D (7279) wifi:restart connect 1s timer for assoc
D (7289) wifi:recv assoc: type=0x10
D (7289) wifi:filter: set rx policy=6
I (7289) wifi:state: assoc -> run (10)
D (7289) wifi:start 10s connect timer for 4 way handshake
D (7299) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

I (7309) wifi:connected with WiFi, aid = 1, channel 11, 40D, bssid = 80:3f:5d:f9:83:d7
I (7309) wifi:security: WPA2-PSK, phy: bgn, rssi: -67
D (7319) wifi:remove all except 80:3f:5d:f9:83:d7 from rc list
D (7319) wifi:clear blacklist
D (7319) wifi:filter: set rx policy=7
I (7329) wifi:pm start, type: 1

D (7329) wifi:Send sta connected event
D (7329) wifi:connect status 1 -> 5
D (7339) wifi:obss scan is disabled
D (7339) wifi:start obss scan: obss scan is stopped
I (7349) wifi station: WIFI_EVENT_STA_CONNECTED
I (7349) wifi station: anything
I (7399) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (10579) esp_netif_handlers: sta ip: 192.168.0.14, mask: 255.255.255.0, gw: 192.168.0.1
I (10579) wifi station: IP_EVENT_STA_GOT_IP Wifi ready
I (10579) wifi station: got ip:192.168.0.14
I (10629) [TCP]: WiFi init tcp server
I (10629) [UDP]: WiFi init udp server
I (10629) [TCP]: WIFI BIT SET: 13
I (10629) wifi station: anything
I (10629) [UDP]: Socket created
I (10639) [TCP]: creating TCP server
I (10639) [UDP]: Socket bound, port 60582
I (10649) [TCP]: Socket created
I (10649) [UDP]: Waiting 
I (10649) [TCP]: Socket bound, port 60583
E (10629) mg_net_services: service_get_time_task started 

I (10659) [TCP]: starting creating TCP server
I (10669) [TCP]: Socket listening
I (11179) time-mod: Time updated to : 501998
I (11189) mg_net_services: HTTP POST Status = 200, content_length = 48
I (11189) mg_net_services: HTTP_EVENT_DISCONNECTED
I (415699) wifi:bcn_timout,ap_probe_send_start
I (424889) wifi:bcn_timout,ap_probe_send_start
I (433899) wifi:bcn_timout,ap_probe_send_start
I (443119) wifi:bcn_timout,ap_probe_send_start
I (452129) wifi:bcn_timout,ap_probe_send_start
I (461769) wifi:bcn_timout,ap_probe_send_start
I (470869) wifi:bcn_timout,ap_probe_send_start
I (486019) wifi:bcn_timout,ap_probe_send_start
I (495749) wifi:bcn_timout,ap_probe_send_start
I (504789) wifi:bcn_timout,ap_probe_send_start
I (514079) wifi:bcn_timout,ap_probe_send_start
I (528929) wifi:bcn_timout,ap_probe_send_start
I (538039) wifi:bcn_timout,ap_probe_send_start
I (547159) wifi:bcn_timout,ap_probe_send_start
I (556789) wifi:bcn_timout,ap_probe_send_start
I (565899) wifi:bcn_timout,ap_probe_send_start
I (575009) wifi:bcn_timout,ap_probe_send_start
I (584029) wifi:bcn_timout,ap_probe_send_start
I (593869) wifi:bcn_timout,ap_probe_send_start
I (603169) wifi:bcn_timout,ap_probe_send_start
I (612189) wifi:bcn_timout,ap_probe_send_start
```

but it's even before the IP is gotten....

gadget-man, 2020-08-01:
That's very odd. I wonder if our issues are linked - I may try removing all http calls from my code and just leaving MQTT to see if it makes any difference.

Are you able to capture Wireshark packets for your connection to see if there's anything causing the reconnect at the AP?  That's what the Espressif devs here usually ask for.

ZweiEuro, 2020-08-01:
I ... I am not sure what exactly to search for as i am not very experienced with wireshark ...

gadget-man, 2020-08-01:
The recommended Macbook's built-in packet capture tool does not require a packet capture card. It can be used to capture 802.11a/b/g/n/ac packets:
https://osxdaily.com/2015/04/23/sniff-packet-capture-packet-trace-mac-os-x-wireless-diagnostics/
Or you can use the tutorial on the official Espressif website to capture packets using Wireshark
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wireshark-user-guide.html

ZweiEuro, 2020-08-01:
I got it captured, but it doesn't seem terribly interesting...

it's a .pcapng just rename it
[cap.txt](https://github.com/espressif/esp-idf/files/5011055/cap.txt)

gadget-man, 2020-08-01:
@esp-HarveyRong  are you able to have a look at this to see what's causing the disconnect?

ZweiEuro, 2020-08-01:
well yeah oyu can just open the file yourself (After renaming it) but there is nothing in it that says "obvious disconnect"

ZweiEuro, 2020-08-02:
I also found https://www.esp32.com/viewtopic.php?f=13&t=15978 which is very similar but for me
```
 auto bytesFree = heap_caps_get_free_size(MALLOC_CAP_8BIT | MALLOC_CAP_SPIRAM);
        ESP_LOGI(TAG, "memReport(): there are %zu MALLOC_CAP_8BIT | MALLOC_CAP_SPIRAM bytes free.", bytesFree);
```
literally reports 0 bytes... which i guess is bad and i need to increase something but don't know what

gadget-man, 2020-08-02:
Very odd that it's zero - to me that suggests that it's not reporting correctly, rather than there actually being zero bytes.

ZweiEuro, 2020-08-02:
> That's very odd, I'd expect you to at least see a load of `I (3496766) wifi: rsn valid: gcipher=3 ucipher=3 akm=5` before the disconnect event starts.

I tried this with more debug messages and I do get these messages as well:
```

D (368890) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

␛[0;32mI (368960) mg_net_services: Post Data raw: {"config_id":"41","firmware_version":"1"}␛[0m
D (369000) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

D (369100) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

D (369550) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

␛[0;32mI (369570) mg_net_services: DATA,  {"activeScenario":"","triggerModule":{"id":"","h_id":"","is_old":"","state":""},"checkAlexa":0,"ifttt":1,"type":"getinfo"}␛[0m
␛[0;32mI (369570) mg_net_services: Set IFTT bool ␛[0m
␛[0;32mI (369580) mg_net_services: HTTP POST Status = 200, content_length = 123␛[0m
␛[0;32mI (369580) mg_net_services: HTTP_EVENT_DISCONNECTED␛[0m
D (369610) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

D (369710) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

D (369810) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

D (369920) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

D (370020) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

D (370120) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

D (370740) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

D (370840) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

D (370940) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

D (371040) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

D (371150) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

D (371760) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

D (371860) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

D (371970) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

D (372070) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

D (372170) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

D (372680) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

D (372780) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

D (372890) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

D (372990) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

D (373090) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

D (373480) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

D (373510) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

D (373600) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

D (373710) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

D (373810) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

D (373910) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

D (374010) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

D (374120) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

D (374660) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

␛[0;32mI (376590) mg_net_services: Post Data raw: {"config_id":"41","firmware_version":"1"}␛[0m
I (380660) wifi:bcn_timout,ap_probe_send_start
D (380670) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

D (380680) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

D (380770) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

I (389770) wifi:bcn_timout,ap_probe_send_start
D (389780) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

D (389790) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

␛[0;31mE (394990) HTTP_CLIENT: Connection failed, sock < 0␛[0m
␛[0;31mE (394990) mg_net_services: HTTP POST request failed: ERROR␛[0m
␛[0;32mI (394990) mg_net_services: HTTP POST Status = 0, content_length = 0␛[0m
␛[0;32mI (394990) mg_net_services: HTTP_EVENT_DISCONNECTED␛[0m
I (398790) wifi:bcn_timout,ap_probe_send_start
D (398790) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

D (398800) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

D (398800) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

D (398900) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

␛[0;32mI (402000) mg_net_services: Post Data raw: {"config_id":"41","firmware_version":"1"}␛[0m
I (407900) wifi:bcn_timout,ap_probe_send_start
D (407910) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

D (407910) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

I (416910) wifi:bcn_timout,ap_probe_send_start
D (416920) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

D (416920) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

D (416930) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

D (417020) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

␛[0;31mE (420490) HTTP_CLIENT: Connection failed, sock < 0␛[0m
␛[0;31mE (420490) mg_net_services: HTTP POST request failed: ERROR␛[0m
␛[0;32mI (420490) mg_net_services: HTTP POST Status = 0, content_length = 0␛[0m
␛[0;32mI (420500) mg_net_services: HTTP_EVENT_DISCONNECTED␛[0m
I (426020) wifi:bcn_timout,ap_probe_send_start
D (426030) wifi:rsn valid: gcipher=3 ucipher=3 akm=5

D (426040) wifi:rsn valid: gcipher=3 ucipher=3 akm=5
```

It just gets printed no matter if the network is broken or not

gadget-man, 2020-08-02:
Hopefully this gives the team at Espressif some ideas as to what might be causing the error.

ZweiEuro, 2020-08-02:
Hopefully

Alvin1Zhang, 2020-08-03:
Thanks for reporting, and sharing the detailed updates, we will look into.

esp-HarveyRong, 2020-08-04:
Usually beacon timeout problems can be debugged as follows:

**1. Insufficient memory leads to beacon timeout**

- An Error like "esf_buf: t=8, l=beacon_len, ..." will appear in the log. The memory is not enough, you can check the heap size when you receive the disconnect event
**2. AP did not send beacon**

- You can check it through packet capture or comparison test. For hardware problems, you can find a development board with no problems to do a comparative test to troubleshoot

**3. The country code set by the AP is inconsistent with the current channel.** 

- For example, the country is set to US, that is, the channel range is 1~11, but the AP's channel is 13.
**4. Other questions**

liuzfesp, 2020-08-04:
Hi @ZweiEuro could you attached your sdkconfig here (better put it as attachment)?

gadget-man, 2020-08-04:
@esp-HarveyRong It doesn't look from the logging that @ZweiEuro provided that there are memory problems, channel is 11 as per the logs. Are you able to review the packet capture details provided to see if it highlights anything to your expert eyes to see if the error is (2) from your list above, if we've already determined it's not (1) or (3).

ZweiEuro, 2020-08-04:
sdk defaults
```

CONFIG_BT_ENABLED=y
CONFIG_BTDM_CTRL_MODE_BLE_ONLY=y
CONFIG_BTDM_CTRL_MODE_BR_EDR_ONLY=n
CONFIG_BTDM_CTRL_MODE_BTDM=n
CONFIG_FREERTOS_USE_TRACE_FACILITY=y
CONFIG_FREERTOS_GENERATE_RUN_TIME_STATS=y
CONFIG_BT_BTC_TASK_STACK_SIZE=4096
CONFIG_ESP32_PHY_CALIBRATION_AND_DATA_STORAGE=y
CONFIG_BOOTLOADER_LOG_LEVEL=5
CONFIG_ESP32_PHY_INIT_DATA_ERROR=y
CONFIG_ESP32_WIFI_STATIC_RX_BUFFER_NUM=10

CONFIG_ESP32_WIFI_DEBUG_LOG_ENABLE=y
CONFIG_ESP32_WIFI_DEBUG_LOG_DEBUG=y
# CONFIG_ESP32_WIFI_DEBUG_LOG_VERBOSE is not set
CONFIG_ESP32_WIFI_DEBUG_LOG_MODULE_ALL=y
# CONFIG_ESP32_WIFI_DEBUG_LOG_MODULE_WIFI is not set
# CONFIG_ESP32_WIFI_DEBUG_LOG_MODULE_COEX is not set
# CONFIG_ESP32_WIFI_DEBUG_LOG_MODULE_MESH is not set
CONFIG_ESP32_WIFI_DEBUG_LOG_SUBMODULE=y
CONFIG_ESP32_WIFI_DEBUG_LOG_SUBMODULE_ALL=y

```

esp-HarveyRong, 2020-08-07:
Hi @ZweiEuro,

Sorry, I did not see any information about the wireless package in the cap.txt you provided, such as: beacon, etc. Can you provide a new wireless package?  Including beacon etc.

[image]

ZweiEuro, 2020-08-07:
I am out for a week but will do this as soon as i can

ZweiEuro, 2020-08-19:
@esp-HarveyRong 
I've tried capturing the beacons like this:
https://wiki.wireshark.org/Wi-Fi
but I don't seem to receive any.  (filter:  wlan.fc.type_subtype == 0x08 )

I have the network from the hub on WPA2 and am connected to it via my PC and my phone and the esp while it's breaking up.
I also get the same results if my laptop is not connected to any network

esp-HarveyRong, 2020-08-25:
@ZweiEuro,

Can you attach a completed Log?
 
When the Rssi value of the AP connected to the device is low, the device will not receive the beacon frame when the environment is complicated, and `bcn_timout,ap_probe_send_start`

ZweiEuro, 2020-08-27:
You mean a complete log from the esp32 ?

ZweiEuro, 2020-10-04:
I have tried this with my pone hotspot as well, and it just sends out 'wifi:bcn_timeout, ap_probe_Send_start' and internet stops 
I at least wish it would have some event for this so i can start reconnecting

HarveyRong-Esp, 2020-10-13:
Hi @ZweiEuro,

Sorry for the delayed reply, I rested for a few days.

Usually, within 6s, if the STA doesn't  receive a beacon frame, it will appear `bcn_timeout, ap_probe_Send_start`. When this behavior occurs, the STA will try to send a `Probe Request` 5 times. If the router replies with a `Probe Response`, the connection will continue. If the `Probe response` from the router is not received, the connection will be disconnected and an event will be thrown. If there is only `bcn_timeout, ap_probe_Send_start` log, there is no event indicating that the STA has received the `Probe Response` and continues to maintain the connection. This is normal behavior.

Can you call the `esp_get_free_internal_heap_size` API to print the internal memory when this issue occurs?

ZweiEuro, 2020-10-13:
I've already tested, and there is loads of heap left. 
for some reason it won't run unless every variable in the handler is either malloced or declared static, making it very hard to program.
So I am guessing i am doing something else wrong about how i need to allocate data.

ZweiEuro, 2020-10-13:
Some output from this code running, it seems to run fine for a while, heap seems to be stable and cleared correctly.
The bigger functions that would build lists later aren't even called in this case.
https://gist.github.com/ZweiEuro/743602b57faf3b13c2712019a8b6a706

HarveyRong-Esp, 2020-10-23:
Generally speaking, WiFi needs to apply for internal memory to receive the beacon. When the internal memory is low, the application will not be able to receive the beacon. When the device does not receive a beacon within 6s, a beacon timeout will appear, causing the WiFi to drop. .
The ʻesp_get_free_internal_heap_size` API can get the current available internal memory size.

ZweiEuro, 2020-11-04:
https://github.com/espressif/esp-idf/issues/5939#issuecomment-722009896

closed 2020-11-04 by ZweiEuro

HarveyRong-Esp, 2020-12-03:
Hi @ZweiEuro, I tried to use the example to reproduce it, but unfortunately we did not show the `bcn_timeout, ap_probe_Send_start` behavior. I think this should be related to the behavior of the router. Can you tell me the AP model used when the issue occurs ? I will continue to try to reproduce.

ZweiEuro, 2020-12-03:
@HarveyRong-Esp 
Just so you are not stumpling in the dark, have you read:
> [#5939 (comment)](https://github.com/espressif/esp-idf/issues/5939#issuecomment-722009896)

The issue is unrelated to wifi, but happened because i stressed te internal file system beyond capacity

gadget-man, 2020-12-04:
We recently had a customer who was seeing disconnects every 6-7 minutes. The only thing that would allow the ESP32 to reconnect was a reboot. However, the customer advised that they moved their LG TV which was on the same network across to a different WiFi network and that solved the problem straight away. Unfortunately I don't have any packet capturing from their network to find out what the specific issue was, but it seems that this other device was creating an issue.

HarveyRong-Esp, 2020-12-08:
Hi @gadget-man, can you create a new issue ticket and describe the issue you encountered in detail? If possible, please provide wireless packet capture