espressif/esp-idf #5677: RTC_CNTL_BROWN_OUT_RST_ENA is not enabled when "Hardware brownout detect & reset" is enabled on menuconfig

JosuGZ, 2020-07-31:
# Environment

- IDF version: v3.2.3-264-g0c7abb897

# Problem Description

I've noticed that RTC_CNTL_BROWN_OUT_RST_ENA is not active when "Hardware brownout detect & reset" is enabled on menuconfig; by printing the reg contents. I think this is misleading.

Output:

    RTC_CNTL_BROWN_OUT_REG: 0x78028000

Code from the IDF:

```
void esp_brownout_init()
{
    REG_WRITE(RTC_CNTL_BROWN_OUT_REG,
            RTC_CNTL_BROWN_OUT_ENA /* Enable BOD */
            | RTC_CNTL_BROWN_OUT_PD_RF_ENA /* Automatically power down RF */
            /* Reset timeout must be set to >1 even if BOR feature is not used */
            | (2 << RTC_CNTL_BROWN_OUT_RST_WAIT_S)
            | (BROWNOUT_DET_LVL << RTC_CNTL_DBROWN_OUT_THRES_S));

    ESP_ERROR_CHECK( rtc_isr_register(rtc_brownout_isr_handler, NULL, RTC_CNTL_BROWN_OUT_INT_ENA_M) );

    REG_SET_BIT(RTC_CNTL_INT_ENA_REG, RTC_CNTL_BROWN_OUT_INT_ENA_M);
}
```

JosuGZ, 2020-07-31:
Now I'm seeing that there is a reset from the ISR, but isn't this slower? It attempts to log to the console.

igrr, 2020-08-01:
@JosuGZ Thanks for pointing this out. The approach with a software interrupt was necessary on ESP32 revision 0, where the hardware BOR was not functional. We will modify this code so that for revision 1 and above, hardware BOR would be used.

JosuGZ, 2020-08-01:
Could that help to fix/mitigate the problem of the chip getting stuck if voltage drops and there is no voltage supervisor?

I'm also not sure about what "close flash" means: RTC_CNTL_BROWN_OUT_CLOSE_FLASH_ENA

https://m.youtube.com/watch?v=cKDv0aN67BY

igrr, 2020-08-03:
@JosuGZ, no, not really, on the ESP32 the brownout detector does not replace an external voltage supervisor. The reason is that the brownout detector is disabled by default after reset, so successive dips in supply voltage can still put the chip into a stuck state, with brownout detector disabled.

The main thing that the BOD on the ESP32 does help with, is detecting situations when the power supply does not provide sufficient current when Wi-Fi is transmitting. On the ESP8266 this brownout would often result in CPU incorrectly executing an instruction, making it look as if the program was crashing.

Alvin1Zhang, 2021-08-24:
Fix with https://github.com/espressif/esp-idf/commit/55970fe3a231d4dfc1dd553728f7ac6c3dc986d4, feel free to reopen.

closed 2021-08-24 by Alvin1Zhang