espressif/esp-idf #5679: interrupt issue

ravikiranb36, 2020-08-01:
`static void IRAM_ATTR rs_485_task(void* arg){
    printf("zsdas");
    
}
void add_rs485_interrupt(){
    static TaskHandle_t handle;
    //RS485 receive interrupt
    gpio_install_isr_service(0);
    gpio_set_intr_type(RS485_RO_PIN, GPIO_INTR_ANYEDGE);
    gpio_isr_handler_add(RS485_RO_PIN, rs_485_task, (void*) &handle);
}`

It gives bellow error when interrupt occured
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400d7fcc: ffffffff ffffffff ffffffff
Core 0 register dump:
PC      : 0x400d7fd0  PS      : 0x00060031  A0      : 0x80080fdc  A1      : 0x3ffb0690
A2      : 0x3ffb2158  A3      : 0x00000000  A4      : 0x800d5337  A5      : 0x3ffb5650
A6      : 0x00000000  A7      : 0x3ffb5670  A8      : 0x80080ec1  A9      : 0x3ffb5590
A10     : 0x3f40013c  A11     : 0x00060023  A12     : 0x80087244  A13     : 0x3ffb5640
A14     : 0x00000003  A15     : 0x00060d23  SAR     : 0x00000017  EXCCAUSE: 0x00000000
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000
Core 0 was running in ISR context:
EPC1    : 0x400d7fd0  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x00000000

ELF file SHA256: 0290b44da0a6ee2d

Backtrace: 0x400d7fcd:0x3ffb0690 0x40080fd9:0x3ffb06b0 0x400820a9:0x3ffb06d0 0x400e53b3:0x3ffb56a0 0x400d561b:0x3ffb56c0 0x4008782e:0x3ffb56e0 0x40085d11:0x3ffb5700

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:7096
load:0x40078000,len:14076
load:0x40080400,len:4304
entry 0x400806e8

felmue, 2020-08-02:
Hi ravikiranb36
I could be wrong, but I don't think printf() is allowed in the interrupt service function. Does it still crash when you comment out the printf() statement? You could try to only set a flag in the interrupt service function and then check that flag in the main task / loop.
Cheers
Felix

Alvin1Zhang, 2020-08-03:
@ravikiranb36 Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

ravikiranb36, 2020-08-07:
Thank you after removed printf() it worked

Alvin1Zhang, 2020-08-07:
Thanks for the updates and glad to hear it worked, feel free to reopen if the issue still happens. Thanks.

closed 2020-08-07 by Alvin1Zhang