espressif/esp-idf #5680: nvs_flash.h: No such file or directory

benas1989, 2020-08-01:
## Environment

- Development Kit:ESP32-Wrover-Kit
- Kit version DevKitC
- Module or chip used:ESP32-WROVER-B
- IDF version: 4.2
- Build System: idf.py
- Compiler version xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0
- Operating System: macOS
- Using an IDE?: Visual studio code
- Power Supply: USB
## Problem Description

Hello,
I try create my first wifi component. I have to add nvs_flash.h  but then i got error:
nvs_flash.h: No such file or directory
All other libraries work well.(esp_wifi.h, freertos/semphr.h and others)
### Expected Behavior
Build no errors.

### Actual Behavior
Got error nvs_flash.h: No such file or directory

### Code to reproduce this issue

wifi.c
```cpp
#include "wifi.h"
#include "esp_log.h"
#include "esp_wifi.h"
#include "nvs_flash.h"
#include "string.h"
#include "freertos/semphr.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_event.h"
#include <stdio.h>

#define TAG "WIFI"

void start_wifi_sta()
{
    ESP_LOGI(TAG, "Starting wifi STA.");
}

void start_wifi_ap()
{
    ESP_LOGI(TAG, "Starting wifi AP");
}

void init_wifi(void *params)
{
    ESP_LOGI(TAG, "Iniatializing wifi driver...");
}

```

wifi.h
```cpp
#ifndef _WIFI_H
#define _WIFI_H
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_event.h"

extern xSemaphoreHandle init_wifi_semaphore;

void init_wifi(void *params);

#endif
```

CMakeLists.txt
```cpp
idf_component_register(
    SRCS "wifi.c"
    INCLUDE_DIRS "."
)
```
Directory structure
[image]

benas1989, 2020-08-01:
ok :) i found problem.
The list of common components is: freertos, newlib, heap, log, soc, esp_rom, esp_common, xtensa, cxx.

But esp_wifi.h is not listed and does not need to be added to CMakeLists. 
How to know what to add and what not?

laukik-hase, 2020-08-02:
@benas1989 You can refer the corresponding ESP-IDF example's CMakeLists.txt as a reference, for the feature you are using. E.g., I2C, WiFi, ADC, etc.

closed 2020-08-02 by benas1989

yureddy, 2023-07-02:
> 

Hi @benas1989, Could you please explain a little more about how you resolved your problem of missing "fatal error: nvs_flash.h: No such file or directory "

KaeLL, 2023-07-02:
Please ... check if you've specified all component dependencies with 'idf_component_register(REQUIRES ...)'. For more information run `idf.py docs -sp api-guides/build-system.html`.

yureddy, 2023-07-02:
thanks, I am able to resolve it.

LubomirBogdanov, 2024-02-26:
I believe adding "nvs_flash" to the set(COMPONENT_REQUIRES ... nvs_flash ...) list in CMakeLists.txt fixes this problem.