espressif/esp-idf #5685: unnamed prototyped parameters not allowed when body is present

benas1989, 2020-08-02:
## Environment

Development Kit:ESP32-Wrover-Kit
Kit version DevKitC
Module or chip used:ESP32-WROVER-B
IDF version: 4.2
Build System: idf.py
Compiler version xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0
Operating System: macOS
Using an IDE?: Visual studio code
Power Supply: USB

## Problem Description

I try to write function to do write i2c registers.I get error:
unnamed prototyped parameters not allowed when body is present

When I change the name of the variable register no longer indicates an error.
Is the word register reserved in esp-id or c?

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

esp_err_t i2c_write(uint8_t address, uint8_t register, uint8_t data)
{
    return ESP_OK;   
}
```

igrr, 2020-08-03:
> Is the word register reserved in esp-id or c?

You are right, `register` is a keyword in C language: https://en.cppreference.com/w/c/keyword/register.

Alvin1Zhang, 2020-08-04:
@benas1989 Thanks for reporting, would you please help check whether your issue has been resolved? Thanks.

closed 2020-08-04 by benas1989