espressif/esp-idf #5690: CAN can driver can not be installed if using -DCMAKE_BUILD_TYPE=Release

ablagodirov, 2020-08-03:
## Environment

- Development Kit:      [ESP32-Wrover-Kit]
- Kit version (for DevKitC): [v4]
- Module or chip used:  [ESP32-WROVER-B]
- IDF version (run ``git describe --tags`` to find it):
    v4.1-rc
- Build System:         [CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0
- Operating System:     [Linux|macOS]
- Using an IDE?: [Yes Clion]
- Power Supply:         [USB]

## Problem Description

After upgrading to IDF 4.1-rc we faced problem with using CAN
Call to can_driver_install fails. And this is reproducible on IDF example 

### Expected Behavior

can_driver_install Success

### Actual Behavior

Getting crash with following traceback:

```
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7016
load:0x40078000,len:13072
ho 0 tail 12 room 4
load:0x40080400,len:3896
0x40080400: _init at ??:?

entry 0x40080688
I (31) boot: ESP-IDF v4.1-rc 2nd stage bootloader
I (31) boot: compile time 19:14:35
I (31) boot: chip revision: 1
I (34) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 2MB
I (55) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot: End of partition table
I (97) boot_comm: chip revision: 1, min. application chip revision: 0
I (105) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x0336c ( 13164) map
I (119) esp_image: segment 1: paddr=0x00013394 vaddr=0x3ffb0000 size=0x01f4c (  8012) load
I (126) esp_image: segment 2: paddr=0x000152e8 vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at /Users/alblagodirov/IDF/4.1/esp-idf/components/freertos/xtensa_vectors.S:1778

I (132) esp_image: segment 3: paddr=0x000156f4 vaddr=0x40080404 size=0x08d74 ( 36212) load
I (156) esp_image: segment 4: paddr=0x0001e470 vaddr=0x00000000 size=0x01ba8 (  7080) 
I (159) esp_image: segment 5: paddr=0x00020020 vaddr=0x400d0020 size=0x13c80 ( 81024) map
0x400d0020: _stext at ??:?

I (198) boot: Loaded app from partition at offset 0x10000
I (198) boot: Disabling RNG early entropy source...
I (199) cpu_start: Pro cpu up.
I (202) cpu_start: Application information:
I (207) cpu_start: Project name:     can_network_master
I (213) cpu_start: App version:      1
I (218) cpu_start: Compile time:     Aug  3 2020 19:15:02
I (224) cpu_start: ELF file SHA256:  beaab218149f77f0...
I (230) cpu_start: ESP-IDF:          v4.1-rc
I (235) cpu_start: Starting app cpu, entry point is 0x40080e78
0x40080e78: call_start_cpu1 at /Users/alblagodirov/IDF/4.1/esp-idf/components/esp32/cpu_start.c:271

I (226) cpu_start: App cpu up.
I (245) heap_init: Initializing. RAM available for dynamic allocation:
I (252) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (258) heap_init: At 3FFB2760 len 0002D8A0 (182 KiB): DRAM
I (264) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (271) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (277) heap_init: At 40089178 len 00016E88 (91 KiB): IRAM
I (283) cpu_start: Pro cpu start user code
I (301) spi_flash: detected chip: generic
I (302) spi_flash: flash io: dio
W (302) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (312) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400d6307  PS      : 0x00060e33  A0      : 0x800d44ee  A1      : 0x3ffb4570  
0x400d6307: can_ll_set_bus_timing at /Users/alblagodirov/IDF/4.1/esp-idf/components/soc/esp32/include/hal/can_ll.h:370
 (inlined by) can_hal_configure at /Users/alblagodirov/IDF/4.1/esp-idf/components/soc/src/hal/can_hal.c:55

A2      : 0x3ffb271c  A3      : 0x0000003f  A4      : 0x3f4010ac  A5      : 0x000000e7  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x00000008  A9      : 0x00000000  
A10     : 0x00000003  A11     : 0x00000010  A12     : 0x00000080  A13     : 0x00060e20  
A14     : 0x3ffb656c  A15     : 0x3f400010  SAR     : 0x00000014  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000018  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

ELF file SHA256: beaab218149f77f0

Backtrace: 0x400d6304:0x3ffb4570 0x400d44eb:0x3ffb45a0 0x400d3af1:0x3ffb45c0 0x400d0a12:0x3ffb45f0 0x40084231:0x3ffb4610
0x400d6304: can_ll_set_bus_timing at /Users/alblagodirov/IDF/4.1/esp-idf/components/soc/esp32/include/hal/can_ll.h:370
 (inlined by) can_hal_configure at /Users/alblagodirov/IDF/4.1/esp-idf/components/soc/src/hal/can_hal.c:55

0x400d44eb: can_driver_install at /Users/alblagodirov/IDF/4.1/esp-idf/components/driver/can.c:386

0x400d3af1: app_main at /Users/alblagodirov/work/can_network_master-4.1/main/can_network_example_master_main.c:221

0x400d0a12: main_task at /Users/alblagodirov/IDF/4.1/esp-idf/components/esp32/cpu_start.c:565 (discriminator 1)

0x40084231: vPortTaskWrapper at /Users/alblagodirov/IDF/4.1/esp-idf/components/freertos/port.c:143

```

### Steps to reproduce

1. Install IDF 4.1 and corresponding toolchain 
2. Complile can_network_master with Cmake and -DCMAKE_BUILD_TYPE=Release
3. Flash and run example

In attachments logs with and without -DCMAKE_BUILD_TYPE=Release
[cmake4.1-rel.txt](https://github.com/espressif/esp-idf/files/5017136/cmake4.1-rel.txt)
[cmake-norel.txt](https://github.com/espressif/esp-idf/files/5017139/cmake-norel.txt)

dshil, 2020-08-06:
@0xjakob @igrr (sorry for mentioning directly) could you please take a look or ping somebody who is aware of `CAN` part?

Dazza0, 2020-08-06:
@dshil Sorry for the delay. I'll be looking into this shortly.

dshil, 2020-08-06:
Hi @Dazza0, no problem and big thanks!

Dazza0, 2020-08-06:
@dshil @ablagodirov It seems like this the same issue described in [this forum post](https://esp32.com/viewtopic.php?f=13&t=16699). We already have a fix merged into our internal master branch with backports to v4.2 and v4.1. I'll link the commit ID once Github syncs with our internal repo. 

For now, a quick fix would be to move all functions in `can.c` out of their assert calls. For example

```c
assert(can_hal_init(&can_context));
```
to
```c
can_hal_init(&can_context);
```

ablagodirov, 2020-08-06:
@Dazza0 Thank you! We will wait for fix delivery!

Dazza0, 2020-08-10:
Closed by bd72a9ab2bd9843d082e1c04d83174ac5c92a8ab

closed 2020-08-10 by Dazza0