espressif/esp-idf #5692: ESP32_S2 with light sleep enabled reboots

felmue, 2020-08-03:
## Environment

- Development Kit: [The ESP32-S2-Kaluga-1]
- Kit version: [v1.2]
- Module or chip used:  [ESP32-S2-WROVER]
- IDF version (run ``git describe --tags`` to find it): v4.3-dev-771-gc77c4ccf6
- Build System: [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0
- Operating System: [macOS]
- Using an IDE?: [No]
- Power Supply: [USB]

## Problem Description

Running example 'esp-idf/examples/wifi/power_save' the ESP32_S2 boots normally, connects to Wi-Fi AP successfully, but then reboot moments after (Less than 2 seconds) with a reset cause rst:0x10 (RTCWDT_RTC_RST). This reboot loop is continuous.

### Expected Behavior

ESP32_S2 to enter into light sleep mode

### Actual Behavior

ESP32_S2 reboots with cuase rst:0x10 (RTCWDT_RTC_RST),boot:0x8 (SPI_FAST_FLASH_BOOT). This reboot loop is continuous.

### Steps to reproduce

1. cd ~/esp
2. cp -r esp-idf/examples/wifi/power_save .
3. cd power_save
4. idf.py set-target esp32s2
5. idf.py menuconfig (only change SSID, Password, flash size: 4MB)
6. idf.py build
7. idf.py flash monitor

### Code to reproduce this issue

The line that causes the issue

```
#if CONFIG_FREERTOS_USE_TICKLESS_IDLE
      .light_sleep_enable = true
#endif
```

Changing the following prevents the ESP32_S2 from rebooting, but will not allow low power mode.

```
      .light_sleep_enable = false
```

## Debug Logs

First start after flashing

```
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x8
load:0x3ffe6108,len:0x17e8
load:0x4004c000,len:0x14c4
load:0x40050000,len:0x2158
entry 0x4004c2d8
I (47) boot: ESP-IDF v4.3-dev-771-gc77c4ccf6 2nd stage bootloader
I (47) boot: compile time 23:42:30
I (47) boot: chip revision: 0
I (50) boot.esp32s2: SPI Speed      : 80MHz
I (55) boot.esp32s2: SPI Mode       : DIO
[... 129 lines cut ...]
I (469) wifi:Init management frame dynamic rx buffer num: 32
I (469) wifi:Init management short buffer num: 32
I (469) wifi:Init dynamic tx buffer num: 32
I (479) wifi:Init static rx buffer size: 1600
I (479) wifi:Init static rx buffer num: 10
I (489) wifi:Init dynamic rx buffer num: 32
I (619) phy: phy_version: 603, 72dfd77, Jul  7 2020, 19:57:05, 0, 2
I (619) wifi:enable tsf
I (619) wifi:mode : sta (7c:df:a1:00:75:3e)
I (619) power_save: esp_wifi_set_ps().
I (619) wifi:Set ps type: 1

I (739) wifi:new:<11,0>, old:<1,0>, ap:<255,255>, sta:<11,0>, prof:1
I (749) wifi:state: init -> auth (b0)
I (749) wifi:state: auth -> assoc (0)
I (749) wifi:state: assoc -> run (10)
I (769) wifi:connected with ZZ1Z, aid = 14, channel 11, BW20, bssid = 20:4e:7f:c7:55:1a
I (769) wifi:security: WPA2-PSK, phy: bgn, rssi: -52
I (779) wifi:pm start, type: 1

I (789) wifi:AP's beacon interval = 102400 us, DTIM period = 1

```

Thank you for looking into this. I appreciate it.

Alvin1Zhang, 2020-08-04:
@felmue Thanks for reporting, we will look into.

thorvald, 2020-08-12:
I'm suffering from the exact same issue, but on the ESP32-S2-Saola-1MI with the ESP32-S2-WROOM-I. Anything we can do to help resolve this?

felmue, 2020-09-17:
Good day

this issue has been resolved for me with commit: https://github.com/espressif/esp-idf/commit/b5e9047e47ad2d32242c599afee2e0a1050a5a6d pushed about 2 days ago.

_Fix reboot or crash when enable lightsleep on esp32s2
Closes WIFI-2248_

Thank you very much. I appreciate it.

Cheers
Felix

Alvin1Zhang, 2020-09-17:
@felmue Thanks for reporting and good to hear the fix was resolved. Would you please help close the issue? Thanks.

closed 2020-09-17 by felmue