espressif/esp-idf #5695: IDF4.2 : PPPOS_Client : issue in "example_send_message_text()" : "dte->send_wait()" : UART Rx "frozen"

mmggbe, 2020-08-04:
Dear,

I have an issue with the example PPPOS_Client when sending a SMS with a SIM800l
Api version is : ESP-IDF v4.2-dev-1995-g636b964c8-dirty

In ppos_client_main(), the issue is in the function "esp_err_t example_send_message_text(), with an extrace of the code below
 
    /* send message */
    char command[MODEM_SMS_MAX_LENGTH] = {0};
    int length = snprintf(command, MODEM_SMS_MAX_LENGTH, "AT+CMGS=\"%s\"\r", phone_num);
    /* set phone number and wait for "> " */
    **dte->send_wait(dte, command, length, "\r\n>", MODEM_PROMPT_TIMEOUT_MS);**
    /* end with CTRL+Z */
    snprintf(command, MODEM_SMS_MAX_LENGTH, "%s\x1A", text);
    dce->handle_line = example_handle_cmgs;
    if (dte->send_cmd(dte, command, MODEM_COMMAND_TIMEOUT_SMS_MS) != ESP_OK) {
        ESP_LOGE(TAG, "send command failed");
        goto err;
    }

The function "dte->send_wait(dte, command, length, "\r\n>", MODEM_PROMPT_TIMEOUT_MS)" enter in Timeout and seems to never listen to UART

This function calls the function : uart_read_bytes() in which I added some debugging at some key steps
{
    **ESP_LOGI(UART_TAG, "Enter uart_read_bytes");  //MaGe**
....
    if(xSemaphoreTake(p_uart_obj[uart_num]->rx_mux,(portTickType)ticks_to_wait) != pdTRUE) {
        return -1;
    }
    **ESP_LOGI(UART_TAG, "Wait to receive char"); //MaGe**
    while(length) {
        if(p_uart_obj[uart_num]->rx_cur_remain == 0) {
            data = (uint8_t*) xRingbufferReceive(p_uart_obj[uart_num]->rx_ring_buf, &size, (portTickType) ticks_to_wait);

            if(data) {
                **ESP_LOGI(UART_TAG, "Data received; char nbr= %d", size); //MaGe**
                p_uart_obj[uart_num]->rx_head_ptr = data;
                p_uart_obj[uart_num]->rx_ptr = data;
                p_uart_obj[uart_num]->rx_cur_remain = size;
            } else {

Te trace is : 
I (14345) pppos_example: default_handle: OK

I (14345) esp-modem: UART cmd WAIT: AT+CMGS="+32475123456"
I (14345) uart: Enter uart_read_bytes
I (14355) uart: Wait to receive char
E (19355) esp-modem: esp_modem_dte_send_wait(284): wait prompt [
>] timeout
I (19355) esp-modem: UART cmd: Welcome to ESP32!
I (22045) uart: Enter uart_read_bytes
I (22045) uart: Wait to receive char
I (22045) uart: Data received; char nbr= 23
I (22045) pppos_example: handle_cmgs: 
> 

E (22045) esp-modem: esp_dte_handle_line(91): handle line failed
W (22055) pppos_example: Unknow line received: 
> 

send_wait() waits for the Modem answer ">CRLF"... but never get it until the time-out is expired (I put the time-out to 5 sec to be sure)
It seems the UART is inactive and becomes active again after the dte->send_wait() returned an error since the modem answer comes a bit later in the trace.

Any idea where to look?

Thanks

Marc

PS : the modem SIM800L init sequence below (missing in the) example, its important to wait 10 for full initialisation otherwise the "AT+COPS?\r" (get_operator_name) returns an error

#if CONFIG_EXAMPLE_MODEM_DEVICE_SIM800

	#define LOW 				0
	#define HIGH				1

	#define MODEM_RST            5
	#define MODEM_PWKEY          4
	#define MODEM_POWER_ON       23

	gpio_pad_select_gpio(MODEM_RST);
	gpio_set_direction(MODEM_RST, GPIO_MODE_OUTPUT);

	gpio_pad_select_gpio(MODEM_PWKEY);
	gpio_set_direction(MODEM_PWKEY, GPIO_MODE_OUTPUT);

	gpio_pad_select_gpio(MODEM_POWER_ON);
	gpio_set_direction(MODEM_POWER_ON, GPIO_MODE_OUTPUT);

	// restart module : needed in case of watchdog
	gpio_set_level(MODEM_RST, LOW);
	gpio_set_level(MODEM_PWKEY, LOW);
	gpio_set_level(MODEM_POWER_ON, LOW);		// Turn off the Modem LDO external power supply
	ESP_LOGI(TAG, "Module GSM un-powered");
	vTaskDelay(500/portTICK_PERIOD_MS);			// must be between 0,5 and 1s

	gpio_set_level(MODEM_RST, HIGH);
	gpio_set_level(MODEM_PWKEY, HIGH);
	gpio_set_level(MODEM_POWER_ON, HIGH);		// Turn on the Modem LDO external power supply first
	ESP_LOGI(TAG, "Module GSM powered");
	vTaskDelay(500/portTICK_PERIOD_MS);			// must be 1,5s up before the raise of PWKEY

	gpio_set_level(MODEM_PWKEY, LOW);
	ESP_LOGI(TAG, "Module GSM PWKEY LOW");
	vTaskDelay(1100/portTICK_PERIOD_MS);		// Pull down PWRKEY for more than 1 second

	gpio_set_level(MODEM_PWKEY, HIGH);
	ESP_LOGI(TAG, "Module GSM PWKEY HIGH");
	vTaskDelay(10000/portTICK_PERIOD_MS);		// modem is ready 10s after pull down of PWKEY

    modem_dce_t *dce = sim800_init(dte);

Alvin1Zhang, 2020-08-05:
Thanks for the report, we will look into.

mmggbe, 2020-08-12:
Dear,

I found the issue : it is linked to the fact that the UART interrupt is not active. The follwing code fixes it (add uart_enable_rx_intr) and there is no Timeout anymore on the UART and the answer is correctly handled.

```
static esp_err_t esp_modem_dte_send_wait(modem_dte_t *dte, const char *data, uint32_t length,
        const char *prompt, uint32_t timeout)
{
    MODEM_CHECK(data, "data is NULL", err_param);
    MODEM_CHECK(prompt, "prompt is NULL", err_param);
    esp_modem_dte_t *esp_dte = __containerof(dte, esp_modem_dte_t, parent);
    // We'd better disable pattern detection here for a moment in case prompt string contains the pattern character
    uart_disable_pattern_det_intr(esp_dte->uart_port);
    uart_enable_rx_intr(esp_dte->uart_port);			//MaGe : needed since deactivated per default in esp_modem_dte_init
    ESP_LOGI(MODEM_TAG, "UART cmd WAIT: %s", data);
    MODEM_CHECK(uart_write_bytes(esp_dte->uart_port, data, length) >= 0, "uart write bytes failed", err_write);
  ...
}
```

BUT, there is now another issue in teh code below : it seems that the Modem is in the wrong PPP mode. Still to be investigated

```
static void esp_handle_uart_data(esp_modem_dte_t *esp_dte)
{
    if (esp_dte->parent.dce->mode != MODEM_PPP_MODE) {
        ESP_LOGE(MODEM_TAG, "Error: Got data event in PPP mode");
        /* pattern detection mode -> ignore date event on uart
         * (should never happen, but if it does, we could still
         * read the valid data once pattern detect event fired) */
        return;
    }
....
}
```

mmggbe, 2020-08-15:
Just to summarize the issue :

 uart_write_bytes() doesn't work anymore when the interrupts are disabled (uart_disable_pattern_det_intr() or uart_disable_rx_intr()) which prevent to use uart_write_bytes() in a polling mode

david-cermak, 2020-08-20:
Hi @mmggbe 

Thanks for the report and the proposed solution.
Could you please test if sending a text message works okay if you:
* comment out this [hang-up command](https://github.com/espressif/esp-idf/blob/master/examples/protocols/pppos_client/components/modem/src/esp_modem.c#L519)
* and allow the data event in command mode removing [this return](https://github.com/espressif/esp-idf/blob/master/examples/protocols/pppos_client/components/modem/src/esp_modem.c#L162)

My impression is that the main reason for the failing `send_wait()` was the previous hang-up. Also could you please share the exact version of the IDF you use?

Anyway, there's definitely a problem with sending a text message which needs fixing. Thanks for your help!

closed 2021-01-05 by espressif-bot