espressif/esp-idf #5698: 无法连接wpa2企业加密ESP-TLS

kaedewang, 2020-08-05:
## Environment

- Development Kit:      ESP32-DevKitC
- Module or chip used:  ESP32-WROOM-32D
- IDF version (run ``git describe --tags`` to find it):
    // v4.3-dev-771-gc77c4ccf6
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0
- Operating System:     wsl
- Using an IDE?: No
- Power Supply:         USB

## Problem Description
ESP32使用wpa2_enterprise例程，无法连接wpa2企业加密EAP-TLS，电脑和手机连接无问题
路由器使用ASUSwrt merlin，安装本地FreeRADIUS Version 3.0.21
根据其他issue
ESP32设置了
config EXAMPLE_VALIDATE_SERVER_CERT=n
config WPA_MBEDTLS_CRYPTO=n

FreeRADIUS设置了
disable_tlsv1_2 = no
disable_tlsv1_1 = yes
disable_tlsv1 = yes
tls_min_version = "1.2"
tls_max_version = "1.2"

FreeRADIUS Debug信息如下
(0) Received Access-Request Id 106 from 127.0.0.1:48424 to 127.0.0.1:1812 length 233
(0)   User-Name = "example@espressif.com"
(0)   NAS-IP-Address = 127.0.0.1
(0)   Called-Station-Id = "D4-5D-64-4D-7B-C8:ASUS"
(0)   NAS-Port-Type = Wireless-802.11
(0)   Service-Type = Framed-User
(0)   Calling-Station-Id = "C8-2B-96-B8-6A-DC"
(0)   Connect-Info = "CONNECT 0Mbps 802.11g"
(0)   Acct-Session-Id = "AA3CF7806C39EB64"
(0)   Acct-Multi-Session-Id = "0ABFDC1446AD11C1"
(0)   Attr-186 = 0x000fac04
(0)   Attr-187 = 0x000fac04
(0)   Attr-188 = 0x000fac01
(0)   Framed-MTU = 1400
(0)   EAP-Message = 0x02af001a016578616d706c65406573707265737369662e636f6d
(0)   Message-Authenticator = 0x032d3ecaee12ccb4bfe5826034ada2bc
(0) # Executing section authorize from file /opt/etc/freeradius3/sites-enabled/default
(0)   authorize {
(0)     policy filter_username {
(0)       if (&User-Name) {
(0)       if (&User-Name)  -> TRUE
(0)       if (&User-Name)  {
(0)         if (&User-Name =~ / /) {
(0)         if (&User-Name =~ / /)  -> FALSE
(0)         if (&User-Name =~ /@[^@]*@/ ) {
(0)         if (&User-Name =~ /@[^@]*@/ )  -> FALSE
(0)         if (&User-Name =~ /\.\./ ) {
(0)         if (&User-Name =~ /\.\./ )  -> FALSE
(0)         if ((&User-Name =~ /@/) && (&User-Name !~ /@(.+)\.(.+)$/))  {
(0)         if ((&User-Name =~ /@/) && (&User-Name !~ /@(.+)\.(.+)$/))   -> FALSE
(0)         if (&User-Name =~ /\.$/)  {
(0)         if (&User-Name =~ /\.$/)   -> FALSE
(0)         if (&User-Name =~ /@\./)  {
(0)         if (&User-Name =~ /@\./)   -> FALSE
(0)       } # if (&User-Name)  = notfound
(0)     } # policy filter_username = notfound
(0)     [preprocess] = ok
(0)     [chap] = noop
(0)     [mschap] = noop
(0)     [digest] = noop
(0) suffix: Checking for suffix after "@"
[... 271 lines cut ...]
												
									   
							  
(3)     [files] = noop
(3)     [expiration] = noop
(3)     [logintime] = noop
Not doing PAP as Auth-Type is already set.
(3)     [pap] = noop
(3)   } # authorize = updated
(3) Found Auth-Type = eap
(3) # Executing group from file /opt/etc/freeradius3/sites-enabled/default
(3)   authenticate {
(3) eap: Expiring EAP session with state 0xc607166ac4b51b6d
(3) eap: Finished EAP session with state 0xc607166ac4b51b6d
(3) eap: Previous EAP request found for state 0xc607166ac4b51b6d, released from the list
**(3) eap: Peer sent packet with method EAP NAK (3)
(3) eap: Peer NAK'd our request for TLS (13) with a request for TLS (13), skipping...
(3) eap: WARNING: !!! We requested to use an EAP type as normal.
(3) eap: WARNING: !!! The supplicant rejected that, and requested to use the same EAP type.
(3) eap: WARNING: !!!     i.e. the supplicant said 'I don't like X, please use X instead.
(3) eap: WARNING: !!! The supplicant software is broken and does not work properly.
(3) eap: WARNING: !!! Please upgrade it to software that works.**
(3) eap: Found mutually acceptable type PEAP (25)
(3) eap: Calling submodule eap_peap to process data
(3) eap_peap: Initiating new TLS session

sagb2015, 2020-08-11:
@kaedewang - Can you help us with the default EAP method set on Freeradius side, typically available in mods_available/eap file? Also what is the value of  EAP_METHOD on esp32 in menuconfig?

Alvin1Zhang, 2020-08-14:
@kaedewang Thanks for reporting, would you please help share if any updates per the questions by @sagb2015 ? Thanks.

kaedewang, 2020-08-18:
@sagb2015 @Alvin1Zhang 
ESP32 EAP method:TLS
EAP setting is almost default.
Only change
 default_eap_type = tls
 private_key_file = ${certdir}/server.key
 certificate_file = ${certdir}/server.pem
 disable_tlsv1_2 = no
 disable_tlsv1_1 = yes
 disable_tlsv1 = yes
 tls_min_version = "1.2"
 tls_max_version = "1.2"
And delete other unused EAP method
 
eap {
        #  Invoke the default supported EAP type when
        #  EAP-Identity response is received.
        #
        #  The incoming EAP messages DO NOT specify which EAP
        #  type they will be using, so it MUST be set here.
        #
        #  For now, only one default EAP type may be used at a time.
        #
        #  If the EAP-Type attribute is set by another module,
        #  then that EAP type takes precedence over the
        #  default type configured here.
        #
        default_eap_type = tls

        #  A list is maintained to correlate EAP-Response
        #  packets with EAP-Request packets.  After a
        #  configurable length of time, entries in the list
        #  expire, and are deleted.
        #
        timer_expire = 60

        #  There are many EAP types, but the server has support
        #  for only a limited subset.  If the server receives
        #  a request for an EAP type it does not support, then
        #  it normally rejects the request.  By setting this
        #  configuration to "yes", you can tell the server to
        #  instead keep processing the request.  Another module
        #  MUST then be configured to proxy the request to
        #  another RADIUS server which supports that EAP type.
        #
        #  If another module is NOT configured to handle the
        #  request, then the request will still end up being
        #  rejected.
        #
        ignore_unknown_eap_types = no

        # Cisco AP1230B firmware 12.2(13)JA1 has a bug.  When given
        # a User-Name attribute in an Access-Accept, it copies one
        # more byte than it should.
        #
        # We can work around it by configurably adding an extra
        # zero byte.
        #
        cisco_accounting_username_bug = no

        #  Help prevent DoS attacks by limiting the number of
        #  sessions that the server is tracking.  For simplicity,
        #  this is taken from the "max_requests" directive in
        #  radiusd.conf.
        #
        max_sessions = ${max_requests}

        ############################################################
        #
        #  Supported EAP-types
[... 650 lines cut ...]
                #  client.  This worked, but it had issues
                #  when the cached password was wrong.  The
                #  server *should* send "E=691 R=0" to the
                #  client, which tells it to prompt the user
                #  for a new password.
                #
                #  The default is to behave as in 2.1.10 and
                #  earlier, which is known to work.  If you
                #  set "send_error = yes", then the error
                #  message will be sent back to the client.
                #  This *may* help some clients work better,
                #  but *may* also cause other clients to stop
                #  working.
                #
        #       send_error = no

                #  Server identifier to send back in the challenge.
                #  This should generally be the host name of the
                #  RADIUS server.  Or, some information to uniquely
                #  identify it.
                #
        #       identity = "FreeRADIUS"
        }

}

kaedewang, 2020-08-28:
I successed to connect wpa2-enterprise by EAP-TLS.
The problem is client key. I used client key with password. After clearing password it is OK.

closed 2020-08-28 by kaedewang

AxelLin, 2020-08-28:
@kaedewang 
The password protected client key should also work.

    ESP_ERROR_CHECK( esp_wifi_sta_wpa2_ent_set_cert_key(client_crt_start, client_crt_bytes,\
                client_key_start, client_key_bytes, NULL, 0) );

The latest 2 parameters is private_key_passwd and private_key_passwd_len.
You can check again by set correct password and password length.

dabbinavo, 2021-06-10:
> The password protected client key should also work.
>
> ```
> ESP_ERROR_CHECK( esp_wifi_sta_wpa2_ent_set_cert_key(client_crt_start, client_crt_bytes,\
>             client_key_start, client_key_bytes, [...]

@AxelLin, sorry for this noob question, but i'm struggling with the understanding of that function.
From my understanding, the following is a common use-case:
* Considering a network configured as WPA2-EAP TLS
* Then, the IT administrator will give to me a **ca-certificate** for server certification and a **client certificate** that is used by my client to authenticate towards the server
* The ca-certificate has to be set with `esp_wifi_sta_wpa2_ent_set_ca_cert()`
* The client certificate has to be set with `esp_wifi_sta_wpa2_ent_set_cert_key()`

But what to pass to `client_key_start`, if i only have gotten the client certificate (.crt) file but no "client key" (that is also called `private_key` in the [header file](https://github.com/espressif/esp-idf/blob/1d7068e4be430edd92bb63f2d922036dcf5c3cc1/components/wpa_supplicant/include/esp_supplicant/esp_wpa2.h#L174))?

sagb2015, 2021-06-11:
@dabbinavo In PKI, both parties (client and server in this case) generate <private, public> key pair and get certificates from trusted CA (which ties public key to the identity). Each side is supposed to know and keep their private keys private. The arg client_key_start refers to private key of the ESP32 device. Please refer to [this](https://en.wikipedia.org/wiki/Public_key_infrastructure) article for more information on PKI and also contact your IT admin about generating the key pair and getting client-certificate signed by CA trusted by your organisation.

dabbinavo, 2021-06-11:
@sagb2015 Thanks for the information! It wasn't clear to me, that the user certificate (client certificate) in .p12 format (PKCS#12) distributed by the IT admin contains both, the certificate and the private key. So to use the ESP-IDF API `esp_wifi_sta_wpa2_ent_set_cert_key(...);`, the certificate and the private key first have to be extracted from the .p12 file and then passed to the arguments `client_cert` and `private_key` (and `private_key_passwd` if a password had to be specified during the extraction process for private key), right?

If so, does some ESP-IDF library running on the ESP32 already provide any functions to extract the certificate and private key from a .p12 file or do I have to include/use a separate library for this (e.g. openssl which [can do that](https://github.com/openssl/openssl/blob/master/demos/pkcs12/pkread.c))?

AxelLin, 2021-08-30:
> @kaedewang
> The password protected client key should also work.
>
> ```
> ESP_ERROR_CHECK( esp_wifi_sta_wpa2_ent_set_cert_key(client_crt_start, client_crt_bytes,\
>             client_key_start, [...]

I take it back.

While testing with v4.3-356-g48ae2309fd9c, I found the same issue as reported by https://github.com/espressif/esp-idf/issues/5698#issuecomment-682328164
(It does not work if the client key needs password, but it works if the key does not need password).

I test again with v4.4-dev-2825-gb63ec47238fd, the same symptom: It does not work if the client key needs password.

@sagb2015 @kapilkedawat  (I think this issue should be re-open)

kapilkedawat, 2021-08-30:
> @sagb2015 Thanks for the information! It wasn't clear to me, that the user certificate (client certificate) in .p12 format (PKCS#12) distributed by the IT admin contains both, the certificate and [...]

Hi @dabbinavo , you can use openssl commands to extract certificate and key.

> While testing with v4.3-356-g48ae2309fd9c, I found the same issue as reported by #5698 (comment)
> (It does not work if the client key needs password, but it works if the key does not need password).

> I test again with v4.4-dev-2825-gb63ec47238fd, the same symptom: It does not work if the client key needs password.

Hi @AxelLin , can you please share the certificates that you have used? If not possible, can you please share the cert type?

AxelLin, 2021-08-30:
@kapilkedawat 
My testing ca, client certificate, key as below.
(For password protected key, the password: whatever)
ca.pem + client.crt + client_without_pwd.key works.
ca.pem + client.crt + client_with_pwd.key + passord(whatever) fails.

[ca.pem.txt](https://github.com/espressif/esp-idf/files/7074565/ca.pem.txt)
[client.crt.txt](https://github.com/espressif/esp-idf/files/7074566/client.crt.txt)
[client_without_pwd.key.txt](https://github.com/espressif/esp-idf/files/7074567/client_without_pwd.key.txt)
[client_with_pwd.key.txt](https://github.com/espressif/esp-idf/files/7074568/client_with_pwd.key.txt)

AxelLin, 2021-08-30:
@kapilkedawat 

I test again, this time it works. (No idea why if fails in my previous test).

With CONFIG_WPA_DEBUG_PRINT=y, it works but it shows some error messages as below (FYI).
(The same error messages on both v4.3 and master)

I (3941) wifi:new:<11,2>, old:<1,0>, ap:<255,255>, sta:<11,2>, prof:1
I (4958) wifi:state: init -> auth (b0)
I (4999) wifi:state: auth -> assoc (0)
I (5047) wifi:state: assoc -> run (10)
I (5048) wpa: wpa2_task prio:2, stack:6656

I (17123) wpa: SSL: Need 1914 bytes more input data
I (20117) wpa: SSL: Need 920 bytes more input data
E (22611) net_utils: wifi_sta_up timeout 20000 ms, bits=0x0
I (24224) wpa: application data is null, adding one byte for ack
I (24233) wpa: >>>>>wpa2 FINISH

E (25240) wpa: RSN: PMKSA cache entry found - PMKID - hexdump(len=16):
E (25241) wpa: 92 98 05 cd 6c 63 ed cf 7b fb e2 44 7d d3 51 ae

I (26259) wifi:connected with IGS-MESH, aid = 2, channel 11, 40D, bssid = 18:a6:f7:ab:ca:66
I (26260) wifi:security: WPA2-ENT, phy: bgn, rssi: -49
I (26262) wifi:pm start, type: 1

W (26265) wifi:Next TBTT incorrect! last beacon:1318892, offset:383, next beacon:11763692, beacon interval:102400, dtim period:0, dtim count:0, listen interval:3, now:23657636

AxelLin, 2021-09-06:
@kapilkedawat
(I'm not sure how it works in my previous test. I guess it was mistakely tested with non password protected key)

Test again with v4.4-dev-2928-gd5f58ab13551 and v4.3.1 debug build, I got below error message on the device:
E (407759) wpa: mbedtls_pk_parse_keyfile returned -0x2e80
E (407759) wpa: Failed to set client pki context
E (407760) wpa: Failed to set client configurations
E (407765) wpa: failed to create ssl handle
I (407770) wpa: TLS: Failed to set TLS connection parameters
I (407776) wpa: EAP-TLS: Failed to initialize SSL.

AxelLin, 2021-09-06:
> @kapilkedawat
> (I'm not sure how it works in my previous test. I guess it was mistakely tested with non password protected key)
>
> Test again with v4.4-dev-2928-gd5f58ab13551 and v4.3.1 debug build, I [...]

Hi @kapilkedawat 
Above error is because mbedtls_cipher_info_from_type() failed which is because the
cipher_alg is MBEDTLS_CIPHER_DES_EDE3_CBC and it is not supported.
Setting CONFIG_MBEDTLS_DES_C=y avoid the error.

Then I can connect to the AP.
Note, even though it works in the end, it fails so may times before successfully join the AP. (I got many disconnect with reason 204, 205).

AxelLin, 2021-09-07:
@kapilkedawat 

I tried diffent -keypbe setting to generate the key via openssl, e.g. PBE-SHA1-RC2-128

```
# openssl pkcs12 -info -in client.p12 -noout
Enter Import Password:
MAC: sha1, Iteration 2048
MAC length: 20, salt length: 8
PKCS7 Encrypted data: pbeWithSHA1And128BitRC2-CBC, Iteration 2048
Certificate bag
PKCS7 Data
Shrouded Keybag: pbeWithSHA1And128BitRC2-CBC, Iteration 2048
```

But I still got below error if CONFIG_MBEDTLS_DES_C=n
**E (111835) wpa: mbedtls_pk_parse_keyfile returned -0x2e80**

Is it necessary to set CONFIG_MBEDTLS_DES_C=y for password protected key?
I'm wondering if I missed something? ( How to test password protected key with CONFIG_MBEDTLS_DES_C=n?)

kapilkedawat, 2021-09-07:
Hi @AxelLin ,

You can try with AES encryptions(128-192-256 bits). I would advise against enabling DES since it is vulnerable to sweet32 attacks.

AxelLin, 2021-09-07:
> Hi @AxelLin ,
> 
> You can try with AES encryptions(128-192-256 bits). I would advise against enabling DES since it is vulnerable to sweet32 attacks.

Could you show me the example to generate AES encryption key? Thanks.

nishanth-radja, 2021-09-07:
Hi @AxelLin ,
Following are the steps to generate the encrypted Crts.
1.openssl genrsa -aes256 -out server.key 4096 {AES encryption of the 	private key}
2. openssl req -new -key server.key -out server.csr -config <required_config>
3. openssl ca -batch -keyfile ca.key -cert ca.pem -in server.csr -key 'whatever' -out server.crt -config <required_config>

For more detailed steps,Pls refer this Doc. [Link](https://drive.google.com/drive/folders/18yMdvHV9ncG595p3sxF7zNnf5OZTejAb?usp=sharing)

weisen2021, 2022-09-21:
@AxelLin 
Hello, I cannot connect to the RADIUS server while debugging EAP-TLS. I am using IDF4.4, the routine is WPA2_Enterprise and the chip is ESP32-D0WDQ6-V3.

I replaced wpa2_ca.pem, wpa2_client.crt, wpa2_client.ket, and wpa2_client.pem in my routine with files in the RADIUS server.

I don't use CA authentication, so I disabled it in the ConfigurAction Editor and selected TLS for the EAP Method.

When I use a Huawei mobile phone to connect to RADIUS, a user certificate is selected. The user certificate is a client. p12 file provided by my colleagues who set up RADIUS.

But I don't understand how ESP32 uses the client.p12 file.

dabbinavo, 2022-09-24:
> When I use a Huawei mobile phone to connect to RADIUS, a user certificate is selected. The user certificate is a client. p12 file provided by my colleagues who set up RADIUS.
> 
> But I don't understand how ESP32 uses the client.p12 file.

If you want to use the .p12 file on ESP32, you have to extract the information from it. You can do this for example with openSSL:
https://github.com/openssl/openssl/blob/master/demos/pkcs12/pkread.c

AxelLin, 2022-09-30:
> Hi @AxelLin , Following are the steps to generate the encrypted Crts. 1.openssl genrsa -aes256 -out server.key 4096 {AES encryption of the private key} 

Hi @nishanth-radja 
Above command does not work, I need to add additional "-traditional" to make it generate PKCS#1 format instead of default PKCS#8 format.

Without "-traditional" , I got E (6105) wpa: mbedtls_pk_parse_keyfile returned -0x2e80.
```
pk_parse_key_pkcs8_encrypted_der()
    mbedtls_pkcs5_pbes2()
        mbedtls_oid_get_cipher_alg() returns MBEDTLS_ERR_PKCS5_BAD_INPUT_DATA
```
Just wondering Is it possible to support PKCS#8 format?

weisen2021, 2022-09-30:
@dabbinavo  
Thank you for your answer. Let me look at the routine first。