espressif/esp-idf #5699: Problem with cpp and V4.3

rinripper, 2020-08-05:
I'm trying to compilate my project, it have a cpp file.

the error the esp-idf cmd says is below

I read in previuos ticket it have something to do with inline fuction in esp_attr.h but I check it and it looks fine

`In file included from C:/Users/mauro/Desktop/GitHut/esp-idf/components/soc/src/esp32/include/hal/cpu_ll.h:18,
                 from C:/Users/mauro/Desktop/GitHut/esp-idf/components/soc/include/hal/cpu_hal.h:23,
                 from C:/Users/mauro/Desktop/GitHut/esp-idf/components/soc/include/soc/cpu.h:27,
                 from C:/Users/mauro/Desktop/GitHut/esp-idf/components/soc/include/soc/spinlock.h:19,
                 from C:/Users/mauro/Desktop/GitHut/esp-idf/components/freertos/xtensa/include/freertos/portmacro.h:85,
                 from C:/Users/mauro/Desktop/GitHut/esp-idf/components/freertos/include/freertos/portable.h:95,
                 from C:/Users/mauro/Desktop/GitHut/esp-idf/components/freertos/include/freertos/FreeRTOS.h:107,
                 from ../main/my_rmt.h:49,
                 from ../main/my_rmt.cpp:33:
C:/Users/mauro/Desktop/GitHut/esp-idf/components/xtensa/include/esp_attr.h:110:34: error: conflicting declaration of C function 'constexpr timer_intr_t operator~(timer_intr_t)'
 FORCE_INLINE_ATTR constexpr TYPE operator~ (TYPE a) { return (TYPE)~(INT_TYPE)a; } \
                                  ^~~~~~~~
C:/Users/mauro/Desktop/GitHut/esp-idf/components/xtensa/include/esp_attr.h:122:29: note: in expansion of macro 'FLAG_ATTR_IMPL'
 #define FLAG_ATTR_U32(TYPE) FLAG_ATTR_IMPL(TYPE, uint32_t)
                             ^~~~~~~~~~~~~~
C:/Users/mauro/Desktop/GitHut/esp-idf/components/xtensa/include/esp_attr.h:123:19: note: in expansion of macro 'FLAG_ATTR_U32'
 #define FLAG_ATTR FLAG_ATTR_U32
                   ^~~~~~~~~~~~~
C:/Users/mauro/Desktop/GitHut/esp-idf/components/soc/include/hal/timer_types.h:72:1: note: in expansion of macro 'FLAG_ATTR'
 FLAG_ATTR(timer_intr_t)
 ^~~~~~~~~
C:/Users/mauro/Desktop/GitHut/esp-idf/components/xtensa/include/esp_attr.h:110:34: note: previous declaration 'constexpr spi_event_t operator~(spi_event_t)'
 FORCE_INLINE_ATTR constexpr TYPE operator~ (TYPE a) { return (TYPE)~(INT_TYPE)a; } \
                                  ^~~~~~~~
C:/Users/mauro/Desktop/GitHut/esp-idf/components/xtensa/include/esp_attr.h:122:29: note: in expansion of macro 'FLAG_ATTR_IMPL'
 #define FLAG_ATTR_U32(TYPE) FLAG_ATTR_IMPL(TYPE, uint32_t)
                             ^~~~~~~~~~~~~~
C:/Users/mauro/Desktop/GitHut/esp-idf/components/xtensa/include/esp_attr.h:123:19: note: in expansion of macro 'FLAG_ATTR_U32'
 #define FLAG_ATTR FLAG_ATTR_U32
                   ^~~~~~~~~~~~~
C:/Users/mauro/Desktop/GitHut/esp-idf/components/soc/include/hal/spi_types.h:42:1: note: in expansion of macro 'FLAG_ATTR'
 FLAG_ATTR(spi_event_t)
 ^~~~~~~~~
C:/Users/mauro/Desktop/GitHut/esp-idf/components/xtensa/include/esp_attr.h:111:34: error: conflicting declaration of C function 'constexpr timer_intr_t operator|(timer_intr_t, timer_intr_t)'
 FORCE_INLINE_ATTR constexpr TYPE operator| (TYPE a, TYPE b) { return (TYPE)((INT_TYPE)a | (INT_TYPE)b); } \
                                  ^~~~~~~~
C:/Users/mauro/Desktop/GitHut/esp-idf/components/xtensa/include/esp_attr.h:122:29: note: in expansion of macro 'FLAG_ATTR_IMPL'
 #define FLAG_ATTR_U32(TYPE) FLAG_ATTR_IMPL(TYPE, uint32_t)
                             ^~~~~~~~~~~~~~
C:/Users/mauro/Desktop/GitHut/esp-idf/components/xtensa/include/esp_attr.h:123:19: note: in expansion of macro 'FLAG_ATTR_U32'
 #define FLAG_ATTR FLAG_ATTR_U32
                   ^~~~~~~~~~~~~
C:/Users/mauro/Desktop/GitHut/esp-idf/components/soc/include/hal/timer_types.h:72:1: note: in expansion of macro 'FLAG_ATTR'
 FLAG_ATTR(timer_intr_t)
 ^~~~~~~~~
C:/Users/mauro/Desktop/GitHut/esp-idf/components/xtensa/include/esp_attr.h:111:34: note: previous declaration 'constexpr spi_event_t operator|(spi_event_t, spi_event_t)'
 FORCE_INLINE_ATTR constexpr TYPE operator| (TYPE a, TYPE b) { return (TYPE)((INT_TYPE)a | (INT_TYPE)b); } \
                                  ^~~~~~~~
C:/Users/mauro/Desktop/GitHut/esp-idf/components/xtensa/include/esp_attr.h:122:29: note: in expansion of macro 'FLAG_ATTR_IMPL'
 #define FLAG_ATTR_U32(TYPE) FLAG_ATTR_IMPL(TYPE, uint32_t)
                             ^~~~~~~~~~~~~~
C:/Users/mauro/Desktop/GitHut/esp-idf/components/xtensa/include/esp_attr.h:123:19: note: in expansion of macro 'FLAG_ATTR_U32'
 #define FLAG_ATTR FLAG_ATTR_U32
                   ^~~~~~~~~~~~~
C:/Users/mauro/Desktop/GitHut/esp-idf/components/soc/include/hal/spi_types.h:42:1: note: in expansion of macro 'FLAG_ATTR'
 FLAG_ATTR(spi_event_t)
 ^~~~~~~~~
C:/Users/mauro/Desktop/GitHut/esp-idf/components/xtensa/include/esp_attr.h:112:34: error: conflicting declaration of C function 'constexpr timer_intr_t operator&(timer_intr_t, timer_intr_t)'
 FORCE_INLINE_ATTR constexpr TYPE operator& (TYPE a, TYPE b) { return (TYPE)((INT_TYPE)a & (INT_TYPE)b); } \
                                  ^~~~~~~~
C:/Users/mauro/Desktop/GitHut/esp-idf/components/xtensa/include/esp_attr.h:122:29: note: in expansion of macro 'FLAG_ATTR_IMPL'
 #define FLAG_ATTR_U32(TYPE) FLAG_ATTR_IMPL(TYPE, uint32_t)
                             ^~~~~~~~~~~~~~
C:/Users/mauro/Desktop/GitHut/esp-idf/components/xtensa/include/esp_attr.h:123:19: note: in expansion of macro 'FLAG_ATTR_U32'
[... 189 lines cut ...]
 #define FLAG_ATTR_U32(TYPE) FLAG_ATTR_IMPL(TYPE, uint32_t)
                             ^~~~~~~~~~~~~~
C:/Users/mauro/Desktop/GitHut/esp-idf/components/xtensa/include/esp_attr.h:123:19: note: in expansion of macro 'FLAG_ATTR_U32'
 #define FLAG_ATTR FLAG_ATTR_U32
                   ^~~~~~~~~~~~~
C:/Users/mauro/Desktop/GitHut/esp-idf/components/soc/include/hal/timer_types.h:72:1: note: in expansion of macro 'FLAG_ATTR'
 FLAG_ATTR(timer_intr_t)
 ^~~~~~~~~
C:/Users/mauro/Desktop/GitHut/esp-idf/components/xtensa/include/esp_attr.h:120:25: note: previous declaration 'spi_event_t& operator<<=(spi_event_t&, int)'
 FORCE_INLINE_ATTR TYPE& operator<<=(TYPE& a, int b) { a <<= b; return a; }
                         ^~~~~~~~
C:/Users/mauro/Desktop/GitHut/esp-idf/components/xtensa/include/esp_attr.h:122:29: note: in expansion of macro 'FLAG_ATTR_IMPL'
 #define FLAG_ATTR_U32(TYPE) FLAG_ATTR_IMPL(TYPE, uint32_t)
                             ^~~~~~~~~~~~~~
C:/Users/mauro/Desktop/GitHut/esp-idf/components/xtensa/include/esp_attr.h:123:19: note: in expansion of macro 'FLAG_ATTR_U32'
 #define FLAG_ATTR FLAG_ATTR_U32
                   ^~~~~~~~~~~~~
C:/Users/mauro/Desktop/GitHut/esp-idf/components/soc/include/hal/spi_types.h:42:1: note: in expansion of macro 'FLAG_ATTR'
 FLAG_ATTR(spi_event_t)
 ^~~~~~~~~
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
`

I

Alvin1Zhang, 2020-08-06:
@rinripper Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, commit ID, hardware and etc. would help us debug further. Thanks.

robert-alfaro, 2020-08-20:
Same compilation errors for me..all from `esp_attr` component.
- commit `c77c4ccf6c43ab09fd89e7c907bf5cf2a3499e3b`
- versions of xtensa, binutils from IDF install.sh from commit above
- hardware: esp32 (devkitC)

0xjakob, 2020-08-21:
@rinripper @robert-alfaro What code did you compile? How can we reproduce this issue? 
@rinripper What was the previous ticket? What did you check and what looks fine?
Please keep in mind that it's very difficult to determine the error by just looking at error messages with almost no context. We're compiling C++ applications constantly on a daily basis in our CI and currently have no similar issue.

We actually have an issue template as @Alvin1Zhang suggested which also shows you how to format code and command line output and it goes through the questions I just asked.
There are a couple of engineers here which are very keen on helping you but we can't do this if there's not enough information. If you're afraid of disclosing confidential information, there are ways around that, too.

Hope to help you soon :)
Best,
Jakob

0xjakob, 2020-08-24:
@BorisKourt Thanks for the reproducer, we'll have a look into this!

0xjakob, 2020-09-08:
Hi @rinripper @robert-alfaro @BorisKourt 

This was an issue in IDF. The `esp_eth` component header files were declaring all includes `extern "C"` which conflicts with the C++ code in `esp_attr.h`. I removed these and now the Ableton project builds. Could you please try to apply the attached patch and check whether your code builds? We will fix this on master. It usually takes a few weeks until the fixes arrive and are back ported on github.

Overall, a good catch in the end, thank you all!

@BorisKourt I suggest to consider changing from `tcpip_adapter.h` to `esp_netif.h` as the former is deprecated. There should be enough examples using `esp_netif` in IDF. If you encounter problems, feel free to post in the forum (esp32.com) or open another issue if you suspect it's a problem in IDF.

[0001-esp_eth-moved-extern-C-after-includes.zip](https://github.com/espressif/esp-idf/files/5187606/0001-esp_eth-moved-extern-C-after-includes.zip)

BorisKourt, 2020-09-09:
Thank you for looking into this, and thank you for the fix. I will also look at switching to `esp_netif.h`.

Alvin1Zhang, 2020-11-13:
Thanks for reporting and sorry for late reply, the fix has been available at https://github.com/espressif/esp-idf/commit/5db977c512337136c4f07e00dbe27c5ba9883196, feel free to reopen if the issue still happens. Thanks.

closed 2020-11-13 by Alvin1Zhang