espressif/esp-idf #5700: certificat MQTTS with web server

SERIDJ, 2020-08-05:
hi,

 iam using  esp idf mqtt library : https://github.com/espressif/esp-idf/blob/master/examples/protocols/mqtt/ssl_mutual_auth/main/app_main.c 

and i want to use some web page to set configuration for this library like wifi/ethernet and mqtt config  ip,port....

I would like to be able to connect to the ESP32 network on another device, open a web browser, navigate to some address, and have ESP32 serve a webpage from its flash memory.

and how i can send certificate with web page to esp32 bcs its need to be convert to assembly format ca.S and client.crt.S, client.key.S ....

ESP-Marius, 2020-08-06:
Hi, 

Not sure what you mean by **for objcopy compatibility**. You can send it as .pem or .der and store it in your flash.

SERIDJ, 2020-08-06:
hi @ESP-Marius 

yes its mentioned in file : 

```
/* * Data converted from C:/Users/mohammed.seridj/Desktop/wrkspse/ssl_mutual_auth/main/client.crt
 * (null byte appended)
 */
.data
.section .rodata.embedded

.global client_crt
client_crt:

.global _binary_client_crt_start
_binary_client_crt_start: /* for objcopy compatibility */
.byte 0x50, 0x6c, 0x65, 0x61, 0x73, 0x65, 0x20, 0x70, 0x61, 0x73, 0x74, 0x65, 0x20, 0x79, 0x6f, 0x75
.byte 0x72, 0x20, 0x63, 0x6c, 0x69, 0x65, 0x6e, 0x74, 0x20, 0x63, 0x65, 0x72, 0x74, 0x69, 0x66, 0x69
.byte 0x63, 0x61, 0x74, 0x65, 0x20, 0x68, 0x65, 0x72, 0x65, 0x20, 0x28, 0x66, 0x6f, 0x6c, 0x6c, 0x6f
.byte 0x77, 0x20, 0x69, 0x6e, 0x73, 0x74, 0x72, 0x75, 0x63, 0x74, 0x69, 0x6f, 0x6e, 0x73, 0x20, 0x69
.byte 0x6e, 0x20, 0x52, 0x45, 0x41, 0x44, 0x4d, 0x45, 0x2e, 0x6d, 0x64, 0x29, 0x0d, 0x0a, 0x00

.global _binary_client_crt_end
_binary_client_crt_end: /* for objcopy compatibility */

.global client_crt_length
client_crt_length: /* not including null byte */
.word 78

```

-how i can add null byte (i dont know  if its addred automatically when i convert crt to der format ) ?

- and  how about those variable .global _binary_client_crt_start/.global _binary_client_crt_end  ?

ESP-Marius, 2020-08-07:
What you are looking at there is a file that will be assembled and linked into the final binary during the build process, and it's not something you would send to the ESP32. 

If you look at the MQTT examples you'll see that the certificate input is simply a  `uint8_t` array.

ESP-Marius, 2020-08-07:
Closing issue, I recommend using our forum, https://www.esp32.com/, for "How to" questions.

closed 2020-08-07 by ESP-Marius