espressif/esp-idf #5702: V4.2的蓝牙有bug，必现问题。

guo652917087, 2020-08-06:
Hi，
   你好，我刚刚更新了V4.2的分支：0bcd26a6106cf91a455a66a287c5885125af22f8
    就发现蓝牙有问题，直接测试blufi的例子，就出现如下日志：

rst:0xc (SW_CPU_RESET),boot:0x16 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7172
load:0x40078000,len:13696
load:0x40080400,len:4000
entry 0x40080688
I (29) boot: ESP-IDF v4.2-dev-1995-g636b964c8-dirty 2nd stage bootloader
I (29) boot: compile time 10:22:40
I (30) boot: chip revision: 3
I (34) boot.esp32: SPI Speed      : 40MHz
I (39) boot.esp32: SPI Mode       : DIO
I (43) boot.esp32: SPI Flash Size : 4MB
I (48) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (72) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (79) boot:  2 factory          factory app      00 00 00010000 00100000
I (87) boot: End of partition table
I (91) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x24adc (150236) map
I (157) esp_image: segment 1: paddr=0x00034b04 vaddr=0x3ffbdb60 size=0x040f8 ( 16632) load
I (164) esp_image: segment 2: paddr=0x00038c04 vaddr=0x40080000 size=0x00404 (  1028) load
I (165) esp_image: segment 3: paddr=0x00039010 vaddr=0x40080404 size=0x07008 ( 28680) load
I (185) esp_image: segment 4: paddr=0x00040020 vaddr=0x400d0020 size=0xaa4c4 (697540) map
I (451) esp_image: segment 5: paddr=0x000ea4ec vaddr=0x4008740c size=0x16ce4 ( 93412) load
I (507) boot: Loaded app from partition at offset 0x10000
I (508) boot: Disabling RNG early entropy source...
I (508) psram: This chip is ESP32-D0WD
I (513) spiram: Found 64MBit SPI RAM device
I (517) spiram: SPI RAM mode: flash 40m sram 40m
I (522) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (530) cpu_start: Pro cpu up.
I (533) cpu_start: Application information:
I (538) cpu_start: Project name:     blufi_demo
I (543) cpu_start: App version:      v4.2-dev-1995-g636b964c8-dirty
I (550) cpu_start: Compile time:     Aug  6 2020 10:20:49
I (556) cpu_start: ELF file SHA256:  2b6610d6b62546c5...
I (562) cpu_start: ESP-IDF:          v4.2-dev-1995-g636b964c8-dirty
I (569) cpu_start: Starting app cpu, entry point is 0x40081af8
I (562) cpu_start: App cpu up.
I (1464) spiram: SPI SRAM memory test OK
I (1464) heap_init: Initializing. RAM available for dynamic allocation:
I (1465) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (1471) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (1477) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (1483) heap_init: At 3FFCC7B8 len 00013848 (78 KiB): DRAM
I (1489) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1496) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1502) heap_init: At 4009E0F0 len 00001F10 (7 KiB): IRAM
I (1508) cpu_start: Pro cpu start user code
I (1513) spiram: Adding pool of 4096K of external SPI memory to heap allocator
I (1535) spi_flash: detected chip: gd
I (1535) spi_flash: flash io: dio
W (1535) spi_flash: Detected size(16384k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (1546) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (1556) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (1706) wifi:wifi driver task: 3ffd87d0, prio:23, stack:3584, core=0
I (1706) system_api: Base MAC address is not set
I (1706) system_api: read default base MAC address from EFUSE
I (1726) wifi:wifi firmware version: 9f3395e
I (1726) wifi:wifi certification version: v7.0
I (1726) wifi:config NVS flash: enabled
[... 1315 lines cut ...]
# end of Wear Levelling

#
# Wi-Fi Provisioning Manager
#
CONFIG_WIFI_PROV_SCAN_MAX_ENTRIES=16
CONFIG_WIFI_PROV_AUTOSTOP_TIMEOUT=30
# end of Wi-Fi Provisioning Manager

#
# Supplicant
#
CONFIG_WPA_MBEDTLS_CRYPTO=y
# CONFIG_WPA_DEBUG_PRINT is not set
# CONFIG_WPA_TESTING_OPTIONS is not set
# CONFIG_WPA_TLS_V12 is not set
# CONFIG_WPA_WPS_WARS is not set
# end of Supplicant
# end of Component config

#
# Compatibility options
#
# CONFIG_LEGACY_INCLUDE_COMMON_HEADERS is not set
# end of Compatibility options

guo652917087, 2020-08-06:
我回退回 版本 00148cd0c6873cc65aa7edfab83566b517832155 分支，就正常。

redchenjs, 2020-08-06:
Same issue.

Alvin1Zhang, 2020-08-10:
Thanks for reporting and sorry for slow turnaround, the fix is available at https://github.com/espressif/esp-idf/commit/10e59593f6e3fdb50e7ec3373f04a58aa5ad2732. Feel free to reopen if the issue still happens. Thanks.

closed 2020-08-10 by Alvin1Zhang